Koenkk/zigbee2mqtt #7727: How to use and install Herdsman Converters

i-000, 2021-06-07:
Hello,

I am migrating from domoticz to hassio, and I am newbie in this platform. I use ESP devices, RFXcom, Z-wave and Zigbee, and I am in the process to migrate all to HA.

I read, and don't find, how use or install Herdsman Converters.

I download all files of Herdsman from github, but the next step?

I read to create the js file, but how I install the Herdsman converters. Is necessary copy in a folder, or execute an order to install.

Maybe is a simple question, but I read and don't find how to use. Please help me use it

Thanks!!!!

robert-pajak, 2021-06-07:
I am not an expert either, but according to my understanding zigbee-herdsman-converters is not an independent project (it is a zigbee2mqtt vendor library)
The responsibility for this library is to provide device's specific conversions (but within Zigbee protocol family)
The whole zigbee2mqtt deployable project (i.e docker image) consist of `main` zigbee2mqtt + zigbee-herdsman + zigbee-herdsman-converters + recently added web UI

Long story short - if you want to try HASS you need:
* Mosquitto (https://hub.docker.com/_/eclipse-mosquitto)
* Zigbee2mqtt (https://hub.docker.com/u/zigbee2mqtt)
* HASS (https://hub.docker.com/r/homeassistant/home-assistant/)

ESP (Tasmota) devices could/should communicate with HASS over 2.4GHz wifi, connecting to your Mostqutto/MQTT server
I don't have any experience regarding RFXcom nor Z-wave so I can't comment

i-000, 2021-06-08:
> I am not an expert either, but according to my understanding zigbee-herdsman-converters is not an independent project (it is a zigbee2mqtt vendor library)
> The responsibility for this library is to [...]

Hi robert-pajak. Thank you for your help!!!

I  need help to use and install herdsman converters. I have Mosquito broker and Zigbee2mqtt add on. To use Herdsmans Converters, is necessary use another mosquitto like eclipse? 

My main doubt is how to get install herdsman converters.

Thanks!!!

robert-pajak, 2021-06-08:
Hi i-000,

If you have properly installed Zigbee2mqtt  you already have zigbee-herdsman-converters  - check content of the following directory: `/app/node_modules/zigbee-herdsman-converters` in your docker zigbee2mqtt instance
Zigbee-herdsman-converters seems to be a specific library, not deployable independent project/application - Zigbee2mqtt orchestrates everything and behind a scene it uses herdsman converters library when communicating with a specific device(s)

robert-pajak, 2021-06-08:
Do you use zigbee2mqtt docker images? 
Or you tried to git pull zigbee2mqtt project? If so you need to ensure all libraries are included and whole zigbee2mqtt application is assembled and ready to go :)
If you are using docker that's how you can see its directory content:
[image]
[image]

i-000, 2021-06-08:
> Do you use zigbee2mqtt docker images?
> Or you tried to git pull zigbee2mqtt project? If so you need to ensure all libraries are included and whole zigbee2mqtt application is assembled and ready to go :)
> If you are using docker that's how you can see its directory content:
> [image]
> [image]

Hi Robert_pajak, like I am newbie in Hassio, I don't understand what is docker, sorry!! If I execute this command docker ps, the system return command not found, maybe my installation is different like you.

I am very grateful for your help

i-000, 2021-06-08:
> You need to provide some basic information, before someone could help you.
> What installation method did you use?
> 
> https://www.home-assistant.io/installation/ :
> Home Assistant offers four different installation methods.
> 
> * **Home Assistant Operating System:** Minimal Operating System optimized to power Home Assistant. It comes with Supervisor to manage Home Assistant Core and pre-installed add-ons. Recommended installation method.
> * **Home Assistant Container:** Standalone container-based installation of Home Assistant Core (e.g. Docker).
> * **Home Assistant Supervised:** Manual installation of the Supervisor.
> * **Home Assistant Core:** Manual installation using Python virtual environment.
> 
> Did you install **Home Assistant Operating System** and used `Supervisor --> Add-on Store` to apply **Mosquito broker add-on** and **Zigbee2mqtt add on**?
> If yes, you already got zigbee-herdsman-converters available. You simply can not see them, because HassOS is access restricted. **Samba share add-on** does not give you access to everything, same for **Terminal & SSH add-on**.
> 
> Why are you asking for zigbee-herdsman-converters? What do you try to achieve?

Hello Alex Kay88!,

I have a Rasp 4. Home Installed executing this command https://github.com/home-assistant/operating-system/releases/download/5.13/hassos_rpi4-64-5.13.img.xz flashing a SD card.

Yes I have installed Mosquito broker add-on and Zigbee2mqtt add on.

I want to add a Moes device. In GitHub there are moes.js file, and I am interested add to it the Moes 4 gang, that in this moment is not implemented. 

I think the Moes.js, is not working properly, like I can't see the Herdsman folders I am thinking are not installed, and Moes.js not work fine.

I add the Moes.js file to zigbee2mqtt folder. Adding moes.js line inside configuration.yaml. But continue manufacturer unsupported, and getting the message no recognized device, at every button press.

Thank You!!!

AlexKay88, 2021-06-08:
So, you are running hassOS on Raspberry Pi4.
That's a very good choice, if you are new to Home Assistant, since it includes everything you need to run it without the hassle of installing every single component yourself manually.

Z2M is completely installed after adding Zigbee2MQTT add-on, so you don't need to install herdsman-converters yourself. You don't have full filesystem access. That's why you can't see the folders you are looking for, but they are there.

Which moes.js do you mean? The one from Z2M GitHub repository?
Pls provide a link to moes.js here.
Pls. also check Z2M supported devices and provide a link to the Moes device you want to use.
Is it this one?
https://www.zigbee2mqtt.io/devices/TS0601_switch.html

i-000, 2021-06-08:
> So, you are running hassOS on Raspberry Pi4.
> That's a very good choice, if you are new to Home Assistant, since it includes everything you need to run it without the hassle of installing every [...]

Hi AlexKay88, thanks for your info, now I have a better view of the system.

The new device that I want use is this one (Model TS004F):

https://es.aliexpress.com/item/1005002133026707.html?acnt=439-079-4345&aff_platform=aaf&ds_e_device=c&albcp=11505901136&ds_e_product_id=es1005002133026707&ds_url_v=2&ds_dest_url=https%3A%2F%2Fs.click.aliexpress.com%2Fdeep_link.htm%3Faff_short_key%3DUneMJZVf&ds_e_product_group_id=743612850874&sk=UneMJZVf&ds_e_adid=476171223963&terminal_id=5236958d0fb44987bb5d94b8a7572378&tmLog=new_Detail&needSmbHouyi=false&albbt=Google_7_shopping&ds_e_product_channel=online&src=google&ds_e_product_country=ES&aff_fcid=a3c30395196649d894d7368ab096588f-1623181236149-09765-UneMJZVf&gclid=EAIaIQobChMI4-XJpeWI8QIVaRkGAB3SMgR5EAQYASABEgINA_D_BwE&albag=115252975111&aff_fsk=UneMJZVf&albch=shopping&ds_e_network=u&albagn=888888&ds_e_product_language=es&isSmbAutoCall=false&ds_e_product_merchant_id=107851193&aff_trace_key=a3c30395196649d894d7368ab096588f-1623181236149-09765-UneMJZVf&gclsrc=aw.ds

Is a Tuya compatible device, but I don't want use Tuya app and either local tuya.

Thank you so much!!!!!

AlexKay88, 2021-06-08:
I can't judge from product picture at AliExpress, if this device is already supported by Z2M or not.
Model TS004F is a good start, but also not enough.

Please share
1) link to moes.js file you referred
2) data/database.db entry for your Moes device
3) did you see a log line like
`Device '0x[ID]' with Zigbee model 'XXXX' and manufacturer name 'YYYYYYYY' is NOT supported, please follow...
`
Please share this line.

i-000, 2021-06-08:
> I can't judge from product picture at AliExpress, if this device is already supported by Z2M or not.
> Model TS004F is a good start, but also not enough.
>
> Please share
>
> 1. link to moes.js file you [...]

I share the info:
1.
This is the file with other Moes devices, but this model is not inside.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/moes.js
2.
{"id":3,"type":"EndDevice","ieeeAddr":"0x60a423fffed5c32f","nwkAddr":8738,"manufId":4098,"manufName":"_TZ3000_xabckq1v","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_xabckq1v","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1623180942054}

3.
Received message from unsupported device with Zigbee model 'TS004F' and manufacturer name '_TZ3000_xabckq1v'

AlexKay88, 2021-06-08:
OK, let's summarize what we got up to now:
**Environment:** hassOS with Zigbee2MQTT add-on
**Device:** Moes 4-gang switch (Zigbee model 'TS004F' and manufacturer name '_TZ3000_xabckq1v')
**--> Device is currently not supported by Z2M.**

Above referenced moes.js is already included in Z2M add-on.
You receive "not supported" log message, because model 'TS004F' / manufacturer name '_TZ3000_xabckq1v' is not included in any `zigbee-herdsman-converters/blob/master/devices/*.js` files.

**To support new devices you need to follow:**
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html   and
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

AlexKay88, 2021-06-08:
TS004F/_TZ3000_xabckq1v has been added to Z2M 2 weeks ago:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2624

i-000, 2021-06-09:
> OK, let's summarize what we got up to now:
> **Environment:** hassOS with Zigbee2MQTT add-on
> **Device:** Moes 4-gang switch (Zigbee model 'TS004F' and manufacturer name '_TZ3000_xabckq1v')
> 
> Device is already included in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js :
> fingerprint: [{modelID: 'TS004F', manufacturerName: '_TZ3000_xabckq1v'}],
> 
> Thus it is already supported by Z2M, but hassOS Zigbee2mqtt add-on still shows it as "not supported".
> 
> @i-000 : Did you apply `Zigbee2mqtt Edge` add-on or without **Edge**? If without "Edge", try Edge version to get latest device definitions.
> If Edge version does not help to get rid of "not supported" log message, Edge version is outdated in comparison to Z2M and needs to be updated by add-on maintainer.

Hi AlexKay88, I am using not Edge Zigbee2mqtt. I install the Edge version, and now is recognized!!!!! And I see in the log start the herdsman converters. Now is time to fight with the settings.

Thanks a lot AlexKay88 for your help and your time!!!!!!

closed 2021-06-09 by i-000

jocamane, 2022-01-08:
If i install zigbee2mqtt my TS0014 apear with "unsupported".
If i install zigbee2mqtt edge it's work ok.
But i want use the z2m normal version.
Why ? is bug?

Koenkk, 2022-01-09:
@jocamane wait for the next release (on 1 february)

jjusko, 2022-02-07:
Hi , will be blitzwolf bw-SHP 15 supported in new version?
Expected date of release?

ipn00tic, 2023-03-08:
> 

Hello,

I have two different unsourced water leak devices but both have the same behavior.

I put it in pairing mode but it is not detected, the message " Device '0x[ID]' with Zigbee model 'XXXX' and manufacturer name 'YYYYYYYY' is NOT supported, please follow..."

what should I do?

ipn00tic, 2023-03-08:
> I can't judge from product picture at AliExpress, if this device is already supported by Z2M or not. Model TS004F is a good start, but also not enough.
>
> Please share
>
> 1. link to moes.js file you [...]

Hello,

I have two different unsourced water leak devices but both have the same behavior.

I put it in pairing mode but it is not detected, the message " Device '0x[ID]' with Zigbee model 'XXXX' and manufacturer name 'YYYYYYYY' is NOT supported, please follow..."

what should I do?