Koenkk/zigbee2mqtt #7728: TuYa TS0505B  blink and error when try change color

shirou93, 2021-06-07:
TuYa TS0505B https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505

Zigbee2MQTT1.19.1 commit: 1.19.1
zStack3x0 20210120

When try change color (sometimes working) led strip blink and error in logs.

Color temp work perfectly but when try change color x_y i have error in logs and led driver instantly not responding.

logs `Zigbee2MQTT:error 2021-06-07 22:05:12: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":32182,"colory":28086}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51782 - 1 - 26 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-06-07 22:05:35: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":26904,"colory":32078}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51782 - 1 - 27 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-06-07 22:05:58: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":19418,"colory":38836}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51782 - 1 - 28 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-06-07 22:10:37: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":11862,"colory":47382}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51782 - 1 - 70 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-06-07 22:11:55: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":19418,"colory":38836}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51782 - 1 - 80 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-06-07 22:12:13: Publish 'set' 'color_temp' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColorTemp({"colortemp":370,"transtime":0}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:error 2021-06-07 22:12:30: Publish 'set' 'state' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'`

Can you check this? @sjorge  thanks

sjorge, 2021-06-07:
@Koenkk those look like a mix of timeout/mac failure, so nothing to do with the converters right?

shirou93, 2021-06-07:
@sjorge  i left driver 15 minutes without ac and error disappeared when repaired.  Sorry for wasted your time.

closed 2021-06-07 by shirou93