Koenkk/zigbee2mqtt #7732: Docker Zigbee2MQTT run

jos149, 2021-06-08:
## What happened
Succesfully installed and running Zigbee2MQTT on Docker. I was wondering how I could log into the console of the Docker image and run Zigbee2MQTT like the normal installation. I managed to log in to the console with `/bin/sh` and the was wondering: how do I start? Running `npm zigbee2mqtt` does not work? I cannot `cd /opt/zigbee2mqtt` so how does this work then? Another thing is, I am logging in into the console with portainer. It throws me out after one minute. Doing it over terminal it throws me out after 1 second. Any help?

## What did you expect to happen
Not being thrown out.

## How to reproduce it (minimal and precise)
Explained above.

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC2531
Adapter firmware version: latest

tomdudu38, 2021-06-08:
hello
isn't your container restarting more and more? 
what says logs for the container (in portainer).

jos149, 2021-06-09:
I found that my container is just running. But in bin/ash everything works so pleased with that!

But now the problem is: how can I run zigbee2mqtt to start connecting zigbee devices?

tomdudu38, 2021-06-09:
if your containr is running, all you have to do is , allow devices to join zigbee2mqtt.and press the button on the device to join.
zigbee2mqtt should be alreaby running if you take this image : https://hub.docker.com/r/koenkk/zigbee2mqtt/
zigbee2mqtt have a http frontend: https://www.zigbee2mqtt.io/information/frontend.html
an it have a configuration.yaml.
the compose : 
`
  version: '3'
  services:
    zigbee2mqtt:
      container_name: zigbee2mqtt
      image: koenkk/zigbee2mqtt
      volumes:
        - ./data:/app/data
        - /run/udev:/run/udev:ro
      devices:
        - /dev/ttyACM0:/dev/ttyACM0
      restart: always
      network_mode: host
      privileged: true
      environment:
        - TZ=Europe/Amsterdam
`
here you can see the volume app/data mounted in ./data where should be your configuration.yaml

tomdudu38, 2021-06-09:
permit_join=true should be enough to get it work
if u press a paring button on a device, you should see the announcement of paring the device in logs,
there is nothing to start  but the container.

jos149, 2021-06-09:
> 
> 
> if your containr is running, all you have to do is , allow devices to join zigbee2mqtt.and press the button on the device to join.
> zigbee2mqtt should be alreaby running if you took this image : https://hub.docker.com/r/koenkk/zigbee2mqtt/
> zigbee2mqtt have a http frontend: https://www.zigbee2mqtt.io/information/frontend.html
> and / or it have a configuration.yaml.

Ok, what is currently happening is that the docker image is running (I am using the koenkk image).
I installed the frontend.

> 
> here (in docker compose or docker command line) you can see the volume app/data mounted in ./data where should be your configuration.yaml:
> https://www.zigbee2mqtt.io/information/docker.html

Within the configuration.yaml which is also in the ./data folder on the host, permit_join is set to TRUE!

> 
> if you want to know how it start you should take a look at the docker file but it's useless
> https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/Dockerfile

I managed to start it and it keeps running properly, so also no problems here. The sensors are just not being detected....

tomdudu38, 2021-06-09:
yes, yes understood.i corrected my previous post sorry:

permit_join=true should be enough to get it work
if u press a paring button on a device, you should see the announcement of paring the device in logs,
there is nothing to start but the container.

jos149, 2021-06-09:
This is the exact thing I am doing. The problem is: nothing happens!

jos149, 2021-06-09:
Setting `permit_join=true` lights up the LED on my CC2531. When set it to false the LED turns off. Maybe this helps?

tomdudu38, 2021-06-09:
ok, so seems your coordinator eard nothing
i would try to reflash de coordinator and i see that in faq:

https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

CC2531/CC2530 coordinator only:
Stop Zigbee2MQTT, unplug the coordinator, wait 10 seconds, plug the coordinator, start Zigbee2MQTT and try to pair the device again.
If none of the above helps, try to reflash the coordinator (does not require repairing of already paired devices).

jos149, 2021-06-09:
Using a usb extension cable did the trick!

closed 2021-06-10 by jos149