Koenkk/zigbee2mqtt #7736: MQTT Invalid color mode received hassio ZNLDP12LM

tomdudu38, 2021-06-08:
## What happened
"MQTT Invalid color mode received" spam in hassio log

hassio:
 `
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:247
Integration: MQTT (documentation, issues)
First occurred: 8 juin 2021, 16:32:41 (133 occurrences)
`
MQTT :
`
Copy to clipboard
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    },
    {
      "topic": "zigbee2mqtt/lampe3/availability"
    }
  ],
  "brightness": true,
  "brightness_scale": 254,
  "color_mode": true,
  "command_topic": "zigbee2mqtt/lampe3/set",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x00158d0004850447"
    ],
    "manufacturer": "Xiaomi",
    "model": "Aqara smart LED bulb (ZNLDP12LM)",
    "name": "lampe3",
    "sw_version": "Zigbee2MQTT 1.19.1"
  },
  "effect": true,
  "effect_list": [
    "blink",
    "breathe",
    "okay",
    "channel_change",
    "finish_effect",
    "stop_effect"
  ],
  "json_attributes_topic": "zigbee2mqtt/lampe3",
  "max_mireds": 500,
  "min_mireds": 150,
  "name": "lampe3",
  "schema": "json",
  "state_topic": "zigbee2mqtt/lampe3",
  "supported_color_modes": [
    "color_temp"
  ],
  "unique_id": "0x00158d0004850447_light_zigbee2mqtt"
}
`
zigbee2mqtt:
`
MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":239,"linkquality":99,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'

`
## What did you expect to happen
no log
## How to reproduce it (minimal and precise)
use  some ZNLDP12LM and make a light groupe in hassio
## Debug info
Zigbee2MQTT version: 
latest
Adapter hardware:
 CC2531
Adapter firmware version:
 latest

Koenkk, 2021-06-10:
This happens because you are on an older HA version, make sure to be on 2021.4.5 as least as mentioned in the changelog: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.3

tomdudu38, 2021-06-13:
i am on 2021.6.3

Koenkk, 2021-06-14:
What z2m version are you using? If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

tomdudu38, 2021-06-14:
i pull the latest tag i, so  1.19.0.
should i try latest-dev?

Koenkk, 2021-06-15:
Can you try if this issue disappears when setting the color temp via HA? Does the error then appear when setting the XY color of the group?

tomdudu38, 2021-06-16:
I tried setting groups and individual lights color_temp via HA  but the issue still there.
And, i tried some custom color_mode (xy, true , false,) in the customize.yaml it does nothing much.
So removed lights groups. and still happen.
I don't  think it's a group issue.
but:
if i look at mqtt log , i can see:
 "color_mode": true
and zigbee2mqtt send :
 color_mode: xy

Koenkk, 2021-06-16:
When setting the color temp of the individual light in HA, does z2m send `color_mode: "color_temp"`? (can you provide the debug log of that).

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

tomdudu38, 2021-06-18:
Publishing 'set' 'color_temp' to 'lampe1'
Debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x00158d0004850409!
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":307,"tuyaRgbMode":117440598}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"xy","color_temp":307,"linkquality":60,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":307,"tuyaRgbMode":4406784}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"xy","color_temp":307,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"xy","color_temp":307,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Failed to ping 'lampe3'
Debug Received MQTT message on 'zigbee2mqtt/lampe1/set' with data '{"state": "ON", "color_temp": 226}'
Debug Skipping state because of Home Assistant
Debug Publishing 'set' 'color_temp' to 'lampe1'
Debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x00158d0004850409!

Koenkk, 2021-06-18:
Could you provide your `configuration.yaml` file?

tomdudu38, 2021-06-19:
sure:
`
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.253'
serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
devices:
  '0x00124b001b783bed':
    friendly_name: htemp1
  '0x00158d0004850447':
    friendly_name: lampe3
    retain: true
    optimistic: true
  '0x00158d0004850409':
    friendly_name: lampe1
    retain: true
    optimistic: true
  '0x00158d000484ec5b':
    friendly_name: lampe2
    retain: true
    optimistic: true
  '0x00158d00042f4d67':
advanced:
  availability_timeout: 60
  availability_passlist:
    - lampe1
    - lampe2
    - lampe3
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  log_level: debug
ota:
  disable_automatic_update_check: true
`

Koenkk, 2021-06-22:
Can you try again with the latest dev branch? After upgrading set the color temperature at least once and provide the complete debug log starting at the point where you set the color temperature of the individual bulb via HA till 10 seconds after this.

tomdudu38, 2021-06-23:
sure 
Received Zigbee message from 'cc2530', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 8 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/cc2530', payload '{"led":true,"linkquality":84}'
Debug Received Zigbee message from 'lampe3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Debug Successfully pinged 'lampe3'
Debug Received MQTT message on 'zigbee2mqtt/lampe3/set' with data '{"state": "ON", "color_temp": 317}'
Debug Skipping state because of Home Assistant
Debug Publishing 'set' 'color_temp' to 'lampe3'
Debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x00158d0004850447!
Debug Received Zigbee message from 'lampe3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":317,"tuyaRgbMode":117440660}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":193,"color_mode":"xy","color_temp":317,"linkquality":92,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":193,"color_mode":"color_temp","color_temp":317,"linkquality":92,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":317,"tuyaRgbMode":3324416}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":193,"color_mode":"color_temp","color_temp":317,"linkquality":92,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Debug Successfully pinged 'lampe2'
Debug Received MQTT message on 'zigbee2mqtt/lampe2/set' with data '{"state": "ON", "color_temp": 312}'
Debug Skipping state because of Home Assistant
Debug Publishing 'set' 'color_temp' to 'lampe2'
Debug Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x00158d000484ec5b!
Debug Received Zigbee message from 'lampe2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":312,"tuyaRgbMode":117440662}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":193,"color_mode":"xy","color_temp":312,"linkquality":55,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":193,"color_mode":"color_temp","color_temp":312,"linkquality":55,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":312,"tuyaRgbMode":11021824}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":193,"color_mode":"color_temp","color_temp":312,"linkquality":52,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received MQTT message on 'zigbee2mqtt/lampe1/set' with data '{"state": "ON", "brightness": 97}'
Debug Publishing 'set' 'brightness' to 'lampe1'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":117440663,"currentLevel":97}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"color_temp","color_temp":327,"linkquality":63,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'genOnOff', data '{"61440":4406784,"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"color_temp","color_temp":327,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"color_temp","color_temp":327,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":4406784,"currentLevel":97}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":97,"color_mode":"color_temp","color_temp":327,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received MQTT message on 'zigbee2mqtt/lampe1/set' with data '{"state": "ON", "brightness": 99}'
Debug Publishing 'set' 'brightness' to 'lampe1'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":117440664,"currentLevel":99}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":99,"color_mode":"color_temp","color_temp":327,"linkquality":63,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
[... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":99,"color_mode":"color_temp","color_temp":306,"linkquality":63,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe1', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440666,"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":99,"color_mode":"color_temp","color_temp":306,"linkquality":63,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Koenkk, 2021-06-26:
Does HA still throw the error now? The payload looks good now: `"color_mode":"color_temp"`

tomdudu38, 2021-06-27:
thanks you koenkk.
today my payload was :
MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":82,"color_mode":"color_temp","color_temp":291,"linkquality":49,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":82,"color_mode":"color_temp","color_temp":291,"linkquality":47,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":110,"color_mode":"xy","color_temp":275,"linkquality":89,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":110,"color_mode":"xy","color_temp":275,"linkquality":92,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":110,"color_mode":"xy","color_temp":275,"linkquality":92,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe2', payload '{"brightness":110,"color_mode":"xy","color_temp":275,"linkquality":89,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":299,"linkquality":99,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":299,"linkquality":99,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":299,"linkquality":99,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":299,"linkquality":97,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/cc2530', payload '{"led":true,"linkquality":89}'
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":82,"color_mode":"color_temp","color_temp":291,"linkquality":47,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe1', payload '{"brightness":82,"color_mode":"color_temp","color_temp":291,"linkquality":42,"power_outage_memory":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'

BUT i  rebooted hassio, set each bulb color_temp in hassio and the logs disapeared,and i can't see those bad  logs come back for now.

tomdudu38, 2021-06-27:
the error still there:
please look my logs, what i did:
- set color_temp in hassio (xy--->color_temp)
- cut the power(physical switch off, no electricity) of the light
- put the light on (physical switch_on) (color_temp--->xy)

Received MQTT message on 'zigbee2mqtt/lampe3/set' with data '{"state": "ON", "color_temp": 269}'
Debug Publishing 'set' 'color_temp' to 'lampe3'
Debug Received Zigbee message from 'lampe3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":269,"tuyaRgbMode":117440744}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'lampe3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":269,"tuyaRgbMode":3324416}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"color_temp","color_temp":269,"linkquality":89,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Device 'lampe3' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"lampe3","ieee_address":"0x00158d0004850447"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"lampe3"},"type":"device_announced"}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"color_temp","color_temp":269,"linkquality":84,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"color_temp","color_temp":269,"linkquality":86,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":269,"linkquality":84,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'lampe3', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":269,"linkquality":94,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/lampe3', payload '{"brightness":115,"color_mode":"xy","color_temp":269,"linkquality":86,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'

Koenkk, 2021-06-28:
Does this issue disappear when setting `availability_timeout` to `0` in your configuration?

tomdudu38, 2021-06-30:
the issue has disappeared with avaibility_timeout to 0.
thanks you sir.

nicx, 2021-08-10:
Unfortunately setting availability_timeout to 0 does not help, still getting a lot of these warning messages:

```
2021-08-10 08:44:26 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received
2021-08-10 08:44:26 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received
2021-08-10 08:44:26 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received
2021-08-10 08:44:26 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid or incomplete color value received
```

@Koenkk any other idea? Should I create another Issue or shall we reopen this one?

hmorkemo, 2021-09-13:
> Unfortunately setting availability_timeout to 0 does not help, still getting a lot of these warning messages:
>
> ```
> 2021-08-10 08:44:26 WARNING (MainThread) [...]

This still happens here as well. zigbee2mqtt 1.21.1, HA Core 2021.9.6, availability_timeout: 0

sygys, 2021-09-14:
Im not sure why this thread is closed. The color mode en and color value warnings are still not fixed. Besides that i get around 200 - 300 warnings a day about the click attribute.

z2mqtt is spamming an awful lot warnings in my logs with no reason.

Im using 100% hue lights which all are supported by z2mqtt and all work like expected. Still i get these warnings allot. Im running z2m version 1.21.1-1, home assistant core 2021.9.3, supervisor 2021.09.0, home assistant OS 6.3

In Which version will this be fixed? and even more important what is causing this? It doesn't make sense to set availability_timeout to 0 to fix a wrong color mode. or any other of the 500 warnings a day i get from z2m

tomdudu38, 2021-09-16:
hi....
this topic was not about hue light, not about color_value. i have no more log since few months.closed.

hmorkemo, 2021-10-11:
I've had both the 1) click attribute and 2) invalid color mode spams, for a long time, also in HA 2021.10.2. But, now I've found a solution to problem 2! The messages occur only for light devices with color temperature capabilities. And ONLY for those where the color temp has never been set. (Seems like a typical uninitialized data error). Once I set the color temp for all my bulbs, the message no longer appears. But problem 1 still isn't solved.

tomdudu38, 2021-11-07:
today, i got this "color mode log back". with hassio and/or  zb2mqtt update.
Logs disapears if setting the color temp, in hassio 
hassio: 2021.11.1
zb2mqtt:1.22.0 commit: 0a1fccf

tomdudu38, 2021-11-18:
my new configuration.yaml:

'0x00158d0004850409':
    friendly_name: lampe1
    retain: true
    optimistic: true
    filtered_attributes: ["color_mode"]
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500

RadioLed2000, 2021-12-06:
z2m sends "color_mode"="xy", but Home Assistant expects boolean (true or false). [https://www.home-assistant.io/integrations/light.mqtt/#color_mode](https://www.home-assistant.io/integrations/light.mqtt/#color_mode)
 This causes an error. 
I add "color_mode" to Filtered attributes and have no errors anymore.
Maybe this option is not needed. This lamp doesn't support colors. Only color temp.

RadioLed2000, 2021-12-06:
BTW, there are wrong min and max values for color_temp. Right are 153-370
https://github.com/Koenkk/zigbee2mqtt/issues/621#issuecomment-483214288

adekloet, 2021-12-17:
the error is back: 2021-12-17 12:03:35 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received

availability_timeout: 0 does not fix the error, tyhe light control is still working from HA though.

Rjaap, 2021-12-18:
> the error is back: 2021-12-17 12:03:35 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
> 
> availability_timeout: 0 does not fix the error, tyhe light control is still working from HA though.

Same here, update to home assistant latest core 2021-12.3, light control works as expected but throws an error: invalid color mode received.

NicolaiPetri, 2022-01-09:
I also see this on some Ikea devices, and it makes the HA powercalc module unhappy:
2022-01-09 16:18:00 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
2022-01-09 16:18:00 DEBUG (MainThread) [custom_components.powercalc.strategy.lut] light.livingroomfloorlamp1: Could not calculate power. color mode unknown
Only happens for some specific bulb types, this bulb is reported as "IKEA LED1903C5/LED1835C6" in Z2M

dummylabs, 2022-02-11:
> I add "color_mode" to Filtered attributes and have no errors anymore. 
@RadioLed2000 How did you achieve this?

RadioLed2000, 2022-02-11:
> How did you achieve this?

Now in the latest Zigbee2mqtt version: 1.23.0-1 I remove "color_mode" from Filtered attributes and have no errors.
Maybe this issue is already fixed.

mrfind, 2022-03-16:
I'm still seeing the "Invalid color mode received" warnings.

I'm on the latest ha version 2022.3.5 and z2m version 1.24.0
I'm using different types of Philips Hue Color lights and it happens for all of them, both for individual lights and groups.
It only happens when setting hs colors or when using "hue_step", not when setting rgb or xy colors or color_temp. In other words: It happens whenever the lights report color_mode: "hs".
This is the log when sending a command to a single light (which is part of 5 groups):

```
  debug Received MQTT message on 'zigbee2mqtt/light/ofc/desk/rght/set' with data '{"color":{"h":100,"s":80}}'
  debug Skipping state because of Home Assistant
  debug Publishing 'set' 'color' to 'light/ofc/desk/rght'
  info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/all', payload '{"brightness":79,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"state":"ON"}'
  info MQTT publish: topic 'zigbee2mqtt/light/ofc/desk/rght', payload '{"brightness":79,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"color_temp_startup":250,"linkquality":131,"state":"ON","update":{"state":"idle"},"update_available":false}'
  info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/all', payload '{"brightness":254,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"state":"ON"}'
  info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/brm', payload '{"brightness":254,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"state":"ON"}'
  info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/cor', payload '{"brightness":254,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"state":"ON"}'
  info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/desk', payload '{"brightness":79,"color":{"h":100,"hue":100,"s":80,"saturation":80,"x":0.3151,"y":0.514},"color_mode":"hs","color_temp":170,"state":"ON"}'
```

Interestingly there are 6 mqtt messages reporting the new state and I get 12 warnings in the ha log - 2 per message?

Setting the color_temp for those lights doesn't change anything.
I added "color_mode" to the filtered_attributes for this light. Result: Only 10 instead of 12 warnings (for the 5 groups?) but a new ERROR in ha log:

```
  2022-03-16 01:40:22 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/light/ofc/desk/rght': '{"brightness":79,"color":{"h":200,"hue":200,"s":80,"saturation":80,"x":0.1999,"y":0.2912},"color_temp":153,"color_temp_startup":250,"linkquality":134,"state":"ON","update":{"state":"idle"},"update_available":false}'
  Traceback (most recent call last):
    File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 47, in wrapper
      msg_callback(msg)
    File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 327, in state_received
      self._update_color(values)
    File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 265, in _update_color
      color_mode = values["color_mode"]
  KeyError: 'color_mode'
```

So that's no solution. Also, I'd have to set the filtered attribute for the groups as well and I don't know how to do that...

I really hope you find a solution for this; I'm working on animated light scenes which need the h/s option, so I get loads of these warnings and after a few thousand of them ha keeps restarting.
Thanks a lot for your work!
Martin

krasatos, 2022-04-10:
Same here, 
Hassio 2022.4.1
z2m 1.25.0-1 
[ZNLDP12LM](https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html#xiaomi-znldp12lm)
```
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:267
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 4:06:54 AM (25 occurrences)
Last logged: 2:30:32 PM

Invalid color mode received
```

ThatTallGuy21, 2022-04-13:
Hey guys-- I was receiving this same error and *I think* I may have resolved it. Still trying to test it out, but so far it has worked, even after restarting HA. I noticed that my color and ambient bulbs (Hue brand in my case) that were throwing this error were never actually set via color or temperature from my HA dashboard. I have numerous automations to turn lights on/off, or switches that would change the state of these different types of bulbs. However, once I went into the dashboard, and changed both color and temperature (when applicable), the warnings/errors disappeared from HA and Zigbee2Mqtt logs.

If this actually does work, my guess is that the default initial value in HA isn't being set by Zigbee2Mqtt so while it's passing over the color/temp value and working correctly in HA, it's not stored in and coming from HA? Idk.. Kind of grasping at straws but it seems to have worked. 

Note: This is best tested by grouping all your lights in Z2M and adding the group light with temp/color to the dashboard and adjusting it until you're happy with the setting.

mrfind, 2022-04-15:
Thanks for sharing that, [ThatTallGuy21](https://github.com/ThatTallGuy21), but sadly it doesn't apply to my situation. I have changed color and/or temperature of my lights loads of times via the dashboard, but the problem persists.
I'm now on ha 2022.4.4 and z2m 1.25.0 and nothing has changed.

@Koenkk Do you have any ideas what else I might try to get to the bottom of this?

Koenkk, 2022-04-15:
@mrfind could you provide the debug log when setting the color temperature and this error pops-up in HA?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

mrfind, 2022-04-15:
@Koenkk there's no error when setting the color temp, only when setting color in hs mode (see my post from 16 March).

Here's the log when I set hs color for a light that's a member of multiple groups - the ha log shows multiple _Invalid color mode received_ warnings:
```
debug Received MQTT message on 'zigbee2mqtt/light/ofc/desk/rght/set' with data '{"state":"on","color":{"hue":139,"saturation":99}}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color' to 'light/ofc/desk/rght'
debug Requested color_temp 103 is lower than minimum supported 153, using minimum!
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/all', payload '{"brightness":254,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/light/ofc/desk/rght', payload '{"brightness":99,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"color_temp_startup":250,"linkquality":105,"state":"ON","update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/all', payload '{"brightness":254,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/brm', payload '{"brightness":254,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/cor', payload '{"brightness":254,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/desk', payload '{"brightness":99,"color":{"h":139,"hue":139,"s":99,"saturation":99,"x":0.1652,"y":0.5318},"color_mode":"hs","color_temp":153,"state":"ON"}'
```

...and here's the log when I try the same with rgb color - the ha log shows NO warnings:
```
debug Received MQTT message on 'zigbee2mqtt/light/ofc/desk/rght/set' with data '{"state":"on","color":{"rgb":"0,255,0"}}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color' to 'light/ofc/desk/rght'
debug Requested color_temp 129 is lower than minimum supported 153, using minimum!
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/all', payload '{"brightness":254,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/light/ofc/desk/rght', payload '{"brightness":99,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"color_temp_startup":250,"linkquality":94,"state":"ON","update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/all', payload '{"brightness":254,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/brm', payload '{"brightness":254,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/cor', payload '{"brightness":254,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/desk', payload '{"brightness":99,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"xy","color_temp":153,"state":"ON"}'
```

mrfind, 2022-04-15:
As per your request, here's also the log for changing the color temp, which does **not** result in warnings:
```
debug Received MQTT message on 'zigbee2mqtt/light/ofc/desk/rght/set' with data '{"state":"on","color_temp":300}'
debug Skipping state because of Home Assistant
debug Publishing 'set' 'color_temp' to 'light/ofc/desk/rght'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/all', payload '{"brightness":254,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/light/ofc/desk/rght', payload '{"brightness":99,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"color_temp_startup":250,"linkquality":72,"state":"ON","update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/all', payload '{"brightness":254,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/brm', payload '{"brightness":254,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/hom/not/cor', payload '{"brightness":254,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"state":"ON"}'
info MQTT publish: topic 'zigbee2mqtt/lightgroup/ofc/desk', payload '{"brightness":99,"color":{"h":36,"hue":36,"s":67,"saturation":67,"x":0.4152,"y":0.3954},"color_mode":"color_temp","color_temp":300,"state":"ON"}'
```
Thanks for your support!

Koenkk, 2022-04-17:
This is because `hs` is not marked as a supported color mode in the HA discovery for the reason below:

@emontnemery I've tested with `supported_color_modes: [ 'xy', 'hs', 'color_temp' ]` but after this HA only sends the HS color when changing the color via the picker wheel. We actually want this to be XY for default (according to the Zigbee spec HS is optional and XY mandatory). Could the first color mode be the preferred one? e.g.:
- `supported_color_modes: [ 'xy', 'hs', 'color_temp' ]` -> XY preferred
- `supported_color_modes: [ 'hs', 'xy', 'color_temp' ]` -> SH preferred

mrfind, 2022-04-17:
Not sure if this is relevant, but just in case: Note that the problem also occurs when changing colors from node-red by writing to MQTT (Mosquitto) directly.

Happy Easter! :-)

Koenkk, 2022-04-18:
@mrfind it will only happen if you control in HS, if you send XY values from node-red you should be fine.

closed 2022-05-26 by github-actions[bot]

kirush0280, 2022-11-23:
```
2022-11-23 11:03:09.301 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.svet_koridor_2
2022-11-23 11:03:09.362 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.svet_koridor_2
2022-11-23 11:03:51.213 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.svet_koridor_1
2022-11-23 11:03:51.378 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.svet_koridor_2
```
Model Zigbee
lumi.light.aqcn02
Производитель Zigbee: LUMI
Aqara smart LED bulb
Version Zigbee2MQTT: [1.28.2]

HA:
Home Assistant 2022.11.4
Supervisor 2022.10.2
Operating System 9.3
Frontend 20221108.0 - latest

phoxy84, 2022-12-06:
I solved my issue by going into the bulb info in Home assistant and set the color temperature once there manually. On the bulbs with the error message the slider was all the way to the left. Now no errors (for now)
[image]

sygys, 2022-12-06:
Haven't been having these errors for months now. Just change the color modes once indeed and it will be fixed

robobobo00, 2023-02-02:
> Haven't been having these errors for months now. Just change the color modes once indeed and it will be fixed

hello, and how i can do that ?

popy2k14, 2023-04-05:
Same issue here for two hue bulbsa which just have color temp and no color.
Log:

```
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:368
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 3. April 2023 um 20:40:14 (258 occurrences)
Last logged: 20:23:08

Invalid color mode received for entity light.kuche_dunstabzug_links
Invalid color mode received for entity light.kuche_dunstabzug_rechts
```

Have now set an color_temp in HA manually (as stated above). Hope this fixes the log entries.

popy2k14, 2023-04-06:
After restarting HA and Z2M issue is back. So no solution.

@Koenkk How can we help to debug this issue.  It seems that Z2M sends xy and HA just want's color_temp.

Here are others with the same issue: https://community.home-assistant.io/t/mqtt-invalid-color-mode-received/314381/8

Koenkk, 2023-04-06:
- Does LTW012 support color XY?
- Can you provide the debug log when color mode changes from a valid one (color temp) to an invalid one (xy)?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

popy2k14, 2023-04-08:
How can i check if it supports XY? I think not, because it just have color_temp values to set.

Here is the log part where i set an color temp, and the error pops up in the HA logs:

```
Logger: homeassistant.components.mqtt.light.schema_json
Source: components/mqtt/light/schema_json.py:377
Integration: MQTT (documentation, issues)
First occurred: 17:04:24 (6 occurrences)
Last logged: 17:04:49

Invalid color mode 'xy' received for entity light.kuche_dunstabzug_links
Invalid color mode 'xy' received for entity light.kuche_dunstabzug_rechts
```

Z2M Debug Logs:
```
Zigbee2MQTT:debug 2023-04-08 17:04:49: Received MQTT message on 'zigbee2mqtt/Küche Dunstabzug rechts/set' with data '{"state":"ON","color_temp":183}'
Zigbee2MQTT:debug 2023-04-08 17:04:49: Publishing 'set' 'state' to 'Küche Dunstabzug rechts'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.1355,"y":0.0405},"color_mode":"xy","color_temp":370,"last_seen":"2023-04-08T17:04:49+02:00","linkquality":98,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2023-04-08 17:04:49: Publishing 'set' 'color_temp' to 'Küche Dunstabzug rechts'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.1355,"y":0.0405},"color_mode":"xy","color_temp":370,"last_seen":"2023-04-08T17:04:49+02:00","linkquality":98,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Alle', payload '{"brightness":126,"color":{"x":0.3334,"y":0.3418},"color_mode":"color_temp","color_temp":183,"state":"ON"}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.3334,"y":0.3418},"color_mode":"color_temp","color_temp":183,"last_seen":"2023-04-08T17:04:49+02:00","linkquality":98,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche', payload '{"brightness":126,"color":{"x":0.3334,"y":0.3418},"color_mode":"color_temp","color_temp":183,"state":"ON"}'
Zigbee2MQTT:debug 2023-04-08 17:04:49: Received MQTT message on 'zigbee2mqtt/Küche Dunstabzug rechts/set' with data '{"state":"ON","color_temp":318}'
Zigbee2MQTT:debug 2023-04-08 17:04:49: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2023-04-08 17:04:49: Publishing 'set' 'color_temp' to 'Küche Dunstabzug rechts'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.3334,"y":0.3418},"color_mode":"color_temp","color_temp":183,"last_seen":"2023-04-08T17:04:49+02:00","linkquality":98,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Alle', payload '{"brightness":126,"color":{"x":0.4273,"y":0.4005},"color_mode":"color_temp","color_temp":318,"state":"ON"}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.4273,"y":0.4005},"color_mode":"color_temp","color_temp":318,"last_seen":"2023-04-08T17:04:49+02:00","linkquality":98,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-04-08 17:04:49: MQTT publish: topic 'zigbee2mqtt/Küche', payload '{"brightness":126,"color":{"x":0.4273,"y":0.4005},"color_mode":"color_temp","color_temp":318,"state":"ON"}'
Zigbee2MQTT:debug 2023-04-08 17:04:50: Received MQTT message on 'zigbee2mqtt/Küche Dunstabzug rechts/set' with data '{"state":"ON","color_temp":336}'
Zigbee2MQTT:debug 2023-04-08 17:04:50: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2023-04-08 17:04:50: Publishing 'set' 'color_temp' to 'Küche Dunstabzug rechts'
Zigbee2MQTT:info  2023-04-08 17:04:50: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.4273,"y":0.4005},"color_mode":"color_temp","color_temp":318,"last_seen":"2023-04-08T17:04:50+02:00","linkquality":98,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-04-08 17:04:50: MQTT publish: topic 'zigbee2mqtt/Alle', payload '{"brightness":126,"color":{"x":0.4391,"y":0.4048},"color_mode":"color_temp","color_temp":336,"state":"ON"}'
Zigbee2MQTT:info  2023-04-08 17:04:50: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug rechts', payload '{"brightness":63,"color":{"x":0.4391,"y":0.4048},"color_mode":"color_temp","color_temp":336,"last_seen":"2023-04-08T17:04:50+02:00","linkquality":98,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
[... cut ...]
Zigbee2MQTT:info  2023-04-08 17:04:53: MQTT publish: topic 'zigbee2mqtt/Küche Dunstabzug links ', payload '{"brightness":63,"color":{"x":0.426,"y":0.4},"color_mode":"color_temp","color_temp":316,"last_seen":"2023-04-08T17:04:53+02:00","linkquality":160,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786688,"latest_version":16783874,"state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2023-04-08 17:04:53: Received MQTT message on 'zigbee2mqtt/Küche Dunstabzug rechts/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2023-04-08 17:04:53: Publishing 'set' 'state' to 'Küche Dunstabzug rechts'

```

Hope this helps.

Koenkk, 2023-04-08:
In this log it switches from xy -> color_temp, can you provide the one where it switches from color_temp -> xy?

Rjaap, 2023-04-17:
In my case the problem was solved when I manually set the attribute to the right value in the configuration.yaml of Home Assistant: color_mode: color_temp.

latinvm, 2023-04-26:
> I solved my issue by going into the bulb info in Home assistant and set the color temperature once there manually. On the bulbs with the error message the slider was all the way to the left. Now no errors (for now) [image]

This also fixed it for me, color_mode is now set to color_temp. Thanks!

maniac79, 2023-09-13:
Hi all.
I also have this issue - already tried manually setting color_mode manually for all my lights, but it didn't help.
My logs are getting spammed by this error message and very regularly different lights will come unresponsive and I suspect it has to do with this (potential) error.
Any idea what else I could try to remedy the situation?
Thx!

popy2k14, 2023-09-13:
@maniac79 can you get an log like described above?

pagdot, 2023-10-20:
I've got this issue with an Ikea LED1924G9 RGB Bulb when changing the color from a remote directly to the bulb (I think using a group) and reporting configured.
With MQTT Explorer I was able to at least temporary fix this by publishing on the homeassistant/light/<ID>/light/config endpoint a config with `supported_color_modes` extended by `hs`. When doing this I loose control in homeassistant, because it sends the color as hs now.

Looking at the log

```
debug 2023-10-20 16:35:36 Received Zigbee message from 'Wohnzimmerdeckenlampe', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454,"currentHue":0,"currentSaturation":254,"currentX":44563,"currentY":20315}' from endpoint 1 with groupID null
debug 2023-10-20 16:35:36 Requested color_temp 160 is lower than minimum supported 250, using minimum!
info  2023-10-20 16:35:36 MQTT publish: topic 'zigbee2mqtt/Wohnzimmerdeckenlampe', payload '{"brightness":215,"color":{"h":0,"hue":0,"s":100,"saturation":100,"x":0.7006,"y":0.2993},"color_mode":"hs","color_options":null,"color_temp":250,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"installed_version":268572245,"latest_version":268572245,"state":"idle"},"update_available":false}'
```

it seems like the bulb sends not only a xy, but also a hs value and zigbee2mqtt repeats both of them and sets "color_mode" to "hs" breaking homeassistant.

I guess this (at least in my case) could be fixed by making sure that color_mode is only set to supported values. I tried to find the code where this happens, but at least searching for "currentSaturation" only showed a test. Can take a look at fixing it, but need help finding the code causing the issue.

Can provide more information (logs, mqtt packets, config values, ...) if it helps fixing the issue. If this is a separate issue, I'll open a new one