Koenkk/zigbee2mqtt #7741: NAS-TH01

scargill, 2021-06-09:
I'm new to Zigbee2mqtt and not sure how to add devices

## Information about the device + link
Device is NAS-TH01 temperature and Humidity sensor but it also does light sensing in LUX - from AliExpress - the box says NEO COOLCAM.

Runs off two AA batteries 0 Zigbee 3.0 and has a buzzer.

It is neat, rectangular desk-mounting and also does light sensing.

If kind of shows up in the zigbee2mqtt pairing info showing only a light intensity value (no temperature,humidity and I think the lught value is nonsense) but then nothing shows up in MQTT. I've lots of othe devices working no problem with Zigbee2mqtt on a sonoff dongle. The diaplay works perfectly detecting all 3 items and battery state

Any help or pointers welcome, this would be popular as it is on AliExpress and cheap.

Anyone?

cybermann22, 2021-06-10:
[image]

Habe hier auch das gleiche Gerät, wird bei mir als TS0201 erkannt. Es werden

Folgende Werte werden geliefert:
[image]

Leider keine Temperatur oder Feuchtigkeit.

FreakErn, 2021-06-10:
I have the same exact device. It is recognized as TS0201. I get Temperature and Humidity but no Lux value.

WildeRNS, 2021-06-11:
You can try this converter: https://pastebin.com/zrqky3LH

scargill, 2021-06-12:
Adding that file made no difference, it turns out the device is PARTLY recognised, Mine is seen as TS0201 - I get the opposite to FreakErn - I get the correct LUX value, but no mention of temperature or humidity.

jorkki22, 2021-06-12:
I can add also that I have 3 of those devices and 2 of them show temp and humidity without LUX and the third one shows only link quality, not sure if its broken or partially recognized. 

There seems to be a related discussion here of a NAS-TH02B with debug logs https://github.com/Koenkk/zigbee2mqtt/discussions/7268

scargill, 2021-06-12:
> I can add also that I have 3 of those devices and 2 of them show temp and humidity without LUX and the third one shows only link quality, not sure if its broken or partially recognized.
> 
> There seems to be a related discussion here of a NAS-TH02B with debug logs #7268

In that thread, queations are asked about raw data but no answers. I get link quality as wewll as lux, but nothing about temperatuer or humidity, This is AleIxpress, surely SOMEONE has these running in zigbee2mqtt...

scargill, 2021-06-12:
Am I beginning to understand a little? In database.db, I have an entry "if":11 which includes "modelId":"TS0201", with info on luminance but no mention of temperature or humudity. Does this mean anything to anyone?

In my configuration.yaml, tthe device showing light from my desktop device is definitely this one:

  '0x804b50fffe047798':
    friendly_name: tuya-temperature
    retain: true

And the entry 11 in the database definitely only shows luminance - is there a way to simply add temperature and humidity variables?

There it is in the database.db file...

{"id":11,"type":"EndDevice","ieeeAddr":"0x804b50fffe047798","nwkAddr":52852,"manufId":4098,"manufName":"_TZ3000_qaaysllp","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1,1024,57346],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0201","manufacturerName":"_TZ3000_qaaysllp","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"msIlluminanceMeasurement":{"attributes":{"61441":50,"measuredValue":17405}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001cd485b8","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1623491302555}

jorkki22, 2021-06-12:
> 
> 
> You can try this converter: https://pastebin.com/zrqky3LH

I added this converter and restarted Z2M, Now it reports also LUX and illuminance, but no temp/humidity. 
There is a log entry with error

```
info  2021-06-12 14:05:27: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"9bd46932","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.19.1"}'
error 2021-06-12 14:05:49: Failed to configure '0xb4e3f9fffe0c4bbe', attempt 1 (Error: ConfigureReporting 0xb4e3f9fffe0c4bbe/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":100}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30447 - 1 - 3 - 1026 - 7 after 10000ms)
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))

```

scargill, 2021-06-12:
I tried copying an attribute for temperature into the db file - 
"msTemperatureMeasurement":{"attributes":{"measuredValue":2336}},
just before the "msIlluminanceMeasurement" attribute
but it just disappeared every time.

I've no idea how to get that to stay put, never mind do anything..... any ideas?

cybermann22, 2021-06-12:
> You can try this converter: https://pastebin.com/zrqky3LH

wie kann ich den Converter in Iobroker hinzufügen?

WildeRNS, 2021-06-12:
> > You can try this converter: https://pastebin.com/zrqky3LH
> 
> I added this converter and restarted Z2M, Now the one that earlier showed temp+humidity works perfectly, but the other one that just showed linkquality reports LUX, illuminance andbattery, but no temp/humidity.
> 
> There is a log entry with error
> 
> ```
> info  2021-06-12 14:05:27: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"9bd46932","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.19.1"}'
> error 2021-06-12 14:05:49: Failed to configure '0xb4e3f9fffe0c4bbe', attempt 1 (Error: ConfigureReporting 0xb4e3f9fffe0c4bbe/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":100}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30447 - 1 - 3 - 1026 - 7 after 10000ms)
>     at Timeout.waiter.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
>     at ontimeout (timers.js:436:11)
>     at tryOnTimeout (timers.js:300:5)
>     at listOnTimeout (timers.js:263:5)
>     at Timer.processTimers (timers.js:223:10))
> ```

I think previous two has another firmware (older).

cybermann22, 2021-06-12:
mit dem Converter bekomme ich zwar Illuminance aber keine Temperatur und keine Feuchtigkeit ;-(

[image]

scargill, 2021-06-12:
I'm wrong- I'm not even getting luminance - I'm getting a fixed value of 52, regardless of light level shown on the display.

wdesponts, 2021-06-17:
Hey everyone, I have this sensor working at 50%, hopefully we can help each other resolve the remainder. I have been able to get all the data except the Temp and Humidity values. Those are part of a new cluster type that needs to be defined...more on that later.

To get the rest working try the following:

Edit the Neo.js file to add the new device
Navigate to `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices`
Edit neo.js
Add the following below the NEW-AB02B0 Device
`{ fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_qaaysllp'}], zigbeeModel: ['0yu2xgi'], model: 'NAS-TH01', vendor: 'Neo', description: 'Temperature, humidity & illuminance display', fromZigbee: [fz.battery, fz.temperature, fz.humidity, fz.illuminance], toZigbee: [], exposes: [e.battery(), e.illuminance(), e.illuminance_lux().withUnit('lx'), e.humidity(), e.temperature()], }, },`
Remove the device if already added to your system
Restart zigbee2mqtt service
Pair the Neo SAS-TH01 device to Zigbee2Mqtt
You'll have access to the battery, illuminance and illuminance_lux data

Now the for the new cluster work, I'll need to differ to others. I was able to find the right file thanks to @Vebryn 's input but I'm at a loss as to editing that file. For that I'll need some help.

RealKodijack, 2021-06-19:
> Now the for the new cluster work, I'll need to differ to others. I was able to find the right file thanks to @Vebryn 's input but I'm at a loss as to editing that file. For that I'll need some help.

Thanks to you I at least see Illumination now. Thanks. How can I help you? What do you need? If you need it quicker:
https://t.me/kodijack

scargill, 2021-06-19:
Hi wdesponts

I found that file - forgive my thickness....
Where you say to add below the NEW_ABO2B0 device - where in the file exactly? This is what that file contains on my system. Go easy on me...

const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_d0yu2xgi'}],
        zigbeeModel: ['0yu2xgi'],
        model: 'NAS-AB02B0',
        vendor: 'Neo',
        description: 'Temperature & humidity sensor and alarm',
        fromZigbee: [fz.neo_t_h_alarm, fz.ignore_basic_report],
        toZigbee: [tz.neo_t_h_alarm],
        exposes: [
            e.temperature(), e.humidity(), exposes.binary('humidity_alarm', ea.STATE_SET, true, false), e.battery_low(),
            exposes.binary('temperature_alarm', ea.STATE_SET, true, false),
            exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.enum('melody', ea.STATE_SET, Array.from(Array(18).keys()).map((x)=>(x+1).toString())),
            exposes.numeric('duration', ea.STATE_SET).withUnit('second'),
            exposes.numeric('temperature_min', ea.STATE_SET).withUnit('Â°C'),
            exposes.numeric('temperature_max', ea.STATE_SET).withUnit('Â°C'),
            exposes.numeric('humidity_min', ea.STATE_SET).withUnit('%'),
            exposes.numeric('humidity_max', ea.STATE_SET).withUnit('%'),
            exposes.enum('volume', ea.STATE_SET, ['low', 'medium', 'high']),
            exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb']),
        ],
    },
];

Vebryn, 2021-06-19:
According to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js, neo_t_h_alarm use manuSpecificTuya cluster, this device use 0xE002 cluster. I think it cannot work

scargill, 2021-06-21:
> { fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_qaaysllp'}], zigbeeModel: ['0yu2xgi'], model: 'NAS-TH01', vendor: 'Neo', description: 'Temperature, humidity & illuminance display', [...]

Adding that directly below thw other fingerprint section simply seems to crash Zigbee2mqtt... any idea eactly where in the neo.js file it is supposed to go?

Vebryn, 2021-06-21:
Configuration can be added into a new js file, cf. https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

scargill, 2021-06-21:
> Configuration can be added into a new js file, cf. https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Thgis isn't obvious to a newby like me... the zigbee2mqtt guide says to make a new file start with (as example

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.sens'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'WSDCGQ01LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'MiJia temperature & humidity sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
module.exports = definition;

Yet what I'm seeing above doesnt start witrhg "definition" - it starts with {fingerprint

So how would one take the demo above and alter it for the model TSO201 "fingerprint"?   Ideas anyone?

scargill, 2021-06-21:
So...

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0yu2xgi'], 
    model: 'NAS-TH01',
    vendor: 'Neo',
    description: 'Temperature, humidity & illuminance display', 
    fromZigbee:  [fz.battery, fz.temperature, fz.humidity, fz.illuminance],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [ e.battery(), e.illuminance(), e.illuminance_lux().withUnit('lx'), e.humidity(), e.temperature()]; 
};

module.exports = definition;

THIS in a file - adds in the module - perfectly - BUT all it outputs is BATTERY and LINKQUALITY - no temperature, humidity or light....

Vebryn, 2021-06-21:
I'm using zigbee2mqtt container. In container's volume, create your js file :

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const e = exposes.presets;

module.exports = [
  {
    fingerprint: [
      {manufacturerName: '_TZ3000_qaaysllp'},
    ],
    zigbeeModel: ['TS0201'],
    model: 'NAS-TH01',
    vendor: 'NEO Coolcam',
    description: 'Temperature, humidity and illumination sensors with display',
    fromZigbee: [fz.battery, fz.illuminance, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.battery(), e.illuminance(), e.temperature(), e.humidity()],
  },
];
```

Then add your file into configuration.yaml :
```
external_converters:
  - external_converters.js
```

Only battery and illuminance are working.

scargill, 2021-06-21:
I have a js file called petes.js, I put in your code – and added the file link into configuration.yaml as suggested..

 

It has appeared again – but still showing only battery and linkquality  in the payload.

 

Pete

mrTuomoK, 2021-06-22:
I got the exact same device working just a moment ago with this configuration:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_qaaysllp'}],
        model: 'NAS-TH01',
        vendor: 'Neo',
        description: 'Temperature, humidity and illuminance sensor plus alarm',
        fromZigbee: [
            fz.battery,
            fz.temperature,
            fz.humidity,
            fz.illuminance
        ],
        toZigbee: [],
        exposes: [e.battery(), e.illuminance(), e.illuminance_lux(), e.temperature(), e.humidity(), e.battery_voltage()],
};

module.exports = definition;
```

I'm still to test if the alarm & buzzer can be gotten to work too...

scargill, 2021-06-22:
mrTuomoK

I just took your code, put it into an empty /opt/zigbee2mqtt/data/petes.js, stopped zigbee2mqtt service, ensured reference to that device was cleared out of /opt/zigbee2mqtt/data/configuration.yaml and restarted the service (with "permit join" enabled.

I then long-perssed the reset on the device as I've done in the past, the device appeared again as before - and as before, the payload contains nothing more than battery and linkquality. 
[image]

mrTuomoK, 2021-06-22:
@scargill That's weird, Here's my entry for the device in the database.db, how does your look like?

```
{"id":62,"type":"EndDevice","ieeeAddr":"0xb4e3f9fffe0c4b2d","nwkAddr":16784,"manufId":4098,"manufName":"_TZ3000_qaaysllp","powerSource":"Battery","modelId":"TS0201","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1,1024,57346],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005n3��\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":1,"modelId":"TS0201","manufacturerName":"_TZ3000_qaaysllp","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":18452}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4134","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":3010}},"msRelativeHumidity":{"attributes":{"measuredValue":5530}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1624342457466}
```

lunarok, 2021-06-22:
> I got the exact same device working just a moment ago with this configuration:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Thank you for your code, this is working for me.

ali1234, 2021-06-25:
I have something that looks like this too. Part of the problem is that because it identifies as TS0201, Z2M thinks it is a different device that only has temperature and humidity. So if you only see battery and link level that is probably why. It is probably reporting illuminance but Z2M ignores it.

See https://github.com/Koenkk/zigbee-herdsman-converters/issues/2718 about misidentification.

If I reset and re-pair the device, the information stored in database.db actually changes. Sometimes it is like this:

```
{
   "id":2,
   "type":"EndDevice",
   "ieeeAddr":"0xb4e3f9fffe07a4b6",
   "nwkAddr":12704,
   "manufId":4098,
   "manufName":"_TZ3000_qaaysllp",
   "powerSource":"Battery",
   "modelId":"TS0201",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":262,
         "inClusterList":[
            0,
            1,
            1024,
            57346
         ],
         "outClusterList":[
            25,
            10
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  "modelId":"TS0201",
                  "manufacturerName":"_TZ3000_qaaysllp",
                  "powerSource":3,
                  "zclVersion":3,
                  "appVersion":68,
                  "stackVersion":0,
                  "hwVersion":1,
                  "dateCode":""
               }
            },
            "msIlluminanceMeasurement":{
               "attributes":{
                  "61441":50,
                  "measuredValue":23445
               }
            },
            "genPowerCfg":{
               "attributes":{
                  "batteryPercentageRemaining":200
               }
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
[... 173 lines cut ...]
            }
         },
         "binds":[
            
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":68,
   "stackVersion":0,
   "hwVersion":1,
   "dateCode":"",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":821693351
   },
   "lastSeen":1624342457466
}
```

ali1234, 2021-06-26:
I discovered something today. Immediately after pairing the device sends this:

```
Zigbee2MQTT:debug 2021-06-26 16:20:04: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,71,10,10,208,41,40,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:07: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,72,10,11,208,41,10,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:09: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,73,10,13,208,41,90,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:11: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,74,10,14,208,41,20,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:13: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,75,10, 6,208,48,2],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:15: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,76,10,15,208,48,2],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-26 16:20:17: Received Zigbee message from '0xb4e3f9fffe07a4b6', type 'raw', cluster '57346', data '{"data":[8,77,10,16,208,32,0],"type":"Buffer"}' from endpoint 1 with groupID 0
```

looking at the last two digits of the first four messages we see this:

```
40,0
10,0
90,0
20,0
```

I had a hunch these could be minimum and maximum values for temperature and humidity. By blowing on the sensor I was able to get the humidity reading to go above 90% and then the device immediately produced this message:

```
type 'raw', cluster '57346', data '{"data":[8,16,10,15,208,48,1],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Then shortly after, when humidity went back below 90%, it produced this message:

```
type 'raw', cluster '57346', data '{"data":[8,29,10,15,208,48,2],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Next I put it on a heater and when temperature went above 40 it immediately sent this:

```
type 'raw', cluster '57346', data '{"data":[8,97,10,6,208,48,1],"type":"Buffer"}' from endpoint 1 with groupID 0
```

With further heating I was able to get humidity down to 19% and got this:

```
type 'raw', cluster '57346', data '{"data":[8,42,10,15,208,48,0],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Then this when it went back up to 22%:

```
type 'raw', cluster '57346', data '{"data":[8,52,10,15,208,48,2],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Then this when temperature dropped to 39 degrees:

```
type 'raw', cluster '57346', data '{"data":[8,65,10,6,208,48,2],"type":"Buffer"}' from endpoint 1 with groupID 0
```

It also seems like the second value is a message sequence number.

TrurlMcByte, 2021-07-13:
Currently partially working after connect to Tuya and reconnect back with next converter
```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const { calibrateAndPrecisionRoundOptions: calibrate } = require("zigbee-herdsman-converters/lib/utils");
const _ = require("lodash");

const e = exposes.presets;
const ea = exposes.access;

const msIlluminance = {
  cluster: "msIlluminanceMeasurement",
  type: "attributeReport",
  convert: (model, msg, publish, options, meta) => {
    const il = msg.data["measuredValue"];
    const ilLux = il === 0 ? 0 : Math.pow(10, (il - 1) / 10000);
    const payload = {
      illuminance: calibrate(il, options, "illuminance"),
      illuminance_lux: calibrate(ilLux, options, "illuminance_lux"),
    };
    const clusters = _.get(msg, ["device", "_endpoints", "1", "clusters"], null);
    if (clusters !== null) {
      const humidity = _.get(clusters, "msRelativeHumidity.attributes.measuredValue", null);
      if (humidity !== null) payload.humidity = calibrate(parseFloat(humidity) / 100.0, options, "humidity");
      const temperature = _.get(clusters, "msTemperatureMeasurement.attributes.measuredValue", null);
      if (temperature !== null)
        payload.temperature = calibrate(parseFloat(temperature) / 100.0, options, "temperature");
    }
    return payload;
  },
};

module.exports = [
  {
    fingerprint: [{ modelID: "TS0201", manufacturerName: "_TZ3000_qaaysllp" }],
    zigbeeModel: ["TS0201"],
    model: "NAS-TH01",
    vendor: "NEO Coolcam",
    description: "Temperature, humidity and illumination sensors with display",
    exposes: [e.battery(), e.illuminance(), e.illuminance_lux(), e.humidity(), e.temperature(), e.battery_low()],
    fromZigbee: [msIlluminance, fz.battery, fz.humidity],
    toZigbee: [],
    meta: { multiEndpoint: true },
  },
];

```
returns  humidity, temperature, illuminance

NODeeJay, 2021-07-14:
I tried your suggestion but I struggle with lodash. 

> ```js
> const _ = require("lodash");
> ```

_Error: Cannot find module 'lodash'_

I tried installing it through command line but HASS OS does not seem to have npm. Also google was not helpful how to add lodash to home assistant.

Any suggestions are very welcome!

TrurlMcByte, 2021-07-15:
lodash included in zigbee2mqtt, also in iobroker, but in code it's used just for simplify getting deep values like ```clusters.msTemperatureMeasurement.attributes.measuredValue``` with checking every level and default value.

KardinalReusen, 2021-07-15:
I get the same error like NODeeJay in my zigbee2mqtt docker container installation.

NODeeJay, 2021-07-15:
Maybe there's the issue?

- core-2021.7.2
- Home Assistant OS 6.1
- supervisor-2021.06.8

running on:
_Oracle VirtualMachine_ (latest) on _Windows 10 Pro 20H2_

_Zigbee2mqtt (1.20.0-1)_ with _TI CC2531 USB_ (Windows COM port to VM)

When I comment out the section, zigbee2mqtt starts fine:
```
/*
const _ = require("lodash");
*/
```

otherwise I get the following error:
```
  throw err;
  ^
Error: Cannot find module 'lodash'
Require stack:
- /app/lib/util/utils.js
- /app/lib/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at /app/lib/util/externally-loaded.js:8:11
    at Script.runInContext (vm.js:143:18)
    at Script.runInNewContext (vm.js:148:17)
    at Object.runInNewContext (vm.js:303:38)
    at loadModuleFromText (/app/lib/util/utils.js:162:8)
    at loadModuleFromFile (/app/lib/util/utils.js:168:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:366) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:366) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[15:08:52] INFO: Handing over control to Zigbee2mqtt Core ...
```

FragMenthor, 2021-07-15:
> Maybe there's the issue?
>
> * core-2021.7.2
> * Home Assistant OS 6.1
> * supervisor-2021.06.8
>
> running on:
> _Oracle VirtualMachine_ (latest) on _Windows 10 Pro 20H2_
>
> _Zigbee2mqtt (1.20.0-1)_ with _TI [...]

Same here, and exactly the same configuration!

TrurlMcByte, 2021-07-17:
same without lodash
```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
// const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
// const reporting = require("zigbee-herdsman-converters/lib/reporting");
// const extend = require("zigbee-herdsman-converters/lib/extend");
// const tuya = require("zigbee-herdsman-converters/lib/tuya");
const {
  calibrateAndPrecisionRoundOptions: calibrate,
  getObjectProperty: gOP,
} = require("zigbee-herdsman-converters/lib/utils");

const e = exposes.presets;
const ea = exposes.access;

const getP = (o, p, d = undefined) => {
  const path = Array.isArray(p) ? p : p.toString().split("."),
    l = path.length;
  let i = 0;
  while (o !== null && i < l) o = gOP(o, path[i++], null);
  return i && i == l ? o : d;
};

const msIlluminance = {
  cluster: "msIlluminanceMeasurement",
  type: "attributeReport",
  convert: (model, msg, publish, options, meta) => {
    const il = msg.data["measuredValue"];
    const ilLux = il === 0 ? 0 : Math.pow(10, (il - 1) / 10000);
    const payload = {
      illuminance: calibrate(il, options, "illuminance"),
      illuminance_lux: calibrate(ilLux, options, "illuminance_lux"),
    };
    const clusters = getP(msg, ["device", "_endpoints", "1", "clusters"], null);
    if (clusters !== null) {
      const humidity = getP(clusters, "msRelativeHumidity.attributes.measuredValue", null);
      if (humidity !== null) payload.humidity = calibrate(parseFloat(humidity) / 100.0, options, "humidity");
      const temperature = getP(clusters, "msTemperatureMeasurement.attributes.measuredValue", null);
      if (temperature !== null)
        payload.temperature = calibrate(parseFloat(temperature) / 100.0, options, "temperature");
    }
    return payload;
  },
};

module.exports = [
  {
    fingerprint: [{ modelID: "TS0201", manufacturerName: "_TZ3000_qaaysllp" }],
    zigbeeModel: ["TS0201"],
    model: "NAS-TH01",
    vendor: "NEO Coolcam",
    description: "Temperature, humidity and illumination sensors with display",
    exposes: [e.battery(), e.illuminance(), e.illuminance_lux(), e.humidity(), e.temperature(), e.battery_low()],
    fromZigbee: [msIlluminance, fz.battery, fz.humidity],
    toZigbee: [],
    meta: { multiEndpoint: true },
  },
];
```

FragMenthor, 2021-07-17:
Thank you, after deleting from Z2M, repairing with Tuya, and adding again in Z2M, it started working, only missing battery value!
[image]

FragMenthor, 2021-07-19:
Well, I restarted all HA and suddenly the conflict between sensors, with wrong identification, is gone. Although this sensor is working, something is wrong. I have a chart with stats (apex) with several sensors, and when I have this one in there, it almost completely freezes my Lovelace page. Can't understand why, because individual history of the sensors seems ok... Just to let know. Thanks again!

FragMenthor, 2021-07-22:
Just to give one more heads up: 2 days later, it stopped updating temperature and humidity, although lux value is still updating regularly

scargill, 2021-07-22:
I started this thread back on June 9 and so much has gone in here I've completely lost the plot of where we are (if anywhere).

Does anyone have a working Zigbee2MQTT solution for the rectangular AliExpress light/temperature/humidity sensor known on the box as NEO COOLCAM?

ali1234, 2021-07-22:
> Does anyone have a working Zigbee2MQTT solution for the rectangular AliExpress light/temperature/humidity sensor known on the box as NEO COOLCAM?

No, there is still no quirk for this sensor. It will only fully work if it has been previously connected to a Tuya hub running Tuya software, and then only for a couple of days. We need to sniff the Tuya traffic (this has been done) and then reprogram Z2M to send the same instructions (not done yet).

TrurlMcByte, 2021-07-22:
> Just to give one more heads up: 2 days later, it stopped updating temperature and humidity, although lux value is still updating regularly

It keeps updating, but with frozen data.

FragMenthor, 2021-07-24:
> > Just to give one more heads up: 2 days later, it stopped updating temperature and humidity, although lux value is still updating regularly
> 
> It keeps updating, but with frozen data.

Yes, you are right, that's it!

scargill, 2021-07-24:
Not the best purchase then..  Oh, well....

FragMenthor, 2021-08-14:
Will this be abandoned?

marciogranzotto, 2021-08-16:
I'm doing some further investigation on https://github.com/zigpy/zha-device-handlers/issues/862#issuecomment-897723664
but I still haven't managed to make it report temperature and humidity without first setting it up with the tuya hub

FragMenthor, 2021-08-17:
It seems it's working in DeconZ: https://zigbee.blakadder.com/Neo_NAS-TH02B.html

jmceara, 2021-08-18:
This worked perfectly for me! Should be added to main code, im my opinion.
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        fingerprint: [{modelID: 'TS0201', manufacturerName: '_TZ3000_qaaysllp'}],
        model: 'NAS-TH01',
        vendor: 'Neo',
        description: 'Temperature, humidity and illuminance sensor plus alarm',
        fromZigbee: [
            fz.battery,
            fz.temperature,
            fz.humidity,
            fz.illuminance
        ],
        toZigbee: [],
        exposes: [e.battery(), e.illuminance(), e.illuminance_lux(), e.temperature(), e.humidity(), e.battery_voltage()],
};

module.exports = definition;
```

scargill, 2021-08-18:
For the benefit of the slower amongst us - where do we put that code - and is anything else needed. I have a ton of devices on my Zigbee network so I know how to add devices - just no idea what to do here as nothing I've tried up to now would make this device work... and did you previously have the device hooked into a Tuya network? I've not done any of that.

RealKodijack, 2021-08-18:
> This worked perfectly for me! Should be added to main code, im my opinion.

Won't do anything for me, sadly. Still just illumination.
(iobroker/zigbee adapter)

FragMenthor, 2021-08-18:
That code works for 1-2 days, and suddenly stops working. When I bought the device, illuminance was not working in Tuya. I complained and they asked Tuya to force an update on the hub, needed my tuya hub device id. And started working. So, something, not quite ususal, is going on here, different from other devices.

TrurlMcByte, 2021-08-19:
> That code works for 1-2 days, and suddenly stops working.
It's not stopping at all, just for 1-2 days. And, usually, temperature only (humidity continue changing). In my case temperature have no changes 25% of time over week. Humidity "stuck" ~5% of time, illuminance working always.

mrTuomoK, 2021-08-19:
> > That code works for 1-2 days, and suddenly stops working.
> 
> 
> 
> It's not stopping at all, just for 1-2 days. And, usually, temperature only (humidity continue changing). In my case temperature have no changes 25% of time over week. Humidity "stuck" ~5% of time, illuminance working always.
> 
> 
> 
> 

For me it has been working for a month at least but the temperature and humidity reports changes of only like one degrees, illuminance works just fine.

scargill, 2021-08-19:
I can see it coming now.... BIN. A shame as it is a very nice-looking and inexpensive unit - but life is too short.

marciogranzotto, 2021-08-19:
> For me it has been working for a month at least but the temperature and humidity reports changes of only like one degrees, illuminance works just fine.

did you first pair it to a tuya hub?
With ZHA I found that if I pair it to my Zemismart hub, then remove it from the app and pair it with my CC2531 or CC2635R, everything works indefinitely. But if I remove the batteries and install it again, temperature and humidity stop working until I do the whole procedure again.

RealKodijack, 2021-08-19:
> I can see it coming now.... BIN. A shame as it is a very nice-looking and inexpensive unit - but life is too short.

And here I am, sitting with a whole bunch of them, thinking it was a good purchase :-)

scargill, 2021-08-19:
I know I'm going to get shot down for this but... isn't there some issue trying to run 2 hubs at once? I'm wary about turning on one of my hubs (I have 3 or 4 different types) for fear of upsetting my ZZH setup with dozens of devices on it. Can anyone confirm or pass comment?

scargill, 2021-08-19:
> > I can see it coming now.... BIN. A shame as it is a very nice-looking and inexpensive unit - but life is too short.
> 
> And here I am, sitting with a whole bunch of them, thinking it was a good purchase :-)

Well I thought it was a good purchase too. Overall I like Zigbee but I've had some duffers - like lights that default to power off - or power on instead of the last state they were in (WiFi devices generally remember their state)

RealKodijack, 2021-08-19:
> I know I'm going to get shot down for this but... isn't there some issue trying to run 2 hubs at once? I'm wary about turning on 

That shouldn't be a problem. Unless you don't want to pair the same device to more than one hub.

>power on (after a power falure) instead of the last state they were in (WiFi devices generally remember their state)

I don't know about your system, but if your device does not support it, why not do it yourself? System resumes from power failure => Set up your stuff the way YOU want it to default. If it's stupid but it works, it ain't stupid :-)

Westcott1, 2021-08-20:
It looks like some progress might be being made -
[https://github.com/zigpy/zha-device-handlers/issues/862](url)
This is a popular device, and I can't see it going unsupported for too long.
I tried one of my 3 with the Sonoff ZBbridge + Tasmota, and that also only gives illumination, no temp nor humidity.
Perhaps hold off binning for a while, if nothing else their display is accurate.

[44 more replies]

closed 2022-02-26 by github-actions[bot]