Koenkk/zigbee2mqtt #7745: Aquara Smart Plug EU - All sensors are not exposed (null value)

cthybert, 2021-06-10:
## What happened
Zigbee2mqtt doesn't show all sensors value for the adapter [Aqara EU smart plug](https://www.zigbee2mqtt.io/devices/SP-EUC01.html)

I'm getting Null value for energy / temp / voltage / current
I only see instant power in W.

[screenshot](http://picc.io/p/Zqb6Crt.png)

## What did you expect to happen

To see those exposed sensors value.

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: Conbee II
Adapter firmware build date: 09-10-2019

Adapter state : 
```
{
    "last_seen": "2021-06-10T15:40:34+02:00",
    "linkquality": 199,
    "power": 1.6,
    "state": "ON",
    "current": null,
    "device": {
        "applicationVersion": 22,
        "dateCode": "09-10-2019",
        "friendlyName": "prise_cave",
        "hardwareVersion": 1,
        "ieeeAddr": "0x04cf8cdf3c79dc55",
        "manufacturerID": 4447,
        "manufacturerName": "LUMI",
        "model": "SP-EUC01",
        "networkAddress": 39404,
        "powerSource": "Mains (single phase)",
        "stackVersion": 2,
        "type": "Router",
        "zclVersion": 3
    },
    "energy": null,
    "temperature": null,
    "voltage": null
}
```

crathyd, 2021-06-16:
It’s same as my situation

cthybert, 2021-06-16:
I don't know, I did the OTA update (through z2m) thinking that it would solve the problem ... and no changes. But maybe the previous firmware was already a newer than z2m was supporting ...

cthybert, 2021-06-18:
anything on the following ?

ThueAntonsen, 2021-07-08:
Same issue here, mine came with the "broken" firmware, but also had a new OTA available for download

ratsoid, 2021-07-23:
In my case, all my 4 outlets have ceased to show the W usage after doing an update to my Conbee II firmware and after I upgraded Zigbee2mqtt to 1.2.0 from 1.9.1

fluffymadness, 2021-08-05:
I have an Electrolama zig-a-zig-ah! (zzh!) stick.
After updating to the newest zigbee2mqtt it also shows the same behavior on my two smartplugs.
Seems to be some kind of regression I think.

rodnheli, 2021-08-09:
Same here - have a brand new Aqara EU smart plug for testing. Came with Firmware 09-22-2020 - only shows watts. Using Conbee II with current deconz firmware.
I'll buy again a Xiaomi ZNCZ04LM - worked out of the box with Firmware 09-06-2019...

coom, 2021-08-10:
Hi,

Same situation here except I've even lost the only things that was working : Watts. Lost it when I tried to remove and repair the device with the latest conbee II firmware.

Here is my logs if it can help :

`Debug Received Zigbee message from '0x54ef44100013a0a6', type 'readResponse', cluster 'genBasic', data '{"appVersion":32}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'readResponse', cluster 'genBasic', data '{"dateCode":"09-22-2020"}' from endpoint 1 with groupID null
Info Successfully interviewed '0x54ef44100013a0a6', device has successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara EU smart plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"SP-EUC01","supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef44100013a0a6","ieee_address":"0x54ef44100013a0a6","status":"successful","supported":true},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara EU smart plug","friendly_name":"0x54ef44100013a0a6","model":"SP-EUC01","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
Info MQTT publish: topic 'homeassistant/switch/0x54ef44100013a0a6/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x54ef44100013a0a6/set","device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0"},"json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0"},"device_class":"power","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6 power","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0"},"device_class":"energy","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_topic":"zigbee2mqtt/0x54ef44100013a0a6","last_reset_value_template":"1970-01-01T00:00:00+00:00","name":"0x54ef44100013a0a6 energy","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/0x54ef44100013a0a6","name":"0x54ef44100013a0a6 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x54ef44100013a0a6","unique_id":"0x54ef44100013a0a6_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Info MQTT publish: topic 'homeassistant/sensor/0x54ef44100013a0a6/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x54ef44100013a0a6"],"manufacturer":"Xiaomi","model":"Aqara EU smart plug (SP-EUC01)","name":"0x54ef44100013a0a6","sw_version":"Zigbee2MQTT 1.21.0"},"device_class":"voltage","enabled_by_default":false,"json_attributes_topic":"zigbee2 [... cut ...]
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":47.63600158691406}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":43.66600036621094}' from endpoint 21 with groupID null
Debug Received Zigbee message from '0x54ef44100013a0a6', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":47.63600158691406}' from endpoint 21 with groupID null
Successfully configured '0x54ef44100013a0a6'
`

Galorhallen, 2021-08-21:
Same problem here with Aquara Smart Plug EU

Koenkk, 2021-08-22:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/6861 (already fixed in the dev branch).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-22 by Koenkk

olylak, 2022-01-10:
> Same issue here, mine came with the "broken" firmware, but also had a new OTA available for download

same for me but with DLKZMK11LM

MrLivelong, 2025-01-20:
I experience the same problem and get an error when reconfiguring the device. The device interview is done correctly.
The strange thing is: all my 10 aqara plugs do have the same firmware, although 2 of them do still report energy and power consumption. The others give te following error:

z2m: Failed to configure 'device_name', attempt 4 (Error: Bind 0x54ef4410003a6cde/1 genOnOff from '0x00124b0014d9a97f/0' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:565:23) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:9) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/lumi.ts:2065:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))