Koenkk/zigbee2mqtt #7746: Issue with using 'tz.tuya_led_control' converter for TS0504A and TS0505A LED controllers

DannyTheGhost, 2021-06-10:
After successfully pairing and configuring in z2m via external converter (due to different ```manufacturerName```) 2 different controllers (TS0504A and TS0505A) I noticed problem with controlling those. When I try to change brightness when color_temp mode is on, it would always switch to RGB mode. Other functionality works fine.
When I was trying to discover the cause of this problem, I was using Zigbee sniffer. This showed me that any attempt to change brightness will always send packet which sets ```tuyaRgbMode``` to 1 and ```ZCL:MoveToHueAndSaturation```  after that
Then i tried to copy and tweak ```tz.tuya_led_control``` converter, when I noticed (by commenting and uncommenting different lines of code) that  
```javascript
if (meta.state.tuyaMode === 0 &amp;&amp; payload.brightness) {  
    await entity.command('genLevelCtrl', 'moveToLevel', {transtime: 0, level: payload.brightness},    
                 {disableResponse: true, disableDefaultResponse: true});    
    await entity.command('lightingColorCtrl', 'tuyaRgbMode', {enable: 0}, {}, {disableDefaultResponse: true});      
    return {state: {brightness: payload.brightness}}; 
}
```
part of code is not working. More specifically, ```meta.state.tuyaMode``` is not passing needed value to run needed code to only change brightness. Is there any workaround or fix for this issue?

iglu-sebastian, 2021-06-11:
Not sure if https://github.com/Koenkk/zigbee-herdsman-converters/pull/2674 helps here

DannyTheGhost, 2021-06-11:
1) LED controller by Tuya Smart app is shown as LED bulb, so I can makke assumption that Zigbee profile for bulbs and controllers are the same (even the model name is the same)
2) It seems like your implementation of ```tuya_led_control``` works perfectly fine, even though i get 
 ```Failed to call 'Receive' 'onZigbeeEvent' (ReferenceError: mapNumberRange is not defined at Object.convert (/app/lib/util/externally-loaded.js:143:17) at /app/lib/extension/receive.js:159:41 at Array.forEach (<anonymous>) at Receive.onZigbeeEvent (/app/lib/extension/receive.js:158:20) at Controller.callExtensionMethod (/app/lib/controller.js:386:44)) ```
 message (I implemented this converters like custom for debug purposes, so I can assume this error only relates to my own implementation, not your code)
 Is there any 'guides' which I can use? I'd like to better understand how z2m works so I could make my own converters more easily.