Koenkk/zigbee2mqtt #7747: Add support for the TS0601 "smart air box" TVOC detector

RubenKelevra, 2021-06-10:
I bought the TuYa compatible device which is called "Smart air box - Environmental Detector".

I've read the new device page, but that's over my head. What do I have to provide you to add support for them?

Via Zigbee it identifies itself as (yet another) TS0601:

`Zigbee2MQTT:warn  2021-06-10 17:06:21: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_8ygsuhe1'`

[image]

jsiemek, 2021-06-10:
Look here: https://github.com/Koenkk/zigbee2mqtt/discussions/7299

It work for me!

Koenkk, 2021-06-10:
Already supported in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-10 by Koenkk

RubenKelevra, 2021-06-11:
Ah! Wonderful, I couldn't find the device via the GitHub search. That's why I opened the ticket :)

RubenKelevra, 2021-06-11:
@Koenkk wrote:

> Already supported in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Sorry that's not an option since I only have one installation which needs to be rock solid :)