Koenkk/zigbee2mqtt #7749: Replace value_template with position_template for cover

m4skime, 2021-06-10:
# Bug Report
## What happened

my curtain motor (zigbee, Lonsonho QS-Zigbee-C01) has stopped working with the Home Assistant 2021.6.3 upgrade with:

```Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt' (log below)```

i reverted to 2021.5.4 (on virtual machine with snapshoot) but can easily upgrade to do tests

I've posted a bug on Home Assistant github but it mentionned as it's a zigbee2mqtt bug :

```It's an issue in zigbee2mqtt, it should use position_template instead ofvalue_template to extract the position```

https://github.com/home-assistant/core/issues/51679#issuecomment-858928515

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: conbee2
Adapter firmware version:

emontnemery, 2021-06-10:
For reference, this is due to: https://github.com/home-assistant/core/pull/50263
It's mentioned in 2021.6 release notes: https://www.home-assistant.io/blog/2021/06/02/release-20216/#breaking-changes

aidbish, 2021-06-11:
have you updated zigbee2mqtt as i believe it was resolved in a recent fix
https://github.com/Koenkk/zigbee2mqtt/issues/7680
see last message from kkoenk in that thread

Koenkk, 2021-06-11:
Already fixed in 1.19.1 indeed.

closed 2021-06-11 by Koenkk