Koenkk/zigbee2mqtt #7751: Philips LightStrips (915005106701) now cause a 'Failed to setup reporting' on Z2M starrtup

TheStigh, 2021-06-11:
Hi,

Currently running release 1.19.1 with a CC1352-P2.

After updating to 1.18.x (I think) suddenly my [Philips LightStrips](https://www.zigbee2mqtt.io/devices/915005106701.html#philips-915005106701) with firmware 1.50.2_r30933 start throwing errors during Z2M startup. After startup, they seem to work correctly. I have 8 of these installed, all of them behave the same at startup.

Are there any changes I've missed?

```
Zigbee2MQTT:error 03-06-2021 21:23:20: Failed to setup reporting for '0x001788010490ac54' - Error: ConfigureReporting 0x001788010490ac54/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 38964 - 11 - 8 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
```

Koenkk, 2021-06-11:
Do you have `report: true` set in your `configuration.yaml`? If yes please migrate away from this feature (should also solve this error). See second point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.17.1

TheStigh, 2021-06-13:
Dough! Resolved. Thanks @Koenkk

closed 2021-06-13 by TheStigh