Koenkk/zigbee2mqtt #7759: Problem with Ikea Fyrtur covers / blinds after update Core 6.0.

Dengaman, 2021-06-11:
After update to core 6.0 the covers/blinds from Fyrtur stopped working with Google home assistant.
When investigating, the entity cover is greyed out, but all the other entities are blue like battery, link quality etc.

I can control the blinds in MQTT with the exposed - position slider but not as entity.

In the entities list the covers have the red restored sign at the end of the line...

I was able to toll back to the Core 5.3 in which all still worked, but are updating Host and Supervisor it is broken again, even with Core 5.3

Dengaman, 2021-06-11:
Logging shows:

`2021-06-11 21:43:06 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Rolgordijn (Slaapkamer)/set', 'device': {'identifiers': ['zigbee2mqtt_0x588e81fffe7b1e6c'], 'manufacturer': 'IKEA', 'model': 'FYRTUR roller blind (E1757)', 'name': 'Rolgordijn (Slaapkamer)', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Rolgordijn (Slaapkamer)', 'name': 'Rolgordijn (Slaapkamer)', 'position_topic': 'zigbee2mqtt/Rolgordijn (Slaapkamer)', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Rolgordijn (Slaapkamer)/set', 'unique_id': '0x588e81fffe7b1e6c_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options
    raise vol.Invalid(
voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__
    raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.`

Koenkk, 2021-06-12:
already fixed in 1.19.0, if you don't see this update see first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-06-12 by Koenkk