Koenkk/zigbee2mqtt #7761: 1.18.1 dont want to restart if PAN_ID is not changed, even nothing has been changed

dummys, 2021-06-12:
Hello, still have problem with slaesh.
I tried reflashing, no luck. Problem come from 18.1.1 z2m version.
Tested another slaesh, same problem.
Latest version of home assistant on the supported Odroid.
EDIT: Changing PAN_ID made it work, but then I lost all my device. Why I need to change PAN_ID if I didn't change anything in hardware/config and only upgraded HA/z2M ????
Here is the full log:

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[02:03:48] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[02:03:49] INFO: MQTT available, fetching server detail ...
[02:03:50] INFO: Previous config file found, checking backup
[02:03:50] INFO: Creating backup config in '/share/zigbee2mqtt/.configuration.yaml.bk'
[02:03:50] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[02:03:51] INFO: Zigbee Herdsman debug logging enabled
[02:03:51] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
Zigbee2MQTT:info  2021-06-12 02:03:55: Logging to console and directory: '/share/zigbee2mqtt/log/2021-06-12.02-03-54' filename: log.txt
Zigbee2MQTT:info  2021-06-12 02:03:56: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2021-06-12 02:03:56: Starting zigbee-herdsman (0.13.71)
2021-06-12T00:03:56.316Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2021-06-12T00:03:56.320Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}'
2021-06-12T00:03:56.326Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2021-06-12T00:03:56.344Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2021-06-12T00:03:56.357Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2021-06-12T00:03:56.359Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2021-06-12T00:04:07.588Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,33,9,135,0,0,1,0,171]
2021-06-12T00:04:07.596Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,0,105]
2021-06-12T00:04:07.596Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,0,105]
2021-06-12T00:04:07.596Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [0] - 105
2021-06-12T00:04:07.597Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWrite - {"status":0}
2021-06-12T00:04:07.597Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:07.599Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWrite - {"id":99,"len":1,"offset":0,"value":{"type":"Buffer","data":[0]}}
2021-06-12T00:04:07.600Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,33,9,99,0,0,1,0,79]
2021-06-12T00:04:07.607Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,0,105]
2021-06-12T00:04:07.608Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,0,105]
2021-06-12T00:04:07.608Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [0] - 105
2021-06-12T00:04:07.609Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWrite - {"status":0}
2021-06-12T00:04:07.609Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:07.611Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWrite - {"id":143,"len":1,"offset":0,"value":{"type":"Buffer","data":[1]}}
2021-06-12T00:04:07.612Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,33,9,143,0,0,1,1,162]
2021-06-12T00:04:07.620Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,0,105]
2021-06-12T00:04:07.620Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,0,105]
2021-06-12T00:04:07.621Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [0] - 105
2021-06-12T00:04:07.621Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWrite - {"status":0}
2021-06-12T00:04:07.622Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:07.623Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWrite - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}}
2021-06-12T00:04:07.624Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,8,33,9,132,0,0,4,0,8,0,0,168]
2021-06-12T00:04:07.633Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,0,105]
2021-06-12T00:04:07.633Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,0,105]
2021-06-12T00:04:07.633Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [0] - 105
2021-06-12T00:04:07.634Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWrite - {"status":0}
2021-06-12T00:04:07.634Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:07.636Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWrite - {"id":131,"len":2,"offset":0,"value":{"type":"Buffer","data":[98,26]}}
2021-06-12T00:04:07.637Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,6,33,9,131,0,0,2,98,26,215]
2021-06-12T00:04:07.644Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,0,105]
2021-06-12T00:04:07.645Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,0,105]
2021-06-12T00:04:07.645Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [0] - 105
[... 32 lines cut ...]
2021-06-12T00:04:10.847Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,111,5,0,107]
2021-06-12T00:04:10.847Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,111,5,0,107]
2021-06-12T00:04:10.847Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 15 - 5 - [0] - 107
2021-06-12T00:04:10.848Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- APP_CNF - bdbStartCommissioning - {"status":0}
2021-06-12T00:04:10.849Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:10.850Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,79,128,1,2,4,203]
2021-06-12T00:04:10.851Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,79,128,1,2,4,203]
2021-06-12T00:04:10.851Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 15 - 128 - [1,2,4] - 203
2021-06-12T00:04:10.851Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":1}
2021-06-12T00:04:10.852Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:11.076Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,192,8,140]
2021-06-12T00:04:11.076Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140]
2021-06-12T00:04:11.077Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140
2021-06-12T00:04:11.077Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8}
2021-06-12T00:04:11.078Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:11.633Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,192,8,140]
2021-06-12T00:04:11.633Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,192,8,140]
2021-06-12T00:04:11.634Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 192 - [8] - 140
2021-06-12T00:04:11.634Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - stateChangeInd - {"state":8}
2021-06-12T00:04:11.635Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-12T00:04:11.636Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,79,128,8,2,0,198]
2021-06-12T00:04:11.637Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,79,128,8,2,0,198]
2021-06-12T00:04:11.637Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 15 - 128 - [8,2,0] - 198
2021-06-12T00:04:11.637Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- APP_CNF - bdbComissioningNotifcation - {"status":8}
```

Koenkk, 2021-06-12:
Can you reproduce this problem by:
- Setting a different pan id, start z2m -> z2m should start now
- Stop and restart z2m -> do you get the pan id duplication error now?

dummys, 2021-06-12:
If I stop and restart, I don't get the duplicate error, because I guess my pan_id is not the same as all my deployed device, right ?

Another problem if I want to follow the best practice and use the full serial device name:
```
Zigbee2MQTT:error 2021-06-12 21:17:02: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_24_0F-if00-port0
```

and the name of the serial device is correct, if we look inside the terminal:

[image]

Koenkk, 2021-06-14:
@dummys when doing this the pan id should be the same, can you check?

- Setting a different pan id, start z2m -> z2m should start now, **now pair some devices**
- Stop and restart z2m -> do you get the pan id duplication error now?

dummys, 2021-06-16:
no because now I changed the pan id. and yes all my device are back but I had to repair ALL of them...

Koenkk, 2021-06-18:
So what if you don't change your pan id? (procedure of https://github.com/Koenkk/zigbee2mqtt/issues/7761#issuecomment-860754446)

dummys, 2021-06-24:
> So what if you don't change your pan id? (procedure of [#7761 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/7761#issuecomment-860754446))

not possible, If I don't change the service didnt start

Koenkk, 2021-06-26:
It's probably the same as https://github.com/Koenkk/zigbee2mqtt/issues/6302 , try reflashing the firmware with a different computer.

dummys, 2021-06-28:
already tried, same problem. I re-paired everything :( Lets close this ticket. anyway, thanks for help

dummys, 2021-07-08:
@Koenkk , sorry to re-open but thats crazy man. I have just restarted my home assistant Odroid device, without any UPDATE/change to z2m config or whatever, and got AGAIN The F***** pan id error. WTF WITH THIS DEVICE SRSLY ?
Last time I setuped new PAN_ID, repaired all. I really don't want to do that each time I reboot my HA device to be honest. WHere is the bug ? I bought the SLAE because in your documentation it stated that it is the best device, but it is a nightmare to repair all device.
I also tried your stuff with removing external antenna, using foil on the dongle, nothing it wont start at all ...

Koenkk, 2021-07-09:
First make sure you are on at least 1.19.0 (big changes to backup mechanism there). 

Next reflash your stick and change the `pan_id` and start (new network will be created).
Now connect pair one router device (e.g. a bulb, not a battery powered one). Next restart z2m, after this is the bulb still in the database?

dummys, 2021-07-11:
> First make sure you are on at least 1.19.0 (big changes to backup mechanism there).
> 
To be honest I don't see why I would upgrade as the existing one must work. I don't want to change stuff before fixing, or understanding why I get this. To be honest I lose so much time with this, all my house is not working everytime update/or restart of HA/Z2m. I changed from CC to slae back in the days because in your page it was the recommenced one, I bought two of it and I don't get the point of this pan_id error. When a coordinator create a zigbee network, and there is router, I don't get the point how the router can tell to the corrdinator that it is a corrdinator. OR I missed something, or something is really wrong of the firmware that you made for slae. I am there to help you fix this, I have 3 weeks vacation from now so I can fully work on this but I need it fixed. It's not possible to re-pair, every device (30) everytime as I plan to have device that will be hard to reset when they will be installed ...

Koenkk, 2021-07-12:
Did you try the procedure I described in https://github.com/Koenkk/zigbee2mqtt/issues/7761#issuecomment-877230153 ?

dummys, 2021-07-12:
> Did you try the procedure I described in [#7761 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/7761#issuecomment-877230153) ?

yes, I answered you on telegram

dummys, 2021-07-12:
I have changed pan_id, channel, network_key. Added a hue bulb. Stop/restart z2m:
Error: AREQ - ZDO - stateChangeInd after 60000ms
Then it try to restart it again annnnd: guess what: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel

Koenkk, 2021-07-12:
I still think the flashed firmware is corrupt (like in https://github.com/Koenkk/zigbee2mqtt/issues/6302). This has nothing to do with the stick or z2m, but rather with the setup that the firmware has been flashed with. Can you reflash the firmware on a completely different setup or with Uniflash? (https://slae.sh/projects/cc2652/#flashing)

dummys, 2021-07-12:
Yes I told you already that I reflash using Windows with python bootloader tool. Now I have upgraded to 1.20-1 latest version. I have restart and get the: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

Koenkk, 2021-07-13:
Good to hear you fixed the problem. In some rare cases (like this and #6302) the cc2538 bsl flasher fails. This can be checked by reading both firmwares back (`cc2538-bsl.py -r`) you will see that the firmwares differ.

dummys, 2021-07-13:
Sure. but it could be nice to understand why, or simply remove the documentation for this flasher.
Also can you add to github the sha1 or md5 of each firmware ? Just to be sure that the actual downloaded file is the correct one.
Thanks for help on the debugging.

closed 2021-07-13 by dummys