Koenkk/zigbee2mqtt #7763: E1745 does not support battery anymore

baderas, 2021-06-12:
## What happened
My E1745 does not report any battery values anyome. Not sure since which version of zigbee2mqtt, currently i'm using 1.19.1.
I recognized it because my batteries were empty and I did still have a reading of 40% (last reported value sometime in the past). After I put in new batteries, no new battery value was send since 3 hours:

`zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:16:05+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:17:42+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:19:10+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:19:41+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:21:35+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:24:02+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:24:02+02:00","linkquality":111,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:29:01+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:30:41+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:30:41+02:00","linkquality":111,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:47:12+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:49:21+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:51:20+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T12:51:20+02:00","linkquality":111,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:21:02+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:23:14+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:25:26+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:27:14+02:00","linkquality":21,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:29:00+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:30:45+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:30:45+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:39:26+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:41:26+02:00","linkquality":111,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:43:12+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:43:12+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:55:23+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:57:26+02:00","linkquality":114,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:59:03+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T13:59:03+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:02:23+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:05:01+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:05:01+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:11:37+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:11:37+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:16:14+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:16:14+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:26:45+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:26:45+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:30:22+02:00","linkquality":99,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:32:41+02:00","linkquality":102,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:34:23+02:00","linkquality":105,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:36:00+02:00","linkquality":108,"occupancy":true,"update":{"state":"idle"},"update_available":false}
zigbee2mqtt/0x000d6ffffe42a529 {"last_seen":"2021-06-12T14:36:00+02:00","linkquality":108,"occupancy":false,"update":{"state":"idle"},"update_available":false}
`

## What did you expect to happen
- reporting 100% (or 80/90%) after changing both batteries
- empty batteries reporting 0% or 10% not 40%

## Debug info
Zigbee2MQTT version: 1.19.1

smulle48, 2021-06-13:
My two sensors are reporting just fine

`MQTT publish: topic 'zigbee2mqtt/xxxx', payload '{"battery":74,"device":{"applicationVersion":17,"dateCode":"20170302","friendlyName":"Gang Sensor","hardwareVersion":1,"ieeeAddr":"0xxxxxxxxxx","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1525/E1745","networkAddress":xxxx,"powerSource":"Battery","softwareBuildID":"1.2.214","stackVersion":87,"type":"EndDevice","zclVersion":1},"linkquality":78,"occupancy":true,"update":{"state":"idle"},"update_available":false}'`

DevDoesOps, 2021-07-07:
I have a similar issue with invalid battery level reporting.  The battery is still reporting at 47%, even just before the batteries completely died.  I've replaced the batteries and 48 hours later it's still reporting 47%.  I have another of these sensors in another room which is also reporting 47% battery - could be a coincidence.

I'm using v1.20.0 with IKEA E1745.

```
MQTT publish: topic 'zigbee2mqtt/*****1, payload '{"battery":47,"linkquality":69,"occupancy":false,"requested_brightness_level":76,"requested_brightness_percent":30,"update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/*****2', payload '{"battery":47,"linkquality":51,"occupancy":false,"update":{"state":"idle"},"update_available":false}'
```

closed 2021-08-14 by github-actions[bot]