Koenkk/zigbee2mqtt #7764: Lupus LS12128 not supported???

renne, 2021-06-12:
# Bug Report
## What happened
After pairing a [Lupus LS12128](https://www.zigbee2mqtt.io/devices/LS12128.html) roller shutter is marked as not supported.

## What did you expect to happen
Supported as stated on [Lupus LS12128](https://www.zigbee2mqtt.io/devices/LS12128.html).

## How to reproduce it (minimal and precise)
Press button on [Lupus LS12128](https://www.zigbee2mqtt.io/devices/LS12128.html) ten seconds for pairing and wait until device is listed in the device list.

## Debug Info
Zigbee2MQTT version: Updated 20 Minutes ago
Adapter hardware: CC2652RB (Slaesh)
Adapter firmware version: ???

## Device:
### About
| key | value |
| --- | --- |
| Friendly name | 0x00124b001fe91d7a |
| Last seen | N/A |
| Device type | EndDevice |
| Zigbee Model | SCM-6-OTA_00.00.03.17TC |
| Support status | not_supported |
| IEEE Address | 0x00124b001fe91d7a |
| Network address | 0x5DFF |
| Firmware build date | 20200702  |
| Power Interview completed | True |

### Exposes
empty_exposes_definition

sjorge, 2021-06-12:
Looks like this is a newer model, the ID you posted is: `SCM-6-OTA_00.00.03.17TC` while the one we have is `SCM-3-OTA_00.00.03.16TC`. 

I'll open a PR to add the new model to the list.

renne, 2021-06-13:
According to the manual shipped with Lupus LS12128 there are firmware versions <= 3.16 and >= 3.17 with different usage of the switch inputs. This shouldn't affect zigbee2mqtt.
It seems Lupus encodes the firmware version into the device model name. We'll have to update the device model name with every new firmware version.

renne, 2021-06-13:
I applied the code from the pull request to my installation. The Lupus LS12128 works (I can hear the relays switching).
How does one configure the shutter up and down run-times?

kds69, 2021-06-15:
Hi,
Actually I wanted to do the same + add **level_config** support (tested successfully), which is not currently the case.
see issue  in  Koenkk / zigbee-herdsman  [https://github.com/Koenkk/zigbee-herdsman/issues/382](url)

renne, 2021-06-15:
@kds69 The issue in your link doesn't seem to exists.
Do you have code for the **level_config** I can test?

kds69, 2021-06-15:
There is no code needed this is available by default in the herdsman's tozigbee converters. You need to extend level_config to your device locally.
see documentation.

renne, 2021-06-15:
@kds69 Can you point me to a documentation?

renne, 2021-06-17:
New firmware:
Zigbee Model: SCM-6-OTA_00.00.03.18TC

closed 2021-07-25 by github-actions[bot]