Koenkk/zigbee2mqtt #7768: Binding failed reporting on cluster 'closuresWindowCovering'

Nornode, 2021-06-12:
## What happened
I'm trying to bind an E1766 (Ikea Fyrtur Open/Close Remote) with a group of Fyrtur, It's not binding and or does not perform the actions expected.

## What did you expect to happen
It would bind and I can control it as it was working "out of the box" with the added functionality of Z2M & Home Assistant.

## How to reproduce it (minimal and precise)
1. Make a group named "Fyrtur" of one Fyrtur Blind (Through UI)
2. Bind the E1766 with the group. (Through MQTT)
3. Read logs: [Logs](https://pastebin.ubuntu.com/p/GkKRrkrgwT/)
  -> `Failed to setup reporting for 'Fyrtur_Loftet/1' cluster 'closuresWindowCovering'`
4. Button clicks still work, there is no action on the blind(s). (I have only paired one blind so far)

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: Slae.sh - CC2652RB
Adapter firmware version: ?
Coordinatior version: {'meta': {'maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20210120, 'transportrev': 2}, 'type': 'zStack3x0'}

kneip68, 2021-06-14:
Don't really know whether it is related, but since an OTA update yesterday I could not bind several remotes (E1743, E1524/E1810) to groups anymore, instead I had to bind directly to device (which was not possible with the older firmware). As I can see you are binding against a group as well, might be worthwhile to try to bind to the device instead. I did not update my E1766 remote for the blinds yet, did not dare.

kneip68, 2021-06-14:
... oh and by the way, if you are on the older firmware, did you delete the default bind group from the remote first?

martyzeq, 2021-06-15:
wow @kneip68 thx for hint, also here unable to bind to group after OTA update of switches, but binding to even multiple devices works... Kudos man :-)

closed 2021-07-23 by github-actions[bot]