Koenkk/zigbee2mqtt #7769: TuYa TS0601_cove not available

vgonzalez84, 2021-06-12:
## What happened
TuYa TS0601_cover not available since Home Assistant core-2021.6.x update

## What did you expect to happen
That the device continues to function properly

## How to reproduce it (minimal and precise)
updating Home Assistant to version core-2021.6.0, core-2021.6.1, core-2021.6.2, core-2021.6.3 ir core-2021.6.4

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: Z-Stack_Home_1.2

elbueno222, 2021-06-13:
I might have the same issue with Ikea Blind, this is the issue I can see in the logs:

```
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: June 12, 2021, 8:55:41 PM (1 occurrences)
Last logged: June 12, 2021, 8:55:41 PM

Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/0xec1bbdfffe2e272b/set', 'device': {'identifiers': ['zigbee2mqtt_0xec1bbdfffe2e272b'], 'manufacturer': 'IKEA', 'model': 'FYRTUR roller blind (E1757)', 'name': '0xec1bbdfffe2e272b', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/0xec1bbdfffe2e272b', 'name': '0xec1bbdfffe2e272b', 'position_topic': 'zigbee2mqtt/0xec1bbdfffe2e272b', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/0xec1bbdfffe2e272b/set', 'unique_id': '0xec1bbdfffe2e272b_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options raise vol.Invalid( voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'. During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
```

I think this could be related to this Breaking change: https://github.com/home-assistant/core/pull/46059

elbueno222, 2021-06-13:
@Koenkk does this Breaking change need to be applied to z2m? home-assistant/core#46059

elbueno222, 2021-06-13:
After restoring an old backup and updating everything to latest versions the blind is working again, so not sure what was causing the issue but it doesn´t look like the Breaking change I mentioned before was the issue (confirmed by Koenkk here https://github.com/Koenkk/zigbee2mqtt/issues/7744#issuecomment-858792073).

@vgonzalez84 did you remove Tuya integration from yaml config file and added from the Integration tab? If you didn´t that might be causing the issue: https://www.home-assistant.io/blog/2021/06/02/release-20216/#integrations-that-removed-deprecated-yaml-import

vgonzalez84, 2021-06-13:
Yes but I don't use a Tuya hub. The cover connect to Home Assistant using Zigbee2Mqtt.

elbueno222, 2021-06-13:
In that case I would recommend you to search for any errors in the zigbee2mqtt log or Home assistant log (if you are using HA) related to the cover and share it here to see if someone can help (I doubt anyone will be able to help with the amount of information you have provided).

vgonzalez84, 2021-06-13:
I search the error in diferentes logs but only the error appears in the Home Assistant log

vgonzalez84, 2021-06-13:
Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 18:22:01 (3 occurrences)
Last logged: 18:22:01

Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Cortina Salón/set', 'device': {'identifiers': ['zigbee2mqtt_0x5c0272fffedec0fe'], 'manufacturer': 'TuYa', 'model': 'Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover)', 'name': 'Cortina Salón', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Cortina Salón', 'name': 'Cortina Salón', 'position_topic': 'zigbee2mqtt/Cortina Salón', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Cortina Salón/set', 'unique_id': '0x5c0272fffedec0fe_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options raise vol.Invalid( voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'. During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Cortina Lucía/set', 'device': {'identifiers': ['zigbee2mqtt_0x5c0272fffee1792d'], 'manufacturer': 'TuYa', 'model': 'Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover)', 'name': 'Cortina Lucía', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Cortina Lucía', 'name': 'Cortina Lucía', 'position_topic': 'zigbee2mqtt/Cortina Lucía', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Cortina Lucía/set', 'unique_id': '0x5c0272fffee1792d_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options raise vol.Invalid( voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'. During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Cortina Dormitorio/set', 'device': {'identifiers': ['zigbee2mqtt_0x804b50fffe4e49d8'], 'manufacturer': 'TuYa', 'model': 'Curtain motor/roller blind motor/window pusher/tubular motor (TS0601_cover)', 'name': 'Cortina Dormitorio', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Cortina Dormitorio', 'name': 'Cortina Dormitorio', 'position_topic': 'zigbee2mqtt/Cortina Dormitorio', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Cortina Dormitorio/set', 'unique_id': '0x804b50fffe4e49d8_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, i [... cut ...]  self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.

elbueno222, 2021-06-13:
If you have version 1.18.1-1 you need to change to new repository to update to v 1.19, mentioned here (first point): https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

vgonzalez84, 2021-06-13:
Thats is the solution!!!
Thank you very much for your help

closed 2021-06-13 by vgonzalez84