Koenkk/zigbee2mqtt #7778: LEDVANCE Smart+ Outdoor Plug

snorro80, 2021-06-13:
I managed to add the LEDVANCE Smart+ outdoor plug manually, but I don't know how contribute to Z2M thru a pull request.
Could someone please add this to Z2M?

[Outdoor Plug info ](https://www.benelux.ledvance.com/nl/professioneel/producten/smart-home/smart-home-producten-met-zigbee-technologie/smart-home-components/smart-outdoor-plug/index.jsp)

Z2M settings:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Plug Value'],
    model: 'AC26940',
    vendor: 'LEDVANCE',
    description: 'Smart Zigbee Outdoor Plug',
    fromZigbee: [fz.on_off],
    toZigbee: [fz.on_off],
    exposes: [e.switch()],
};

module.exports = definition;
```

Koenkk, 2021-06-14:
Can you share the data/database.db entry of this device?

snorro80, 2021-06-14:
Here you go.

`{"id":18,"type":"Router","ieeeAddr":"0xf0d1b800001a5c2e","nwkAddr":52470,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Unknown","modelId":"Plug Value","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug Value","manufacturerName":"LEDVANCE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"00203045"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b000be8abed","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"00203045","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1623696177760}
`

Koenkk, 2021-06-15:
Added, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-15 by Koenkk

JankoSchwalbe, 2021-11-30:
I have an outdoor plug with ID AC03360 that looks the same as yours. Is there an easy way to add it for testing? Currently I'm getting "unsupported device"

XPAction, 2021-12-11:
Ive just got one AC31267 that  cant connect