Koenkk/zigbee2mqtt #7779: New Device Iluminize Zigbee 3.0 LED Controller 511.050

kniggei, 2021-06-13:
I managed to add the Iluminize LED-Controller 511.050 manually, but I don't know how contribute to Z2M thru a pull request.
Could someone please add this to Z2M?

https://www.iluminize.com/de/shop/led-steuerung/led-controller/product/519-5110-80-zigbee-controller-5a.html

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['511.050'],
    model: '511.050',
    vendor: 'Iluminize',
    description: 'Zigbee 3.0 LED-Controller für 5in1 RGB+CCT LEDs',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

----------------------------------------------------------------------------

{"id":31,"type":"Router","ieeeAddr":"0x680ae2fffe3a6f25","nwkAddr":6039,"manufId":4107,"manufName":"iluminize","powerSource":"Mains (single phase)","modelId":"511.050","epList":[1,242],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"511.050","manufacturerName":"iluminize","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r47"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"lightingBallastCfg":{"attributes":{}},"genPowerCfg":{"attributes":{}},"genDeviceTempCfg":{"attributes":{}},"genIdentify":{"attributes":{}},"genScenes":{"attributes":{}},"lightingColorCtrl":{"attributes":{}},"msIlluminanceMeasurement":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.5.3_r47","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1623598944266}

------------------------------------------------------------------------------

Please help.

Koenkk, 2021-06-14:
Please also provide the min and max color temp: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

kniggei, 2021-06-14:
info  2021-06-14 19:24:39: Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":155,"colorTempPhysicalMax":450}

Koenkk, 2021-06-15:
Added. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-15 by Koenkk