Koenkk/zigbee2mqtt #7781: Elko SmartStikk Enkel

bjornidar, 2021-06-13:
## Information about the device + link
https://proff.elko.no/smartstikk-enkel-16a/smartstikk-enkel-16a-plus-ph-article3274-3119.html

Should be able to read current consumption - not available from device file below.
Tried with the "electrical_measurement" without any success. 

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['SOCKET/OUTLET/1'],
        model: 'EKO09716',
        vendor: 'Schneider Electric',
        description: 'Elko Plus SmartStikk Enkel',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        exposes: [e.switch()],
};
module.exports = definition;`

TheStigh, 2021-06-13:
Hei Bjørn,

Do you have the doc of this smartstikk ?

bjornidar, 2021-06-13:
From ZigbeeAlliance
https://zigbeealliance.org/zigbee_products/elko-smartsocket-single-16a/

https://proff.elko.no/getfile.php/13162720/produkter/doc/EKO09716/SmartStikk%20Enkel%2016A.pdf
https://proff.elko.no/getfile.php/13162544/produkter/doc/EKO09716/Datasheet4540088.pdf

TheStigh, 2021-06-13:
> 
> 
> Is this of any help?
> 
> From ZigbeeAlliance
> https://zigbeealliance.org/zigbee_products/elko-smartsocket-single-16a/
> 
> https://proff.elko.no/getfile.php/13162720/produkter/doc/EKO09716/SmartStikk%20Enkel%2016A.pdf
> https://proff.elko.no/getfile.php/13162544/produkter/doc/EKO09716/Datasheet4540088.pdf

Try the code in my initial post

bjornidar, 2021-06-14:
error: Failed to configure 'Stikk 2 Kjøkken', attempt 2 (TypeError: Cannot read property 'bind' of undefined at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24) at Object.configure (/app/lib/util/externally-loaded.js:19:29) at Configure.configure (/app/lib/extension/configure.js:137:45) at Configure.onZigbeeEvent (/app/lib/extension/configure.js:119:18) at Controller.callExtensionMethod (/app/lib/controller.js:386:44))

debug: Received MQTT message on 'zigbee2mqtt/Stikk 2 Kjøkken/set' with data '{"state":"OFF"}'

debug: Publishing 'set' 'state' to 'Stikk 2 Kjøkken'
info: MQTT publish: topic 'zigbee2mqtt/Stikk 2 Kjøkken', payload '{"energy":null,"linkquality":135,"power":null,"state":"OFF"}'

debug: Received MQTT message on 'zigbee2mqtt/Stikk 2 Kjøkken/set' with data '{"state":"ON"}'

debug: Publishing 'set' 'state' to 'Stikk 2 Kjøkken'

info: MQTT publish: topic 'zigbee2mqtt/Stikk 2 Kjøkken', payload '{"energy":null,"linkquality":144,"power":null,"state":"ON"}'

TheStigh, 2021-06-14:
Uhh, sorry, try to change `const endpoint = device.getEndpoint(11);` to `const endpoint = device.getEndpoint(1);`

When you paste code/log, select the pasted code here and press the `insert code` button.  Makes it formatted and easier to read.

[image]

bjornidar, 2021-06-14:
I see that sometimes when I try to insert code it's not "handled" by the editor.
_If you see my first post the `-signs are surrounding my code..._

Tried changing to getEndpoint(1) with the same result - do you know if I have to remove/ rejoin the device or is it OK just to restart Z2M?

`info MQTT publish: topic 'zigbee2mqtt/Stikk 2 Kjøkken', payload '{"energy":null,"linkquality":156,"power":null,"state":"ON"}'
debug Received MQTT message on 'zigbee2mqtt/Stikk 2 Kjøkken/set' with data '{"state":"OFF"}'
debug Publishing 'set' 'state' to 'Stikk 2 Kjøkken'
info MQTT publish: topic 'zigbee2mqtt/Stikk 2 Kjøkken', payload '{"energy":null,"linkquality":156,"power":null,"state":"OFF"}'`

TheStigh, 2021-06-14:
> Tried changing to getEndpoint(1) with the same result - do you know if I have to remove/ rejoin the device or is it OK just to restart Z2M?

As a minimum you should go into the Z2M UI and do a `Reconfigure`. Sometimes best are to rejoin.

After doing the `Reconfigure`, please copy the entire log part from where you see it does the /set for `configuring` until part where it says `success `or `failed`. Can you also please share your database.db entry?

bjornidar, 2021-06-14:
Tried to reconfigure;

`Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Stikk 2 Kjøkken","transaction":"bs3si-1"}'
Info Configuring 'Stikk 2 Kjøkken'
Error Failed to configure 'Stikk 2 Kjøkken', attempt 2 (TypeError: Cannot read property 'bind' of undefined at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:24) at Object.configure (/app/lib/util/externally-loaded.js:19:29) at Configure.configure (/app/lib/extension/configure.js:137:45) at Configure.onMQTTMessage (/app/lib/extension/configure.js:88:32) at Controller.callExtensionMethod (/app/lib/controller.js:386:44) at processTicksAndRejections (internal/process/task_queues.js:93:5))
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Stikk 2 Kjøkken"},"error":"Failed to configure (Cannot read property 'bind' of undefined)","status":"error","transaction":"bs3si-1"}'`

TheStigh, 2021-06-14:
Please share the database.db entry for this device

bjornidar, 2021-06-14:
`{"id":3,"type":"Router","ieeeAddr":"0x680ae2fffe11774d","nwkAddr":33177,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"SOCKET/OUTLET/1","epList":[6,242],"endpoints":{"6":{"profId":260,"epId":6,"devId":9,"inClusterList":[0,3,4,5,6,1794,1800,2820,2821,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SOCKET/OUTLET/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.006.014 R"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.006.014 R","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1623672183111}`

TheStigh, 2021-06-14:
Change the Endpoint to 6 and try

closed 2021-07-23 by github-actions[bot]

Sabesto, 2021-12-13:
I bought the 2-outlet version of this device (only one is controlled/metered). I have gotten this far, but power
(Instantaneous measured power in seMetering) is reported with a value 1000 times higher then the correct value.
```
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['SOCKET/OUTLET/2'],
        model: 'EKO09738',
        vendor: 'Schneider Electric',
        description: 'Zigbee smart plug with power meter',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.power_on_behavior],
        toZigbee: [tz.on_off, tz.power_on_behavior],
        exposes: [
            e.switch(),
            e.power(),
            e.energy(),
            exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on']).withDescription('Controls the behaviour when the device is powered$
            e.current(),
            e.voltage()
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(6);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await endpoint.read('haElectricalMeasurement', ['acCurrentDivisor', 'acCurrentMultiplier']);
            await endpoint.read('seMetering', ['multiplier', 'divisor']);
        },
    },
];
```

This results in:
Power 239588W
Energy 0.8kWh
Current 1.08A
Voltage 228V
`
MQTT publish: topic 'zigbee2mqtt/Varmtvannbereder Sokkel', payload '{"current":1.09,"energy":0.77,"linkquality":191,"power":241195,"power_on_behavior":"on","state":"ON","voltage":230}'
`

Poking all seMetering attributes i could find in the zigbee compliance document yields this:
`Read result of 'seMetering': {"multiplier":1,"divisor":1000,"summaFormatting":75,"status":0,"currentSummDelivered":[0,670],"extendedStatus":"0x0000000000000000","unitOfMeasure":0,"meteringDeviceType":0,"instantaneousDemand":237005}`

michaelthunemberglund, 2022-03-08:
Hi there. 
I am experiencing problems with the "power" meter readings in HA with Zigbee2MQTT.
HA is reporting actual power readings for a couple seconds, then show "null" for a second resulting HA to remove the power information, then back to actual readings repeating endlessly. 
When the equipment connectet to the socket is in standymode with power draw around 1-4watt its reporting continuously, but when the load exceed a certain level it starts to report "null" as a state every 2-3 seconds. 

Anybody have a fix for this?