Koenkk/zigbee2mqtt #7783: IKEA E1757 Fyrtur position not reported

illivanillie, 2021-06-13:
## What happened
Ikea Fyrtur cover position is often not reported, if controlled via open_cover or close_cover.
The position is correctly published, if the position is set.
The position will be reported, after the refresh button on the HA add-on is clicked.
`zigbee2mqtt/FRIENDLY_NAME/get with payload {"state": ""} ` will give me ‚Error No converter available for 'get' 'state' ()‘
## What did you expect to happen

The position should be reported. 
The main problem here is that in HA the arrows are not functional because of the assumed/last reported state. So open cover cannot be closed and vice versa. There are workarounds, for sure, but it is inconvenient for other members of the house.

## How to reproduce it (minimal and precise)
I have no idea. Just set up like documented via discovery. I have around 10 of them and its the same with all. Sometimes it works, sometimes not. The covers are not linked to the amplifier, but latter is in the network.
I‘ve been waiting now for months but the June update did not change anything.

## Debug info
Zigbee2MQTT version: no update available
Adapter hardware: CC26X2R1
Adapter firmware version: 3.X.0 20210120

If log infos are any help for you, I can try to provide it.

Greetings!

illivanillie, 2021-06-13:
and also in the add on: device: exposes: state it says ‚CLOSE ? OPEN‘ for me. I don‘t know if the question marks between it is common or not

illivanillie, 2021-06-13:
And last seen is always N/A.
But now that’s it I think..

Nornode, 2021-06-14:
Did you update the Fyrtur firmware? - They have had trouble in the past.

illivanillie, 2021-06-14:
I am updating since yesterday. Takes hours and hours. 
Is it ok if I reconfigure via the button? Or do I need to factory reset and rejoin them to the network?
Btw: what is the command send by the refresh button? Maybe I could implement this in scripts/automations.

sjorge, 2021-06-14:
> Is it ok if I reconfigure via the button? Or do I need to factory reset and rejoin them to the network?

Hitting reconfigure should be fine, if it spews timeouts, hit a button on the device and hit reconfigure again.

> Btw: what is the command send by the refresh button? Maybe I could implement this in scripts/automations.

You can trigger it by sending '' to the `zigbee2mqtt/0x0000000000/get/position` e.g.

```bash
mosquitto_pub -t zibgee2mqtt/cover/livingroom/side1/get/position -m ''
```

illivanillie, 2021-06-14:
@sjorge The configuration for reporting is correct. For now it is working again and your payload message also gives me a response at the moment.
I will have to wait until it is not working again and will update you guys on how the firmware update changed the issue.

illivanillie, 2021-06-14:
Update:

It is NOT working. Disappointing..
But! mqtt get position manually works perfectly fine for now.
The workaround now will be sending the message in a loop whenever script.open/close_cover is started.

If you have any further suggestions for a native fix, I would appreciate it.

greetings!

illivanillie, 2021-06-16:
In case someone has the same issue, this is my automation from HA:
`alias: Cover get postion
description: Asks for position of whichever cover was changed
trigger:
  - platform: event
    event_type: call_service
    event_data:
      domain: mqtt
      service: publish
      service_data:
        payload: OPEN
  - platform: event
    event_type: call_service
    event_data:
      domain: mqtt
      service: publish
      service_data:
        payload: CLOSE
  - platform: event
    event_type: call_service
    event_data:
      domain: mqtt
      service: publish
      service_data:
        payload_position: null
condition: []
action:
  - repeat:
      count: '60'
      sequence:
        - service: mqtt.publish
          data:
            topic: >-
              zigbee2mqtt/{{trigger.event.data.service_data.topic.split("/")[1]}}/get
            payload: '{"position": ""}'
        - delay:
            hours: 0
            minutes: 0
            seconds: 5
            milliseconds: 0
mode: parallel
max: 20 `

triggers on open, close and set position.
asks for position every 5 seconds for 1 minute.
parallel for 20 covers (in case you ask a group to open)

AshtonKem, 2021-07-04:
I had similar issues until today, but running an OTA update *before* reconfiguring the devices fixed it.

MattWestb, 2021-08-04:
@AshtonKem is your blinds still working OK after have updating them with the last OTA ??

Im curious if the OTA is helping with connection / reporting issues with the blinds.

AshtonKem, 2021-09-06:
I’ve had some intermittent issues with one blind not responding to commands to open/close, which appear to have been resolved by explicitly calling the cover.set_position action in my automations rather than setting device state. Otherwise, no issues with it reporting state back.

MattWestb, 2021-09-06:
IKEA have one more new update for there blinds in the pipe its certificated but not released in the test feed. 
Its looks they is updating all "modern" controller devices but was having problem with the first try that was braking group binding of controllers.

Hopefully they is fixing some more bugs and not implanting to much new ones ;-))

closed 2021-09-14 by github-actions[bot]

illivanillie, 2021-09-14:
@MattWestb Could you please keep us updated on the firmware update?

MattWestb, 2021-09-14:
IKEA have certified the last "modern" controllers (no of the motion sensors and the old round dimmer) so all of them is done but they have not showing up in the test feed as the blinds.
For the blinds its better waiting little then its being released in the test for not braking them like many users was doing with there remotes for some mounts ago then the blinds is little more problematic then one not working remote.

I thinking they is releasing the set very soon then the Android app was updated yesterday but no new firmware for the HUB but it can being its not needed.

I keep one eye on the firmware and reporting then they is being released in the test feed and also if user is getting problems or if its working OK then its released.

Xitro01, 2022-01-30:
I have the issue that the Tradfri Blind is not automatically updating its position since I updated the fw to: 2.3.088.
I have to manually press the refresh button now.

Is there any way to fix this behaviour? I have already tried the reconfigure button without any succes.

Thanks in advance.

Xitro01, 2022-01-31:
> In case someone has the same issue, this is my automation from HA:
> 
> ```
> alias: Cover get postion
> description: Asks for position of whichever cover was changed
> trigger:
>   - platform: event
>     event_type: call_service
>     event_data:
>       domain: mqtt
>       service: publish
>       service_data:
>         payload: OPEN
>   - platform: event
>     event_type: call_service
>     event_data:
>       domain: mqtt
>       service: publish
>       service_data:
>         payload: CLOSE
>   - platform: event
>     event_type: call_service
>     event_data:
>       domain: mqtt
>       service: publish
>       service_data:
>         payload_position: null
> condition: []
> action:
>   - repeat:
>       count: '12'
>       sequence:
>         - service: mqtt.publish
>           data:
>             topic: >-
>               zigbee2mqtt/{{trigger.event.data.service_data.topic.split("/")[1]}}/get
>             payload: '{"position": ""}'
>         - delay:
>             hours: 0
>             minutes: 0
>             seconds: 5
>             milliseconds: 0
> mode: parallel
> max: 20 
> ```
> 
> triggers on open, close and set position. asks for position every 5 seconds for 1 minute. parallel for 20 covers (in case you ask a group to open)

Any idea why this is not working for me? It does not seem to trigger.

Xitro01, 2022-01-31:
Ok, temporarily fixed this issue with the following automation:

`- alias: Tradfri cover get position (FIX)
  description: Asks for position of whichever cover was changed
  trigger:
    - platform: event
      event_type: call_service
      event_data:
        domain: cover
  condition: []
  action:
    - repeat:
        count: "12"
        sequence:
          - service: mqtt.publish
            data:
              topic: >-
                zigbee2mqtt/Tradfri Blind/get
              payload: '{"position": ""}'
          - delay:
              hours: 0
              minutes: 0
              seconds: 5
              milliseconds: 0
  mode: parallel
  max:
    20`

Improvements are welcome.

AshtonKem, 2022-02-01:
I don’t understand the zigbee protocol very well, but this sure seems like something you’d have to do if there isn’t a correct timeout or retry configured in the coordinator somewhere. 

—
Ashton Kemerling

Sent from my iPhone.

miguelangel-nubla, 2022-09-03:
It was solved for me by hitting apply on Cover.currentPositionLiftPercentage on the reporting tab as mentioned here (https://github.com/Koenkk/zigbee2mqtt/issues/7783#issuecomment-860479828)