Koenkk/zigbee2mqtt #7784: Coordinator's leds switch off after zigbee2mqtt 1.19.1 is started

PWnet, 2021-06-14:
Hi,

my configuration is:

zigbee2mqtt: 1.19.1 docker version
coordinator: cc2538_cc2592 (Zigbee Stick v3.1) with firmware JH_2538_2592_ZNP_USB_20201010

with version 1.19.1 the coordinator's leds switch off after zigbee2mqtt is started and all work fine.
If I enable device joining the leds switch on again and then switch off when I disable device joining.

In my configuration I have:

> serial:
>   disable_led: false
>   port: /dev/ttyACM0

No change if I remove "disable led"

No issues with version 1.18.3.

Thank you

Roberto

sjorge, 2021-06-14:
I think this is expected behavior, as it was introduced by https://github.com/Koenkk/zigbee-herdsman/commit/8d50ba0cd78369b7f93911f00d6a47f7972667df

robertalexa, 2021-06-14:
I get the same behaviour on mine.

You have mentioned that there is no change if you remove disabled_led:false - as mentioned in the documentation, disabled_led has a default value of false, thus no change.
[image]

PWnet, 2021-06-14:
> I think this is expected behavior, as it was introduced by [Koenkk/zigbee-herdsman@8d50ba0](https://github.com/Koenkk/zigbee-herdsman/commit/8d50ba0cd78369b7f93911f00d6a47f7972667df)

This type of coordinator device already had this behaviour.
The device has four led:
1-adapter led (usually always on)
2-transmit led (blink during zigbee communications)
3-join led (on when joining is enable)
4-unused (always off)

This new feature switch off all the leds if joining is disabled. When joining is enabled all the led are switched on with their original function.

Anyway, it's not a big problem, perhaps only noticeable with this particular device.

(I hope you can understand my poor english)
Roberto

Giancky79, 2021-06-17:
https://github.com/Koenkk/zigbee2mqtt/issues/7754

PWnet, 2021-07-22:
Same issue with 1.20.0

closed 2021-08-31 by github-actions[bot]