Koenkk/zigbee2mqtt #7785: Crash when calling remove_all

YodaDaCoda, 2021-06-14:
Running z2m v1.19.1

Following the [groups](https://www.zigbee2mqtt.io/information/groups.html) documentation to call `zigbee2mqtt/bridge/request/group/members/remove_all` with a device specified results in a traceback and z2m restarts. This occurs whether or not a group is specified in the payload (docs specifically say it can be excluded for `remove_all` but I checked with and without).

Pastebin of log output: https://pastebin.ubuntu.com/p/3nPBnMmcsm/

Koenkk, 2021-06-14:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

YodaDaCoda, 2021-07-15:
@Koenkk Sorry, I lost track of this. Was it included in the v1.20 release or do I still need to switch to the dev branch to test?

Koenkk, 2021-07-15:
It is in 1.20.0 indeed

closed 2021-07-23 by github-actions[bot]