Koenkk/zigbee2mqtt #7786: Group Binding with Tradfri E1743 ON/Off Switch

BremerButjer, 2021-06-14:
# Bug Report
## What happened
Try to Switch (ON/OFF) some plugs via dircet group binding descriped here:
https://www.zigbee2mqtt.io/devices/E1743.html
and here:
https://www.zigbee2mqtt.io/information/binding.html

If i press on/off on the remoteswitch, the state is published but no ONOff Command will send to the group.
Here is the Output:
Binding:
`Jun 14 15:11:08 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:11:08: Saving state to file /opt/zigbee2mqtt/data/state.json
Jun 14 15:11:55 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:11:55: Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"from": "Tradfri-Switch-B", "to": "Gartenpumpe"}'
Jun 14 15:11:55 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:11:55: binding cluster 'genOnOff' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:01 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:01: Successfully bound cluster 'genOnOff' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:01 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"Tradfri-Switch-B","to":"Gartenpumpe"},"type":"device_bind"}'
Jun 14 15:12:01 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:12:01: binding cluster 'genLevelCtrl' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:03 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:03: Successfully bound cluster 'genLevelCtrl' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:03 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"Tradfri-Switch-B","to":"Gartenpumpe"},"type":"device_bind"}'
Jun 14 15:12:03 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:12:03: binding cluster 'closuresWindowCovering' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:03 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:03: Successfully bound cluster 'closuresWindowCovering' from 'Tradfri-Switch-B' to 'Gartenpumpe'
Jun 14 15:12:04 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"closuresWindowCovering","from":"Tradfri-Switch-B","to":"Gartenpumpe"},"type":"device_bind"}'
Jun 14 15:12:04 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:12:04: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genOnOff","genLevelCtrl","closuresWindowCovering"],"failed":[],"from":"Tradfri-Switch-B","to":"Gartenpumpe"},"status":"ok"}'
`
Control:
`Jun 14 15:17:18 raspberrypi npm[1480]: Zigbee2MQTT:debug 2021-06-14 15:17:18: Received Zigbee message from 'Tradfri-Switch-B', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Jun 14 15:17:18 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:17:18: MQTT publish: topic 'zigbee2mqtt/Tradfri-Switch-B', payload '{"action":"off","battery":74,"linkquality":144}'
Jun 14 15:17:18 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:17:18: MQTT publish: topic 'zigbee2mqtt/Tradfri-Switch-B/battery', payload '74'
Jun 14 15:17:18 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:17:18: MQTT publish: topic 'zigbee2mqtt/Tradfri-Switch-B/action', payload 'off'
Jun 14 15:17:18 raspberrypi npm[1480]: Zigbee2MQTT:info  2021-06-14 15:17:18: MQTT publish: topic 'zigbee2mqtt/Tradfri-Switch-B/linkquality', payload '144'`

## What did you expect to happen
It should switch the plugs

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.19.0
Adapter hardware: CC2652R

ahemwe, 2021-06-17:
Did you run any firmware update on the remote? 
I run a E1743 bound to a LED1842G3 (by means of a group) which used to worked well for a long time.

This week, I've updated E1743's firmware to 2.2.010 (20190723) and even re-binding does not work any more.

Interestingly, I've another one of that remote which I did not update, but this shows the same firmware release and works well. I won't update that.

Might be an issue with the firmware.

BremerButjer, 2021-06-18:
> Did you run any firmware update on the remote?
> I run a E1743 bound to a LED1842G3 (by means of a group) which used to work well for a long time.
> 
> This week, I've updated E1743's firmware to 2.2.010 (20190723) and even re-binding does not work any more.
> 
> Interestingly, I've another one of that remote which I did not update, but this shows the same firmware release and works well. I won't update that.
> 
> Might be an issue with the firmware.

Yes, I updated the remote to 2.2.0.010   :-/

Maybe someone has the experience to investigate. 
My workarround: i build a simple rule in Openhab. Works the same way, but with the detour via the software.

ahemwe, 2021-06-18:
Thanks for your feedback.

I would be happy to provide some debug logs, but do not know how to do so.

deadly667, 2021-06-18:
Have the same problem. Update E1743 from 2.2.0.010 to 2.3.075 and now groups doesnt work. It worked on 2.2.0.010 and after update it do nothing. It is bound correctly and group is working when I publish mqtt message manually. 

[image]

Here is debug log. Nothing else pops up, no errors. 

Received Zigbee message from 'Hall remote Ikea', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea', payload '{"action":"on","battery":60,"click":"on","device":{"applicationVersion":33,"dateCode":"20210526","friendlyName":"Hall remote Ikea","hardwareVersion":1,"ieeeAddr":"0xccccccfffe9948b5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1743","networkAddress":10439,"powerSource":"Battery","softwareBuildID":"2.3.075","stackVersion":98,"type":"EndDevice","zclVersion":3},"last_seen":"2021-06-18T16:48:35+02:00","linkquality":54,"update":{"state":"idle"}}'
MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea/action', payload 'on'
MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea/click', payload 'on'

deadly667, 2021-06-18:
When I now compared with one E1743 which is still on 2.2.010 and working ok with groups here are differences in log:
Bedroom Matija Night Switch - 2.2.010
Hall remote Ikea - 2.3.075

Debug Received Zigbee message from 'Bedroom Matija Night Switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8
Debug Received Zigbee message from 'Hall remote Ikea', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

with new version of FW groupId is 0 instead 8 with old one and then group is not triggered with new one.

Here you can see logs for both old and new.

Debug Received Zigbee message from 'Bedroom Matija Night Switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 8
Info MQTT publish: topic 'zigbee2mqtt/Bedroom Matija Night Switch', payload '{"action":"on","battery":87,"click":"on","device":{"applicationVersion":33,"dateCode":"20190723","friendlyName":"Bedroom Matija Night Switch","hardwareVersion":1,"ieeeAddr":"0xccccccfffe972bba","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1743","networkAddress":29385,"powerSource":"Battery","softwareBuildID":"2.2.010","stackVersion":98,"type":"EndDevice","zclVersion":3},"last_seen":"2021-06-18T17:14:35+02:00","linkquality":48,"update":{"state":"available"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/Bedroom Matija Night Switch/action', payload 'on'
Info MQTT publish: topic 'zigbee2mqtt/Bedroom Matija Night Switch/click', payload 'on'
Debug Received Zigbee message from 'Bedroom Matija Night light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/group_matija_night_light', payload '{"brightness":112,"color":{"x":0.46,"y":0.195},"color_mode":"xy","color_temp":2,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/Bedroom Matija Night light', payload '{"brightness":112,"color":{"x":0.46,"y":0.195},"color_mode":"xy","color_temp":2,"device":{"applicationVersion":80,"dateCode":"","friendlyName":"Bedroom Matija Night light","hardwareVersion":1,"ieeeAddr":"0x588e81fffed6233a","manufacturerID":4098,"manufacturerName":"_TZ3000_odygigth","model":"HG06106B","networkAddress":24489,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2021-06-18T17:14:35+02:00","linkquality":51,"state":"ON"}'

Debug Received Zigbee message from 'Hall remote Ikea', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea', payload '{"action":"on","battery":60,"click":"on","device":{"applicationVersion":33,"dateCode":"20210526","friendlyName":"Hall remote Ikea","hardwareVersion":1,"ieeeAddr":"0xccccccfffe9948b5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1743","networkAddress":10439,"powerSource":"Battery","softwareBuildID":"2.3.075","stackVersion":98,"type":"EndDevice","zclVersion":3},"last_seen":"2021-06-18T17:14:57+02:00","linkquality":33,"update":{"state":"idle"}}'
Info MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea/action', payload 'on'
Info MQTT publish: topic 'zigbee2mqtt/Hall remote Ikea/click', payload 'on'

MattWestb, 2021-06-18:
All new controller firmware in the test feed with 2.3.075 have broken group binding but binding devices (that normally not working with IKEA devices) is working OK.
And its looks like IKEA have blocking firmware downgrading of there remotes so no easy way back.

Using the test feed is not recommended i one production stem and the firmware is not zigbee certified and its only waiting  IKEA is relenting one working updated so can getting the remotes working normal a gen.

ahemwe, 2021-06-19:
Thanks for the hint that binding on device level works  @MattWestb
This helped me a lot. Additionally, I switched off, the test firmware configuration. For some other ikea devices, I still see ota updates. Is there a way to flush update data after switching the test server off?

MattWestb, 2021-06-19:
In the normal feed its some firmware that looks being broken for some device that the system trying applying on the devices but its not working so i think its the best is ignoring it or disabling OTA.

And device binding of lights is not zigbee standard then its group binding so very likely its going back to normal later and you can only using group binding but its being one later problems ;-))

I have doing temporary bindings for the light in my bedroom for the new IKEA E14 WS2 lights that is controlled with E1743 remotes and its working OK.

closed 2021-07-28 by github-actions[bot]

kokangit, 2021-08-19:
I have the same problem with all 4 of my remotes (E1743). They all have firmware 2.2.010 from when I bought the, so I have not done any upgrade.
I have done unbind from default_bind_group and the bind to a new group where I have an IKEA bulb. But when pressing on or off it always only sends to groupID 0. The bulb does not get any trigger.
If I put the bulb into group 0 for testing, I can then control the bulb with the remote. But that is not a working solution, since other devices will also send to this group so the light will go on when a motion detector detects motion.
So this problem seems to be present even in firmware version 2.2.010.