Koenkk/zigbee2mqtt #7795: Tradfi get topics

louia, 2021-06-15:
## What happened

I cannot get the state of my bulb. 
My bulb name is "ampouleSalon"
I checked this topics :
- zigbee2mqtt/ampouleSalon/get/state
- zigbee2mqtt/ampouleSalon/state/get

While i can set the state brightness etc with this topic :
- zigbee2mqtt/ampouleSalon/set/state 

with payload "off" or "on"

the doc isn't clear.

sjorge, 2021-06-15:
What payload are you sending? Both of these work fine:
```
mosquitto_pub -h amethyst.acheron.be  -t zigbee2mqtt/light/office/main/panel1/get/state -m ''
mosquitto_pub -h amethyst.acheron.be  -t zigbee2mqtt/light/office/main/panel1/get -m '{"state": ""}'
```

louia, 2021-06-15:
I don’t understand why i need to publish to get state… I want to have the
state of my bulb when the state change. Imagine, i turn on the bulb via
mqtt publish. The topic « /get/state » tell me 0 or 1 (on or off). It’s
possible ?

sjorge, 2021-06-15:
Thats not how MQTT works, you publish something to `<dev>/get/state`, z2m will listen on those topics and if it sees something publish a message there, it will check with the device and publish like usual to `<dev>` or `<dev>/state` if you have attribute instead of json configured.

louia, 2021-06-17:
when i publish to  "zigbee2mqtt/ampouleSalon/get/state" with payload "on" nothing append but when i send the same payload to "zigbee2mqtt/ampouleSalon/set/state" the lamp turn on. Me, all i want to have is the topic that tell me "on or off", i want the tpoic that represent the state of my devices but the topic "zigbee2mqtt/ampouleSalon/get/state" or "zigbee2mqtt/ampouleSalon/get" doesnt works

sjorge, 2021-06-17:
Asking the bulb for it's state
```
[root@amethyst ~]# mosquitto_pub -t zigbee2mqtt/light/office/desk/bulb/get/state -m ''
```

```
Zigbee2MQTT:debug 2021-06-17 13:48:31: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/get/state' with data ''
Zigbee2MQTT:debug 2021-06-17 13:48:31: Publishing get 'get' 'state' to 'light/office/desk/bulb'
Zigbee2MQTT:debug 2021-06-17 13:48:31: Received Zigbee message from 'light/office/desk/bulb', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-06-17 13:48:32: MQTT publish: topic 'zigbee2mqtt/light/office/desk/group', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_temp":234,"state":"OFF"}'
Zigbee2MQTT:info  2021-06-17 13:48:32: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":234,"color_temp_startup":65535,"linkquality":90,"power_on_behavior":"off","state":"OFF"}'
```

Setting the state to 'ON'
```
[root@amethyst ~]# mosquitto_pub -t zigbee2mqtt/light/office/desk/bulb/set/state -m 'on'
```

```
Zigbee2MQTT:debug 2021-06-17 13:48:46: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/set/state' with data 'on'
Zigbee2MQTT:debug 2021-06-17 13:48:46: Publishing 'set' 'state' to 'light/office/desk/bulb'
Zigbee2MQTT:debug 2021-06-17 13:48:46: Received Zigbee message from 'light/office/desk/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-17 13:48:46: Received Zigbee message from 'light/office/desk/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-06-17 13:48:46: MQTT publish: topic 'zigbee2mqtt/light/office/desk/group', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_temp":234,"state":"ON"}'
Zigbee2MQTT:info  2021-06-17 13:48:46: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":234,"color_temp_startup":65535,"linkquality":90,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2021-06-17 13:48:46: MQTT publish: topic 'zigbee2mqtt/light/office/desk/group', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_temp":234,"state":"ON"}'
Zigbee2MQTT:info  2021-06-17 13:48:46: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":234,"color_temp_startup":65535,"linkquality":90,"power_on_behavior":"off","state":"ON"}'
```

Asking for the state again
```
[root@amethyst ~]# mosquitto_pub -t zigbee2mqtt/light/office/desk/bulb/get/state -m '
```

```
Zigbee2MQTT:debug 2021-06-17 13:48:51: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/get/state' with data ''
Zigbee2MQTT:debug 2021-06-17 13:48:51: Publishing get 'get' 'state' to 'light/office/desk/bulb'
Zigbee2MQTT:debug 2021-06-17 13:48:51: Received Zigbee message from 'light/office/desk/bulb', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-06-17 13:48:52: MQTT publish: topic 'zigbee2mqtt/light/office/desk/group', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_temp":234,"state":"ON"}'
Zigbee2MQTT:info  2021-06-17 13:48:52: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb', payload '{"brightness":254,"color":{"hue":42,"saturation":42,"x":0.3693,"y":0.3695},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":234,"color_temp_startup":65535,"linkquality":93,"power_on_behavior":"off","state":"ON"}'
```

Seems to be working fine, to get the state you send an empty payload to the get topic, it will replace on the normal topic. To set it, you send ON/OFF/toggle to the set topic, it will again respond on the normal topic.

You can also see the replies the bulb gives to z2m in the debug level, we know that z2m is reaching out to the bulb and it is replying with the actual state.

louia, 2021-06-18:
thank you so much for the help to understand the mqtt topic's and z2m.
It work's very well

closed 2021-07-12 by louia