Koenkk/zigbee2mqtt #7797: ID Lock 150 broken in DEV

TheStigh, 2021-06-15:
@Koenkk 

Current Dev ( `1.19.1-dev, commit #497ca18` ) has broken the integration. 

All previous binary_sensor's are now switch's. None of the switches updates its state.

[Pairing log](https://pastebin.com/RhhJTbEz)

Availability log for creating the switch:
```
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    }
  ],
  "command_topic": "zigbee2mqtt/0x680ae2fffe6bb522/set/master_pin_mode",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x680ae2fffe6bb522"
    ],
    "manufacturer": "Datek",
    "model": "Zigbee module for ID lock 150 (0402946)",
    "name": "0x680ae2fffe6bb522",
    "sw_version": "Zigbee2MQTT 1.19.1-dev"
  },
  "json_attributes_topic": "zigbee2mqtt/0x680ae2fffe6bb522",
  "name": "0x680ae2fffe6bb522 master_pin_mode",
  "payload_off": "false",
  "payload_on": "true",
  "state_topic": "zigbee2mqtt/0x680ae2fffe6bb522",
  "unique_id": "0x680ae2fffe6bb522_switch_master_pin_mode_zigbee2mqtt",
  "value_template": "{{ value_json.master_pin_mode }}"
}
```

Z2M Release 1.19.1-1 :
[image]

Z2M Dev 1.19.1-dev, commit #497ca18 :
[image]

Koenkk, 2021-06-15:
@robertalexa I think this happens because of  https://github.com/Koenkk/zigbee2mqtt/pull/7760/files#diff-5cd90a075b18a077fab2f7f5a4066f63eade9d49e5ef0b023bc3878272786e9eR361 . I read your rationale in https://github.com/Koenkk/zigbee2mqtt/pull/7760 but I think e.g. `"payload_off": "false",` should be `"payload_off": false,` instead

robertalexa, 2021-06-15:
@Koenkk I agree with your logic, and when i did my tests both (string) and (bool) values sent commands successfully, with the workaround in place.

What i did see is that boolean values will get converted to strings:
true -> "True"
false -> "False"

Mqtt documentation states the following:
[image]

So having values as boolean in z2m will not play ball with that, cause they will be converted to strings, but when they come back as strings things are not good anymore.

I would be more than happy to make changes to my code, but i have no freaking idea how to avoid the issue above.

Is there like a high up level in toZigbee to check payloads if they are equal to "True" to convert it to (bool) true? Somewhere that is not device specific? I would really appreciate your input on this, i do not mind doing the work and the testing.

Rob

TheStigh, 2021-06-15:
@robertalexa Sure, here's the log.

In this case, the actual state of attribute  `"relock_enabled":false` and trying to enable by activating the switch:

```
Zigbee2MQTT:debug 15-06-2021 20:18:18: Received MQTT message on 'zigbee2mqtt/0x680ae2fffe6bb522/set/relock_enabled' with data 'true'
Zigbee2MQTT:debug 15-06-2021 20:18:18: Publishing 'set' 'relock_enabled' to '0x680ae2fffe6bb522'
Zigbee2MQTT:info  15-06-2021 20:18:19: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6bb522', payload '{"battery":127.5,"door_state":"open","idlock_lock_fw":"1.5.7","linkquality":26,"lock_mode":"auto_off_away_off","lock_state":"unlocked","master_pin_mode":true,"relock_enabled":true,"rfid_enable":true,"service_mode":"deactivated","sound_volume":"high_volume","state":"UNLOCK"}'
Zigbee2MQTT:debug 15-06-2021 20:18:19: Received Zigbee message from '0x680ae2fffe6bb522', type 'attributeReport', cluster 'closuresDoorLock', data '{"16389":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  15-06-2021 20:18:19: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe6bb522', payload '{"battery":127.5,"door_state":"open","idlock_lock_fw":"1.5.7","linkquality":26,"lock_mode":"auto_off_away_off","lock_state":"unlocked","master_pin_mode":true,"relock_enabled":true,"rfid_enable":true,"service_mode":"deactivated","sound_volume":"high_volume","state":"UNLOCK"}'
```
It does change attribute `relock_enabled`  to `true` and the lock are updated but as the attribute doesn't update the `switch` value (still remain `false`), the switch cannot be used anymore.

robertalexa, 2021-06-16:
@TheStigh

Apologies for my late reply.

The fact that the payload gets "interpreted" correctly kinda shows that the PR https://github.com/Koenkk/zigbee2mqtt/pull/7760 does what it is meant to do. The payloads in my testing looked exactly the same.

It makes me wonder if the problem sits somewhat on HA side.

Can you please confirm that you have tried re-pairing that device? Actually, to clarify, you have removed the device in z2m, in turns that removes the device in HA. Then you re-pair the device in z2m, that creates your entity in HA.

I just wonder if this is some weird leftover on HAs side purely because the entity was a different type before.

Let me know about the re-pairing please and we will go from there.

PS: i know it might be a nuissance to do that, but if it works, the long term benefit is greater than the short term inconveniece.

Rob

TheStigh, 2021-06-16:
@robertalexa 

I do have two of these, one in production and one in dev.
Production are latest release and working as expected at these entities are `binary_sensor`

Updated Dev the other day and realized that buttons created for masterpin, rfid and relock didn't show the state.
After investigations, I did see that these earlier binary_sensor's was now swtiches.

My procedure when testing abnormalities:
- Stopping Z2M
- Deleting database.db and state.json
- Using MQTT Explorer and deletes all topics related to the device
- Resetting device to factory defaults
- Starting Z2M
- Pairing

In this case, I did the procedure above, though switch to Z2M Prod, came with binary_sensors. Did the procedure again, but reverted back to latest Z2M Dev, now as switches.

Issue are that the switch never update the state from the actual corresponding state of the lock.
If the state of the attribute are off/false and I use the switch to turn ON, it does send correctly and the attribute does change to on/true on the lock itself, but the switch never updates to the new state. Like it doesn't read the json returned.

As this change from binary_sensor to switch are not done by me, I don't know where to start regarding troubleshooting.

robertalexa, 2021-06-16:
Hey dude, thanks for confirming.

However nowhere in your steps you have not mentioned deleting the HA entity after removing it from z2m. My point was the HA entity might have broken leftovers.

For example you could try the following, probably much simplier process.
1. Access the z2m Frontend
2. Delete the device. This will delete it from z2m and HA
3. Allow Join
4. Pair
5. Check HA again

I have done the switch changes, which is why I am trying to figure where the issue is, and if there is one, how to correct it.

PS: your approach sounds good, but before doing the Pairing step you would need to go to HA and delete the entity for that device.

Rob

TheStigh, 2021-06-16:
Bro, I actually "normally" do what you said, but in this case I will do these steps again just to make sure that all are done per your request.

robertalexa, 2021-06-16:
"Bro" the reason i am asking for this is that the issue you described is in HA and not z2m.

Yet your re-pairing process is all about z2m and nothing handling the leftover entity in HA.

I am not trying to say that this will fix you issue. I am trying to understand the behaviour so i know if i actually need to fix the code, and implicitly your issue :)

But i can't begin fixing something when it all behaves correctly at my end.

Let me know how it goes and i would be more than happy to try to investigate further :)

Koenkk, 2021-06-16:
I've just checked with my Xiaomi plug (HA 2021.6.5) which exposes power outage memory (binary with `true` and `false`). The switch never works correctly in both scenarios (`"payload_off": "false",` or with  `"payload_off": false`). The issue here is that HA expects string values while z2m sends boolean values. HA doesn't support the boolean values. 

With (current implementation)

```
                                payload_on: typeof expose.value_on == 'boolean' ? 'true' : expose.value_on,
                                payload_off: typeof expose.value_off == 'boolean' ? 'false' : expose.value_off,
```
 
the switch always stays off in Home Assistant 

With:

```
                                payload_on: expose.value_on,
                                payload_off: expose.value_off,
```

Home Assistant sends a string (`True`/`False`), which makes z2m return `True`/`False` as string making it appear to work correctly, but it will actually always stick in the true state.

@robertalexa @TheStigh since this currently doesn't work correctly for the binary exposes having a boolean value I've disabled it for those ()

@emontnemery do you know what is the reason for HA not supporting `payload_on`/`payload_off` being a boolean for switches?

robertalexa, 2021-06-16:
@Koenkk Your findings make absolute sense, this is exactly what I have discovered and how @sjorge and I ended up putting those changes together.

The only difference is, in my testing, my HA switches actually work and stay in the correct states..... how? I am not sure, maybe it relates to the reporting specific to each device? In my case, I have used the Hive TRV to test - this is the only device I have that has switches (binaries that should be switches)

So as far as my testing involved, I was seeing the expected results.

I agree that all this can be avoided by allowing boolean in the MQTT payload_on and payload_off.

@Koenkk can you think of a workaround so that we can still fully support switches?

Rob

Koenkk, 2021-06-16:
> @Koenkk can you think of a workaround so that we can still fully support switches?

This should be fixed in HA I believe, thats why  I pinged @emontnemery

TheStigh, 2021-06-16:
Hi Robert, 

Ok, so @Koenkk  confirmed the same behavior. If you later want me to do tests, do not hesitate to contact me :)

robertalexa, 2021-06-16:
Thank you Stigh.

Let's see what  Erik comes back with and we go from there.

emontnemery, 2021-06-17:
@Koenkk What do you expect to be changed in Home Assistant?
`payload_on` and `payload_off` must be strings as per the documentation. Also, MQTT only supports sending and receiving raw bytes, not abstract types like a `bool`.

I think the confusion arises because this configuration: `"payload_off": false` is valid because the boolean will be coerced to a string, and that string representation is what received payload will be compared against.

robertalexa, 2021-06-18:
@Koenkk Given the above, which is somewhat expected, is there like a high up level in toZigbee to check payloads if they are equal to "True" to convert it to (bool) true? Somewhere that is not device specific?

emontnemery, 2021-06-18:
What you could do in zigbee2mqtt is perhaps to try to parse payloads as JSON, and also to send data after JSON serialization; a naked bool is valid JSON.

This Home Assistant configuration should then work:
```
  "payload_off": "false",
  "payload_on": "true",
```

robertalexa, 2021-06-18:
@emontnemery You would think that will do the trick, that is what I thought in the PR :))) But it seems like something is unhappy. Have a look at Koen's comment above https://github.com/Koenkk/zigbee2mqtt/issues/7797#issuecomment-862470950 describing the current code and the behaviour after that

emontnemery, 2021-06-18:
It's not very clear to me what Koen means with this:

> The issue here is that HA expects string values while z2m sends boolean values. HA doesn't support the boolean values.

How can z2m send a boolean value, MQTT only supports sending raw bytes?

It would be nice to get logs from `mosquitto_sub` or something like that showing exactly what is sent and received?

Koenkk, 2021-06-18:
@emontnemery currently when the `payload_off` is set to `false` (boolean), Home Assistant will send `False` when controlling the device (I guess the Python bool to string conversion). If it would send `false` instead this problem would be fixed.

> How can z2m send a boolean value, MQTT only supports sending raw bytes?

The published payloads of z2m are JSON, so they can contains types (boolean, string, number, etc.). e.g. a payload could be `{"occupancy": true}`. This already works well with Home Assistant with the following discovery payload https://github.com/Koenkk/zigbee2mqtt/blob/9bd46932c365e886ed0b1f9b8740d8cce632230c/lib/extension/homeassistant.js#L349 (in such case `payload_on` and `payload_off` are booleans).

emontnemery, 2021-06-18:
> currently when the payload_off is set to false (boolean), Home Assistant will send False when controlling the device (I guess the Python bool to string conversion). If it would send false instead this problem would be fixed (since z2m will auto convert this to a boolean)

Yes, and that's because HA allow non-strings in the configuration which are silently converted to strings. I think that should not be allowed at all because it is extremely confusing (as proven by this issue).
If we make some exception where a `bool` is stringified to either `true` or `false` instead of `True` and `False`, that just things even more confusing IMO.

This value template is not correct if the data is a JSON boolean and you want to compare it with strings `"true"` and `"false"`; the output of the template will be either `"True"` or `"False"`:
```
 value_template: `{{ value_json.${expose.property} }}`,
```

You need to modify the template to make sure the output is `"true"` or`"false"`:
```
 value_template: `{% if value_json.${expose.property} %} true {% else %} false {% endif %}`,
```

Koenkk, 2021-06-18:
The `value_template` was the missing piece of the puzzle, with this change it works: https://github.com/Koenkk/zigbee2mqtt/commit/91c16812dfc09edac37210a3e144d01347152bea
- `payload_off`/`payload_on` will always be a string (as HA requires)
- HA will send `true`/`false` to control
- Will allow the value being a boolean (since it converters it using the `value_template`)

Thanks again for the help @emontnemery !

closed 2021-06-18 by Koenkk

robertalexa, 2021-06-18:
Despite not me doing any work, thank you all for your help!

@TheStigh For the sake of everyone's sanity, would you be kind to test Koen's latest commit on the dev branch? Not that i don't trust his testing, but 2 tests are better than one.

TheStigh, 2021-06-18:
@robertalexa  Haha, sure I will, later today if it is ok.