Koenkk/zigbee2mqtt #7800: Availability changes don't change group state in HomeAssistant

bogdanm, 2021-06-15:
## What happened

I enabled availability (https://www.zigbee2mqtt.io/information/availability.html) in zigbee2mqtt in order to synchronize the state of some Ikea bulbs that are turned on/off from "dumb" light switches. That seems to work fine for individual bulbs. However, the same lights are part of a group in zigbee2mqtt. These groups are added manually to HomeAssistant using the instructions at https://www.zigbee2mqtt.io/integration/home_assistant.html#groups. The problem is that while the lights do change state from ON to OFF in HomeAssistant, the same thing doesn't happen with the corresponding group, which stays ON even after all the bulbs are detected as OFF.

## Debug info
Zigbee2MQTT version: version 1.19.1 (commit #9bd4693)
Adapter hardware: zzh! CC2652R Multiprotocol RF Stick
Adapter firmware version: `{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}`

robertalexa, 2021-06-16:
Hey Bogdan,

I am not proficient in the groups feature but here are my thoughts, as I feel you might be mis-interpreting the purpose of the groups.

1. Availability has nothing to do with states, or groups. Availability just pings devices regularly so you know if they are still alive or disconnected. This way if a device gets flagged as unavailable you can have an automation sending you a phone notification to go investigate.

2. I think groups of devices, and especially how they are created in HA, are not aware of individual devices in the group. Instead, I think the theory is that you "turn 3 bulbs into 1". So you forget about individual control for each bulb, but instead they all operate as one single one, thus the LIGHT entity in HA (which is not a group, is a single light).

To put that into context, when you press a button in your house, or an automation, you turn the LIGHT entity in HA on - which corresponds to a group of devices in z2m, which will turn on all 3 light bulbs.

By the sounds of your expectations, if you want both individual control and multiple-bulbs-at-once you need to rethink your automations. Forget about the group and the HA light entity created for the group, and instead you create automations that will action all 3 bulb individually, but at the same time.

I hope this helps and clarifies some bits for you. I do not think that the issue you have described is an issue with the software, but rather a misundertanding of its use case.

Let me know if I can help you further with achieving your goal, i do not mind sharing ideas if you need them

bogdanm, 2021-06-16:
Hi @robertalexa, thanks for your comments. I mostly agree with your comments, but still I think that there's a disconnect somewhere, although at this point I'm not sure that the problem is in zigbee2mqtt anymore.

To clarify, I am using "proper" Zigbee groups created in zigbee2mqtt, not HomeAssistant groups (which are logical groups). I do not want to control the bulbs individually (even the physical switches turn them all on or all off at the same time), I just have them individually listed in HomeAssistant as a result of the MQTT discovery process. I added manually the Zigbee light groups in HomeAssistant, as explained in my initial post.

While I understand that availability isn't directly related to groups, the disconnect is this in my view:

- I turn the lights on from the physical switch and I see them as ON in the HomeAssistant interface. The GUI element that corresponds to the  Zigbee group in which all the bulbs belong also appears as ON.
- I turn off the lights from the physical switch.

The availability feature works fine and after a while it detects that the bulbs are no longer reachable. At this point, the individual bulbs appear as OFF in the HomeAssistant interface. However, the GUI element that corresponds to the  Zigbee group in which all the bulbs belong still appears as ON, and I don't think this should happen. At this point I can't tell if the issue is in zigbee2mqtt or in HomeAssistant.

closed 2021-07-24 by github-actions[bot]

cirdanmiriel, 2022-02-25:
I am facing the same problem here: I defined a group of 3 bulbs in zigbee2mqtt and if i physically switch them off, each of them appears unavailable after a timeout (as wanted) but the group continues to display the last current state.