Koenkk/zigbee2mqtt #7802: homeassistant_legacy_entity_attributes config not working

stickpin, 2021-06-15:
## What happened

I am running the latest dev build.

I've set the config to remove legacy attributes based on this PR (https://github.com/Koenkk/zigbee2mqtt/pull/7683) but I don't see any change on the Home Assistant side.

`homeassistant_legacy_entity_attributes: false`

All attributes are still present.

Any idea what can be the issue?

FYI @frenck

frenck, 2021-06-15:
@stickpin You have to restart Home Assistant for it to take effect. Once the state attribute are in HA, it will hold on to it in the state machine.

So:
- Set: `homeassistant_legacy_entity_attributes: false`
- Restart z2m
- Restart Home Assistant

stickpin, 2021-06-15:
@frenck thanks a lot for a quick reply.
I was missing the Restart Home Assistant part. :)

Now it's perfect!
Thanks a lot for this PR! :)

Maybe it worth mentioning in the documentation that after the change also Home Assistant has to be restarted.

closed 2021-06-15 by stickpin

frenck, 2021-06-15:
Good idea, @stickpin: Koenkk/zigbee2mqtt.io#795