Koenkk/zigbee2mqtt #7806: Failed to interview,device has not successfully been paired, shows no ID while sending message

tchnk, 2021-06-16:
# Bug Report
## What happened
I get the error "Failed to interview '0x00124b001ba3fb87', device has not successfully been paired" i followed your instructions by replacing the battery multiply times and go near the CC2531.
The Device i want to pair is a Tuya Door/Window Sensor like this one: https://de.aliexpress.com/item/1005002512326796.html?spm=a2g0o.productlist.0.0.43a110d9RSwsbb&algo_pvid=edabb207-8402-430a-bb91-14581cf6d423&algo_exp_id=edabb207-8402-430a-bb91-14581cf6d423-6

I think it is strange that it shows a device ID while trying to pair, but after failed pairing i can toggle the device and it shows "Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'".

Debug protocol: https://hastebin.com/lokipezani.yaml

## What did you expect to happen
I expected that the device would pair with the CC2531 and my Raspberry Pi 1b

## How to reproduce it (minimal and precise)
With the latest firmeware and Zigbee2mqtt Version try to pair the Door Sensor by presssing its pairing-/reset-Button for 6 seconds like it says in the manual.

## Debug Info
Zigbee2MQTT version: 1.19.1 (commit #9bd46932)
Adapter hardware: CC2531
Adapter firmware version: latest
Debug protocol: https://hastebin.com/lokipezani.yaml

karolVimn, 2021-07-05:
I also have this issue. 
Zigbee2MQTT version: 1.20.0-1
Adapter: Conbee2
Adapter firmware: latest

Device model: Tuya ZM-CG205 bought here: https://www.aliexpress.com/item/1005002565444440.html?spm=a2g0s.9042311.0.0.2fee4c4dSGwhhp

elgerg, 2021-07-28:
I also have the save device:
ZM-CG205
My issue was closed last week :(
https://github.com/Koenkk/zigbee2mqtt/issues/7503

Koenkk, 2021-07-28:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7553 , issue should be fixed in the latest-dev.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-07-28 by Koenkk