Koenkk/zigbee2mqtt #7809: Fantem ZB003-X some sensors are missing

serviceiotcc, 2021-06-16:
## What happened
After the [sensor](https://www.zigbee2mqtt.io/devices/ZB003-X.html) is connected to the network, there is no information from many entities. The only thing that works is motion detection and signal strength.
## What did you expect to happen
Data from all sensors (temperature, illumination, humidity, and so on)
## How to reproduce it (minimal and precise)
I have tried several times to remove and add the sensor to the network, at a distance of a meter from the controller, but it does not help.
## Debug info
Zigbee2MQTT version: 1.18.3-1
Adapter hardware: CC2652

[image]

robertalexa, 2021-06-17:
@serviceiotcc 
Hi, I do not own this device so can't guarantee my suggestion will help you, but i think it might.

The documentation show that the device exposes the following:
- Reporting_time
- Reporting_enable
- Pir_enable
- Led_enable

If I understand the documentation correctly, it might by that the device does not have all this things enabled by default, thus explaining why no values are being set.

Have a look at the documentation to see what mqtt command you need to send for each of those exposed items.

While setting those values i suggest you have a look at the z2m log to see what happens.

Seeing that you use HA, you can send MQTT commands from HA by going to Configuration -> Integrations -> MQTT (press configure)

An example would be:
topic: zigbee2mqtt/FRIENDLY_NAME/set
payload: {"reporting_enable": true}

Remember to replace FRIENDLY_NAME with your device.

Hope this helps

serviceiotcc, 2021-06-22:
Many thanks! It all worked!

robertalexa, 2021-06-22:
Happy to hear

closed 2021-12-06 by github-actions[bot]