Koenkk/zigbee2mqtt #7816: Lights in group turn on immediately after turning off

denysdovhan, 2021-06-18:
## What happened

I have [4 Aqara Bulbs](https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html) joined in a group called `main_light`. Sometimes, when I turn off lights they turn off, but immediately turn on after.

## What did you expect to happen

I expect the light to turn off when I turn them off, without turning on again.

## How to reproduce it (minimal and precise)

I didn't find a reliable way to reproduce this problem. It just happens from time to time, usually, after lights weren't turned on for a long time. Other bulbs that aren't joined into groups work fine. 

Here is a video of how this happens: https://user-images.githubusercontent.com/3459374/122544137-2b1b8500-d035-11eb-8dbf-1a1cf9bb2c4c.MOV

My group:

```yaml
'1':
  friendly_name: living_room_main_light
  transition: 2
  devices:
    - 0x00158d0002d3c856/1
    - 0x00158d000450cca4/1
    - 0x00158d00045ec07b/1
    - 0x00158d0002d71dab/1

```

## Debug info

Zigbee2MQTT version: 1.19.1 (commit #9bd4693)
Adapter hardware: CC2531
Adapter firmware version: `{"meta: 
{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}`

My setup is described here: https://denysdovhan.com/smart-home/hardware/#zigbee-network
docker-compose is here: https://github.com/denysdovhan/smart-home/blob/master/docker-compose.yaml

Koenkk, 2021-06-18:
Please provide the debug log of the moment this happens.

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

denysdovhan, 2021-06-25:
I have finally managed to catch this situation once again. Here's the log: https://pastebin.com/ynVdH2KC

denysdovhan, 2021-06-25:
And here it happened again: https://pastebin.com/K1XZ0Hrw

Koenkk, 2021-06-26:
@denysdovhan if you look at your log you see the following, first the group is turned off:

```
debug 2021-06-25 22:12:57: Received MQTT message on 'zigbee2mqtt/living_room_main_light/set' with data '{"state": "OFF"}'
```

However just 8 seconds later another command is received to turn the lights on:

```
debug 2021-06-25 22:13:05: Received MQTT message on 'zigbee2mqtt/living_room_main_light/set' with data '{"state": "ON", "transition": 1.0, "color_temp": 500}'
```

You have to figure out who and why a command turn on command is send to z2m, this doesn't look like a z2m bug.

frenck, 2021-07-21:
Got the logs upstream now as well :)

> However just 8 seconds later another command is received to turn the lights on

true!

Here is all MQTT communication:

```
2021-07-21 21:59:50 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on zigbee2mqtt/living_room_main_light/set: '{"state": "OFF"}', mid: 246
2021-07-21 21:59:50 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/living_room_main_light: b'{"brightness":0,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"state":"ON"}'
2021-07-21 21:59:50 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/living_room_main_light: b'{"brightness":0,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"state":"OFF"}'
2021-07-21 21:59:56 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/living_room_main_light: b'{"brightness":251,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"state":"OFF"}'
2021-07-21 21:59:57 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on zigbee2mqtt/living_room_main_light: b'{"brightness":251,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"state":"ON"}'
2021-07-21 21:59:57 DEBUG (MainThread) [homeassistant.components.mqtt] Transmitting message on zigbee2mqtt/living_room_main_light/set: '{"state": "ON", "transition": 1.0, "color_temp": 500}', mid: 247
```

The last Transmit is caused by the message received just above that, with an "ON" state for this group, 7 seconds after the original "OFF" request was sent.

Because an "ON" message was received, it kicks an turn_on sequence, which includes all listeners, like MQTT, to which Home Assistant indeed will send out its status change.

The message you are pointing out, is not a cause, its a result of the state change received.

frenck, 2021-07-21:
> Here is a video of how this happens: user-images.githubusercontent.com/3459374/122544137-2b1b8500-d035-11eb-8dbf-1a1cf9bb2c4c.MOV

This is a bit of a misleading video, mainly because the UI accepts the input and checks a bit later to request the actual state, since at that point not everything has been confirmed off (that happens a couple of seconds later according to logs), it switches back.

While maybe not nice, it is not showing the actual problem, and thus the video is a tiny bit misleading for that matter.

denysdovhan, 2021-07-28:
@Koenkk seems like this is actually a Z2M problem. Can I help somehow to resolve the issue?

Koenkk, 2021-07-28:
State `ON` is published for the group because one off the bulbs in the group (`living_room_main_light_bulb_2`) reports itself as being on: (first message, looks to be a Xiaomi bulb where `100` indicates the state of the bulb: https://github.com/Koenkk/zigbee-herdsman-converters/blob/dc7bbdc1414c82bf270566628622ea70ec5966a3/converters/fromZigbee.js#L5119)

```
debug 2021-06-25 22:13:05: Received Zigbee message from 'living_room_main_light_bulb_2', type 'attributeReport', cluster 'genBasic', data '{"65281":{"10":3955,"100":1,"101":251,"102":500,"3":101,"5":542,"7":"0x0000000000000000","8":279,"9":1281}}' from endpoint 1 with groupID 0
info  2021-06-25 22:13:05: MQTT publish: topic 'zigbee2mqtt/living_room_main_light', payload '{"brightness":251,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"state":"ON"}'
info  2021-06-25 22:13:05: MQTT publish: topic 'zigbee2mqtt/living_room_main_light_bulb_2', payload '{"brightness":251,"color":{"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_temp":500,"linkquality":23,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

The default behaviour is that if one of the devices in the group turns `ON`, the whole group is assumed to be `ON`. See https://www.zigbee2mqtt.io/information/groups.html#state-changes for more details (and how to disable this). 

Next question, why does the Xiaomi bulb reports itself as `ON`, looking at the logs just before the `ON` report I see: 

```
debug 2021-06-25 22:13:05: Device 'living_room_main_light_bulb_2' announced itself
```

A device announce is typically send when a device boots (e.g. when device is connected to AC again or when the firmware of the device crashes). I guess the default behaviour for this bulb is to turn on when it boots, so the report of the ON state looks valid. Some Xiaomi bulbs support `power_outage_memory` letting you control the behaviour on boot, see https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html#power_outage_memory-binary 

@frenck I still find it strange that Home Assistant sends a turn on command when it receives a state update (but that is not relevant for this issue).

denysdovhan, 2021-08-31:
I've been testing possible solutions with `transition: 0` and `power_outage_memory: true`, but the problem still persists.

Though `transition: 0` made it happen a bit rarely.

denysdovhan, 2021-10-04:
Still unresolved.

denysdovhan, 2021-11-06:
Still not resolved.

closed 2021-11-14 by github-actions[bot]