Koenkk/zigbee2mqtt #7818: RGBgenie ZB-5001 resubmit from stale issue

drhaber, 2021-06-18:
This remote a had a previous attempt to be added but the request has since [gone stale](https://github.com/Koenkk/zigbee2mqtt/issues/6980).

drhaber, 2021-07-22:
I have gotten ZB-5001 to work by duplicating the [existing RBG-5004](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/rgb_genie.js)

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RGBgenie ZB-5001'],
    model: 'ZB-5001',
    vendor: 'RGB Genie',
    description: 'Zigbee 3.0 remote control',
    fromZigbee: [fz.command_recall, fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery],
    exposes: [e.battery(), e.action(['recall_*', 'on', 'off', 'brightness_stop', 'brightness_move_up', 'brightness_move_down'])],
    toZigbee: [],
};

module.exports = definition;    `

Koenkk, 2021-07-24:
Added, thanks!

closed 2021-07-24 by Koenkk