Koenkk/zigbee2mqtt #7819: Cannot add BTicino F20T60A

Cr4z33, 2021-06-18:
I already tried two same BTicino modules (in case the first one was faulty).

I also tried running Zigbee2MQTT (in a Docker) on two different hardwares: a Synology DS918+ and a computer with Ubuntu 20.10.

The USB dongle is a Texas Instruments TI CC2531 with proper flashed firmware.

The problem is I cannot add this module and I keep getting the same error on both hardwares:

`Zigbee2MQTT:info  2021-06-18 18:31:26: Logging to console and directory: '/app/data/log/2021-06-18.18-31-26' filename: log.txt
Zigbee2MQTT:info  2021-06-18 18:31:26: Starting Zigbee2MQTT version 1.19.1 (commit #9bd4693)
Zigbee2MQTT:info  2021-06-18 18:31:26: Starting zigbee-herdsman (0.13.111)
Zigbee2MQTT:info  2021-06-18 18:31:29: zigbee-herdsman started
Zigbee2MQTT:info  2021-06-18 18:31:29: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2021-06-18 18:31:29: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-06-18 18:31:29: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-06-18 18:31:29: Allowing new devices to join.
Zigbee2MQTT:warn  2021-06-18 18:31:29: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-06-18 18:31:29: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-06-18 18:31:29: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:info  2021-06-18 18:31:29: Connected to MQTT server
Zigbee2MQTT:info  2021-06-18 18:31:29: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-06-18 18:31:29: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"9bd4693","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0x00124b001cd4b858","panID":6754},"permit_join":true,"version":"1.19.1"}'
Zigbee2MQTT:info  2021-06-18 18:32:50: Device '0x000474000089a3e4' joined
Zigbee2MQTT:info  2021-06-18 18:32:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000089a3e4","ieee_address":"0x000474000089a3e4"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-06-18 18:32:50: Starting interview of '0x000474000089a3e4'
Zigbee2MQTT:info  2021-06-18 18:32:50: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000474000089a3e4","ieee_address":"0x000474000089a3e4","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-06-18 18:32:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000474000089a3e4"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-06-18 18:32:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000474000089a3e4"},"type":"pairing"}'
Zigbee2MQTT:error 2021-06-18 18:35:12: Failed to interview '0x000474000089a3e4', device has not successfully been paired`

jd766, 2021-06-21:
Bticino devices are a bit difficult to pair at times, I have the same power meter and it took many retries to pair it.   
One thing that seems to help with all bticino devices is doing a power cycle after the pairing attempt.

closed 2021-07-29 by github-actions[bot]

fred-c1, 2022-08-27:
This helped me:
- This device ONLY works on channel 11  => shame on Legrand / BTicino
- If the paring process does not start (no pop up messages on zigbee2mqtt): Restart the coordinator (for example, unplug + plug the USB dongle / coordinator)
- If the device leave the zigbee network just after pairing/joining: Press the botton on the device during the pairing process at a rate of about 1 per second