Koenkk/zigbee2mqtt #7824: Still showing " Template variable warning " in HA with Z2M 1.19.1

whxciotw, 2021-06-19:
## What happened
HA still shows the warning about " Template variable warning"
HA version: 2021.6.5
> 2021-06-19 12:06:05 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'update_available' when rendering '{{ value_json.update_available}}'
2021-06-19 12:06:06 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC652

Koenkk, 2021-06-20:
`update_available` and `click` are legacy, you can remove them by disabling the legacy api, add to your `configuration.yaml`.

```yaml
advanced:
  legacy_api: false
```

closed 2021-06-20 by Koenkk

whxciotw, 2021-06-21:
Thanks for your reply.

I have added to the configuration as below
```
advanced:
  legacy_api: false
```
and then restart Z2M and HA, the warning of `update_available` will not show any more. 

But the warning of `click` still shows as below
`Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'`

Koenkk, 2021-06-22:
@whxciotw try also adding:

```
device_options:
  legacy: false
```

whxciotw, 2021-06-23:
> @whxciotw try also adding:
> 
> ```
> device_options:
>   legacy: false
> ```

still showing the `Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'` after adding to device_option

whxciotw, 2021-06-23:
> @whxciotw try also adding:
> 
> ```
> device_options:
>   legacy: false
> ```

Thanks, now the warnings didn't show any more after adding to device_option.

Stooovie, 2021-10-31:
Still an issue here in Z2M 1.21.2-1  and HA 2021.10.6, with no solution on my system, see https://github.com/Koenkk/zigbee2mqtt/issues/9433

Stooovie, 2021-10-31:
> > @whxciotw try also adding:
> > ```
> > device_options:
> >   legacy: false
> > ```
> 
> Thanks, now the warnings didn't show any more after adding to device_option.

When I do this, Z2M stops communicating with HA altogether and only can be revived by restoring a HA snapshot.