Koenkk/zigbee2mqtt #7825: Bound Gledopto controller not polled for updates

malaakso, 2021-06-19:
## What happened
Created a group with Gledopto GL-C-008-1ID and bound IKEA E1743 OnOff and LevelCtrl cluster to the group. Turned light ON with remote, but light state in zigbee2mqtt stays OFF (light is ON).

## What did you expect to happen
I expected zigbee2mqtt to poll light for status when receiving command from remote and show state as ON.

## How to reproduce it (minimal and precise)
See above.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC2652RB
Adapter firmware version: 20210430

malaakso, 2021-07-20:
Not stale. Still same behavior with 1.20.0.

elgerg, 2021-07-21:
I had a similar problem with one of the LED controllers and associated 2.4ghz remote.

I fixed it by adding the following reporting options:
[image]

Doing something similar might work for you?

malaakso, 2021-07-21:
> I had a similar problem with one of the LED controllers and associated 2.4ghz remote.
> 
> I fixed it by adding the following reporting options:
> [image]
> 
> Doing something similar might work for you?

Thanks for the suggestion, but I forgot to mention in the first post that my LED controller does not support reporting. Per documentation, when binding remote to a group polling should be automatically performed for those devices that don't support reporting. But that doesn't seem to be working.

malaakso, 2021-08-09:
I repeated the test with a Philips Hue bulb, that also doesn't support reporting. With that, polling works. So for some reason the Gledopto controller is handled differently.

malaakso, 2021-09-09:
I can see that https://github.com/Koenkk/zigbee2mqtt/blob/4a51e0c0c99dfeb4b319048db270c9eb69db419e/lib/extension/bind.js#L84 should handle Gledopto controllers, but that is based on ZDO manufacturer ID, which I'm not sure how to check what my device reports. @Koenkk any advice?

Koenkk, 2021-09-10:
@malaakso you can check the data/database.db `manufacturerID`, this one should match https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/manufacturerCode.ts#L978 (`0x124F` or `4687`)

malaakso, 2021-09-12:
> @malaakso you can check the data/database.db `manufacturerID`, this one should match https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/manufacturerCode.ts#L978 (`0x124F` or `4687`)

Thanks @Koenkk !

`{"id":11,"type":"Router","ieeeAddr":"0x00124b0021f09015","nwkAddr":54945,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-008","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-008","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"3.0.1"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":155,"colorTempPhysicalMax":495,"colorTemperature":155,"currentX":45940,"currentY":19595,"enhancedCurrentHue":1792,"currentSaturation":255,"colorMode":1}},"genLevelCtrl":{"attributes":{"currentLevel":252}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"3.0.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1631336700380}`

So Gledopto have used in older controllers a manufacturer ID of 0. How should this be properly handled?

Koenkk, 2021-09-12:
@malaakso can you experiment with adding an additional property `manufacturerName` below https://github.com/Koenkk/zigbee2mqtt/blob/4a51e0c0c99dfeb4b319048db270c9eb69db419e/lib/extension/bind.js#L86 ? (you can use the `"manufName":"GLEDOPTO"` there)

malaakso, 2021-09-16:
> @malaakso can you experiment with adding an additional property `manufacturerName` below
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/4a51e0c0c99dfeb4b319048db270c9eb69db419e/lib/extension/bind.js#L86
> 
> ? (you can use the `"manufName":"GLEDOPTO"` there)
@Koenkk That seems to work. Would something like this be accepted?
`
diff --git a/lib/extension/bind.js b/lib/extension/bind.js
index ce4dd297..903502ba 100755
--- a/lib/extension/bind.js
+++ b/lib/extension/bind.js
@@ -87,6 +87,9 @@ const pollOnMessage = [
             zigbeeHersdman.Zcl.ManufacturerCode.GLEDOPTO_CO_LTD,
             zigbeeHersdman.Zcl.ManufacturerCode.MUELLER_LICHT_INT,
         ],
+        manufacturerNames: [
+            'GLEDOPTO',
+        ],
     },
     {
         cluster: {
@@ -117,6 +120,9 @@ const pollOnMessage = [
             zigbeeHersdman.Zcl.ManufacturerCode.GLEDOPTO_CO_LTD,
             zigbeeHersdman.Zcl.ManufacturerCode.MUELLER_LICHT_INT,
         ],
+        manufacturerNames: [
+            'GLEDOPTO',
+        ],
     },
     {
         cluster: {
@@ -143,6 +149,9 @@ const pollOnMessage = [
             zigbeeHersdman.Zcl.ManufacturerCode.GLEDOPTO_CO_LTD,
             zigbeeHersdman.Zcl.ManufacturerCode.MUELLER_LICHT_INT,
         ],
+        manufacturerNames: [
+            'GLEDOPTO',
+        ],
     },
 ];

@@ -438,7 +447,8 @@ class Bind extends Extension {

             for (const endpoint of toPoll) {
                 for (const poll of polls) {
-                    if (!poll.manufacturerIDs.includes(endpoint.getDevice().manufacturerID) ||
+                    if ((!poll.manufacturerIDs.includes(endpoint.getDevice().manufacturerID) &&
+                        !poll.manufacturerNames.includes(endpoint.getDevice().manufacturerName)) ||
                         !endpoint.supportsInputCluster(poll.read.cluster)) {
                         continue;
                     }
`

Koenkk, 2021-09-18:
Yes that is fine!

DJTerentjev, 2022-11-19:
@malaakso is this fixed was merged with main code?

malaakso, 2022-11-20:
@DJTerentjev unfortunately not. I didn't have time to get a development environment set up and didn't want to submit an untested merge request. In the meantime, I moved the Gledopto controllers to another location where they are not bound to a remote.

DJTerentjev, 2022-11-22:
@malaakso I checked and it works. Thank you! If you don't mind I will  merge it with main code.