Koenkk/zigbee2mqtt #7831: Phillips Hue Motion Sensor - Temperature N/A

SHxKM, 2021-06-19:
## What happened

I'm moving from ZHA to Z2M (well, hopefully).

## What did you expect to happen

After pairing the Phillips Hue motion sensor (shows up as `Hue motion sensor (9290012607)` in the device screen), I expect to see the temperature reported.

## How to reproduce it (minimal and precise)

Pair a Phillips Hue motion sensor, check if it reports temperature.

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: Conbee II
Adapter firmware version:

JDFS404, 2021-06-30:
Just let it wait couple of hours, I had the same with my Outdoor sensor and some of my indoor sensors (have 5 Hue Motion Sensors, 1 Hue Motion Sensor Outdoor).

SHxKM, 2021-07-03:
Yep. It eventually started recording.

closed 2021-07-03 by SHxKM