Koenkk/zigbee2mqtt #7837: cache_state as device specific option to make smoke detector work

pbel78, 2021-06-20:
Dear Mr. Koenkk. First thank you a lot for you continued support of zigbee2mqtt!

Many things work out of the box but currently, I'm running into an issue with state caching (at least that is what I guess):
As I'm using zigbee2mqtt together with home assistant state caching is on. This is perfect for 99% of the devices. So I don't want to turn it off generally

The problem is with the Xiaomi JTYJ-GD-01LM/BW
https://www.zigbee2mqtt.io/devices/JTYJ-GD-01LM_BW.html

In the case of fire, it reports correctly: smoke: true
After the fire: As in the following messages the smoke value is probably not set (I assume it is sending battery level)
As this is now enriched with the cached values smoke is still true.
This means that the following fires are not detected as there is no state change to smoke: true

And no. My appartment is not regulary on fire but I do test the smoke detectors from time to time.

So far the only way to reset it I found is to : stop the zigbee2mqtt container. Set the value manually in state.json to false and restart zigbee2mqtt (is the observation correct that changing the value while the container is running doesn't help?).

Regards
Petr

pbel78, 2021-06-20:
I continued testing and found the following:
If the smoke detector is triggered and fire is removed while the detector is on=> it will send a smoke false message.
If the smoke detector is triggered and the battery is removed => it will not send a message - that is clear. But it also doesn't send a message when the power is restored and fire is out
pressing the button on the device after power was restored will send a smoke: false message

So I would say: that can be classified as a user error on my side: but maybe it would help somebody else to add this documentation that after testing the button should be pressed. I know that is not an issue of zigbee2mqtt but as it could help somebody maybe something you could consider.

Koenkk, 2021-06-20:
To update the documentation, click the pencil icon in the right top of https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/JTYJ-GD-01LM_BW.md

closed 2021-06-20 by Koenkk