Koenkk/zigbee2mqtt #7841: Ikea Tradfri LED1937T5

szatan81, 2021-06-20:
## Information about the device + link

https://www.ikea.com/pl/pl/p/tradfri-zarowka-led-e27-470-lumenow-bezprzewodowy-przyciemniany-biale-spektrum-w-ksztalcie-rurki-biale-szklo-matowe-90461916/

## data/database.db entry of the device

{"id":47,"type":"Router","ieeeAddr":"0x680ae2fffe497f56","nwkAddr":47863,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbT120E27WSopal470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRIbulbT120E27WSopal470lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20200826","swBuildId":"1.0.010"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20200826","swBuildId":"1.0.010","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1624105175173}

Thanks

Koenkk, 2021-06-26:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2716

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-26 by Koenkk

szatan81, 2021-07-05:
Unfortunately device is still not supported.

It is added as:
 zigbeeModel: ['TRADFRI bulb E27 opal 470lm', 'TRADFRI bulb E27 W opal 470lm'],
 model: 'LED1937T5',

Maybe it should be fixed to:
 zigbeeModel: ['TRADFRI bulb T120 E27 opal 470lm', 'TRADFRI bulb T120 E27 W opal 470lm'],
 model: 'LED1937T5',

Thanks.

Koenkk, 2021-07-05:
You are right, added now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)