Koenkk/zigbee2mqtt #7843: Update impossible - corrupt

Garycooper07, 2021-06-20:


Garycooper07, 2021-06-20:
Hi ! I tried to update zigbee2mqtt on my raspberry Pi 4 with CC2531. I'm using dev branch and when I type "git fetch", I've got this issue :

"error: inflate: data stream error (invalid code lengths set)
fatal: l'objet empaqueté a46f06a31234d5f15322e7af9a98c1f427a92ea8 (stocké da                                                                                            ns .git/objects/pack/pack-e93a439156722b0c55211e08a0aa5520dd81ca21.pack) est                                                                                             corrompu
fatal: L'hôte distant a fermé la connexion de manière inattendue"

I tried with only "git pull" but same issue.

All  my devices work fine

## Debug info
Zigbee2MQTT version:  Zigbee2MQTT version 1.18.3-dev
Adapter hardware: CC2531
Adapter firmware version:

robertalexa, 2021-06-22:
@Garycooper07 This is an issue with your repo. You probably run git pull at some point, something happened, git started writing to a file and failed. That now means you have corrupted files.

Have a look at this answer describing the steps you need to fix your repo.

The alternative is to make a backup of your devices and configuration, delete everything and set the repo from scratch, restoring your things after.

Garycooper07, 2021-06-22:
Argh ... could you help me to do a backup of devices and configuration then restoring ?
Thks

robertalexa, 2021-06-22:
No problem.

All you need to do is make a copy of your data folder somewhere in a safe location.

Given that you are running commands, you could try this from your z2m folder
cp -R data ~/data

This will copy the folder and its contents in your users home directory ( ~ ) inside a folder called data.

I hope this helps and makes sense.

PS: i just realised i forgot to copy the answer describing how to fix git without a full re-clone
https://stackoverflow.com/a/23749070

I would start with that, if you can't fix git, only then attempt a backup and reclone. Once again, sorry for forgetting to attach that link

Garycooper07, 2021-07-04:
Solved after backup, delete everything and set the repo from scratch, restoring my things

closed 2021-07-04 by Garycooper07