Koenkk/zigbee2mqtt #7846: Z2M Bridge online status

jd766, 2021-06-21:
The topic `{base_topic}/bridge/state` reports the last published status of the bridge, this is apparently updated when z2m starts or stops. 
Of course, if the application crashes, this topic stays as it is, not informing us of the unavailability of the bridge. 
If I publish a different payload to the topic, it isn't changed until the next stop/start, the bridge effectively ignores it.

My suggestion is to make the bridge react to a change in the payload of `{base_topic}/bridge/state` and overwrite it with the correct status, in order to allow an interactive status check.

malaakso, 2021-06-21:
MQTT "last will and testament" would be the correct way to implement notification of ungraceful exit.

andrew-d, 2021-06-24:
It looks like z2m already uses the "last will and testament" feature - here:
https://github.com/Koenkk/zigbee2mqtt/blob/abd8a0926403d2b6d6c9a1a175f917262926c72f/lib/mqtt.js#L19-L23

rindlerblabla, 2021-06-24:
My experience is that it post 'offline' just right, but that 'online' is repeated now and then.

jd766, 2021-06-28:
Looks like last will and testament works, I don't know why in my previous tests killing z2m didn't result in the topic being updated, anyway I'd say this is solved, issue closed.

closed 2021-06-28 by jd766