Koenkk/zigbee2mqtt #7849: Would it be feasible to make z2m groups discoverably by Home Assistant? They could be exposed as a single entity using HA's MQTT [discovery](https://www.home-assistant.io/docs/mqtt/discovery/) feature.

thomasfals, 2021-06-22:
Would it be feasible to make z2m groups discoverably by Home Assistant? They could be exposed as a single entity using HA's MQTT [discovery](https://www.home-assistant.io/docs/mqtt/discovery/) feature.

_Originally posted by @m0wlheld in https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-760778762_

robertalexa, 2021-06-22:
I highly doubt that this will be possible. A zigbee group is not the same as a HA group, by any stretch of the imagination.

Think of it this way:
- each device has a set of "exposes" which define what sort of attributes get exposed, both to z2m Frontend and HA.
- each of these attributes are of a certain type - e.g. binary, enum etc.
- each of these types have a matching type in HA - so you have a 1 to 1 translation. 1 attribute to 1 entity
- a zigbee group has no knowledge of attributes of each device inside the group. It just acts as a "traffic officer" saying "hey you 2 devices, you need to run this command", and then each device goes and does that.
- so because of the above, you can't create a group in HA based on the devices inside per say.

I hope I have covered this correctly, and to my knowledge I think i did. But maybe I am overcomplicating.

In HA if you create a simple group, the group will not display the "type" specific attributes. For example there is a difference between a group and a light group, and only the light group would allow you to control brightness and effect for example.

A zigbee group has no such "type", so how would you judge that the group in HA needs to be a light group.

Hope this helps

Koenkk, 2021-06-22:
Dupe of #2410 and already implemented in the dev branch.

closed 2021-06-22 by Koenkk