Koenkk/zigbee2mqtt #7851: TuYa PIR sensor / RH3040

biz-tek, 2021-06-22:
Hi everyone, try to connect https://aliexpress.ru/item/1005001977463358.html 
- several months ago, was _not supported_
- yesterday _connected_ as **TuYa PIR sensor / RH3040**, but, occupancy sensor always detected after first motion detection

sorry, questions:
- is it right model connection?
- what should I do to fix it?
 
[image]
[image]

biz-tek, 2021-06-24:
after HA restart device working as expected

closed 2021-06-24 by biz-tek

krt8wz, 2022-01-26:
Is it still working for you? Unfortunately I have same issue with RH3040. Occupancy is always on...

[image]

<html>
<body>
<h1 class="card-header" style="color: rgb(225, 225, 225); font-family: Roboto, Noto, sans-serif; font-size: var(--ha-card-header-font-size, 24px); letter-spacing: -0.012em; line-height: 48px; padding: 12px 16px 16px; display: flex; margin-block: 0px; font-weight: normal; justify-content: space-between; align-items: center; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="card-header-text">system_health</div><ha-button-menu corner="BOTTOM_START" slot="toolbar-icon" style="display: inline-block; position: relative;"><div><slot name="trigger"><ha-icon-button slot="trigger" style="display: inline-block; outline: none;"><mwc-icon-button title="Copy menu" style="display: inline-block; outline: none; --mdc-ripple-color: currentcolor; -webkit-tap-highlight-color: transparent; vertical-align: top; --mdc-theme-on-primary: currentColor; --mdc-theme-text-disabled-on-light: var(--disabled-text-color);"><button class="mdc-icon-button mdc-icon-button--display-flex" aria-label="Copy menu" style="font-size: 24px; width: var(--mdc-icon-button-size, 48px); height: var(--mdc-icon-button-size, 48px); padding: calc( (var(--mdc-icon-button-size, 48px) - var(--mdc-icon-size, 24px)) / 2 ); display: inline-flex; position: relative; box-sizing: border-box; border: none; outline: none; background-color: transparent; fill: currentcolor; color: inherit; text-decoration: none; cursor: pointer; user-select: none; z-index: 0; overflow: visible; align-items: center; justify-content: center; vertical-align: top;"><span><slot><ha-svg-icon style="display: block; align-items: center; justify-content: center; position: relative; vertical-align: middle; fill: currentcolor; width: var(--mdc-icon-size, 24px); height: var(--mdc-icon-size, 24px);"><svg preserveAspectRatio="xMidYMid meet" focusable="false" viewBox="0 0 24 24"><g><path d="M19,21H8V7H19M19,5H8A2,2 0 0,0 6,7V21A2,2 0 0,0 8,23H19A2,2 0 0,0 21,21V7A2,2 0 0,0 19,5M16,1H4A2,2 0 0,0 2,3V17H4V3H16V1Z"></path></g></svg></ha-svg-icon></slot></span></button></mwc-icon-button></ha-icon-button></slot></div><mwc-menu></mwc-menu></ha-button-menu></h1><div class="card-content" style="padding: 0px 16px 16px; margin-top: -8px; color: rgb(225, 225, 225); font-family: Roboto, Noto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Version | core-2021.12.10
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.9.7
Operating System Family | Linux
Operating System Version | 5.10.91
CPU Architecture | x86_64
Timezone | Europe/Warsaw

</div>
</body>
</html>

biz-tek, 2022-01-27:
working good on same versions of Mosquitto & Zigbee2mqtt, i have supervised HA, but don't think it's matter in that case
try to remove batteries and place they back

krt8wz, 2022-01-27:
Unfortunately It didn't fix it. In history I see  that it's cleared one per minute and after another second it's back to occupation state... so maybe my sensore is broken?

DarkVillager, 2022-01-29:
> Unfortunately It didn't fix it. In history I see that it's cleared one per minute and after another second it's back to occupation state... so maybe my sensore is broken?

I have the same issue. I have no clue how to fix it.

kukudemajia, 2022-04-09:
I have two TuYa PIR sensors(RH3040).It's the same issue. occupancy sensor always detected after first motion detection.

DarkVillager, 2022-04-09:
Please reopen.

alex-ander-is, 2023-11-09:
Still an issue :(

kyberias, 2024-10-26:
I can confirm this is still an issue.