Koenkk/zigbee2mqtt #7852: seMetering calculating 'energy' from 'currentSummDelivered' fails on HAN Smart Meters

TheStigh, 2021-06-22:
Hi,

I'm experiencing an issue with the Datek Eva HAN Meter Reader device. From cluster `seMetering` it reads `currentSummDelivered` (attribute 0x0000:current summation delivered ) and it works between whole hour and for the next 25-30 minutes. After this the value get reported as 0 (zero).

What I've learned are that the Smart Meter send this value once per hour (whole hour). But after 25-30 minutes (fluctuate between this interval) I receive 
`debug 22-06-2021 16:30:12: Received Zigbee message from '0x086bd7fffeae5d53', type 'attributeReport', cluster 'seMetering', data '{"currentSummReceived":[0,0]}' from endpoint 1 with groupID 0`

This makes me assume the Smart Meter update it's value to 0 approx half way between each hour causing the new report with value [0,0]. I've tried to use different reportableChange ( 0, 1, [0,10], [1,1] etc ) without luck.
Should I add some code to check if value suddenly drops to 0 and then avoid updating attribute?

Any suggestion how I could resolve this?
Any suggestion with a code example would be appreciated :)

Note: I know this works outside Z2M so there is nothing wrong with the device and/or Smart Meter.

Documentation of the device says:
_min: 10, max: 3600, reportable_change: 0_

Device (code updated for a new PR to-come):
```
module.exports = [
    {
        zigbeeModel: ['Meter Reader-123'],
        model: 'HSE2905E',
        vendor: 'Datek',
        description: 'Datek Eva AMS HAN power-meter sensor',
        fromZigbee: [fz.metering, fz.electrical_measurement, fz.temperature],
        toZigbee: [],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const options = {min: constants.repInterval.MINUTE, max: constants.repInterval.HOUR, change: 0};
            await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'seMetering', 'msTemperatureMeasurement']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.rmsVoltage(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.HOUR, change: 0});
            await reporting.rmsCurrent(endpoint, {min: constants.repInterval.MINUTE, max: constants.repInterval.HOUR, change: 0});
            await endpoint.read('haElectricalMeasurement', ['rmsVoltagePhB', 'rmsVoltagePhC', 'rmsCurrentPhB', 'rmsCurrentPhC'], options);
            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.instantaneousDemand(endpoint, {min: 10, max: 3600, change: 0});
            await reporting.currentSummDelivered(endpoint, {min: 10, max: 3600, change: [1, 1]});
            await reporting.currentSummReceived(endpoint);
            await reporting.temperature(endpoint);
        },
        exposes: [e.power(), e.energy(), e.current(), e.voltage(), e.current_phase_b(), e.voltage_phase_b(), e.current_phase_c(),
            e.voltage_phase_c(), e.temperature()],
    },
];
```

fromZigbee.js:
```
    metering: {
        /**
         * When using this converter also add the following to the configure method of the device:
         * await readMeteringPowerConverterAttributes(endpoint);
         */
        cluster: 'seMetering',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = {};
            const multiplier = msg.endpoint.getClusterAttributeValue('seMetering', 'multiplier');
            const divisor = msg.endpoint.getClusterAttributeValue('seMetering', 'divisor');
            const factor = multiplier && divisor ? multiplier / divisor : null;

            if (msg.data.hasOwnProperty('instantaneousDemand')) {
                let power = msg.data['instantaneousDemand'];
                if (factor != null) {
                    power = (power * factor) * 1000; // kWh to Watt
                }
                payload.power = precisionRound(power, 2);
            }

            if (factor != null && (msg.data.hasOwnProperty('currentSummDelivered') ||
                msg.data.hasOwnProperty('currentSummReceived'))) {
                let energy = 0;
                if (msg.data.hasOwnProperty('currentSummDelivered')) {
                    const data = msg.data['currentSummDelivered'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    energy += value * factor;
                }
                if (msg.data.hasOwnProperty('currentSummReceived')) {
                    const data = msg.data['currentSummReceived'];
                    const value = (parseInt(data[0]) << 32) + parseInt(data[1]);
                    energy -= value * factor;
                }
                payload.energy = precisionRound(energy, 2);
            }

            return payload;
        },
    },
```

From /lib/reporting.js
```
    currentSummDelivered: async (endpoint, overrides) => {
        const p = payload('currentSummDelivered', 5, repInterval.HOUR, [1, 1], overrides);
        await endpoint.configureReporting('seMetering', p);
    },
```

Koenkk, 2021-06-22:
Changing `payload.energy = precisionRound(energy, 2);` to:

```
if (energy !== 0) {
  payload.energy = precisionRound(energy, 2);
}
```

should prevent the incorrect values reported by the device to be published.

If this works remove the `fz.metering` from this device and add `fz.datatek_metering`, add to `fromZigbee.js`

```js
datatek_metering: {
    cluster: 'seMetering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = converters.metering.convert(model, msg, publish, options, meta);
        // Filter incorrect 0 energy values reported by the device:
        // https://github.com/Koenkk/zigbee2mqtt/issues/7852
        if (result.energy === 0) {
            delete result.energy;
        }
        return result;
    },
},
```

TheStigh, 2021-06-22:
> Changing `payload.energy = precisionRound(energy, 2);` to:
> 
> ```
> if (energy !== 0) {
>   payload.energy = precisionRound(energy, 2);
> }
> ```

Did you commit this change now to fromZigbee.js ?

TheStigh, 2021-06-23:
Working perfect! As always you're the man! Thanks Koen ! 🥇

closed 2021-06-23 by TheStigh