Koenkk/zigbee2mqtt #7853: SmartThings SZ-SRN12N - Siren sound level has no effect

The-Deep-Sea, 2021-06-22:
## What happened

I bought three SmartThings SZ-SRN12N devices.
The volume is always the same regardless of the value passed by "level".

PS: One of the devices is louder than the others. However, the packaging was also a little different, although the model number printed on the device is identical.

## What did you expect to happen

The volume should change depending on the passed value.

## How to reproduce it (minimal and precise)

Send a "Warning" message to the device as described in https://www.zigbee2mqtt.io/devices/SZ-SRN12N.html.
Use different values ​​for "level".

## Debug info
Zigbee2MQTT version: 1.19.1 commit: 9bd4693
Adapter hardware: CC2652RB
Adapter firmware version: 20210120

Koenkk, 2021-06-22:
The included level is a standard Zigbee command, so my current guess is that device just does not support this (unless proven otherwise e.g. if it works with the original gateway).

The-Deep-Sea, 2021-06-23:
Okay, the fact with the standard wasn't that clear to me before. So I would say your guess is plausible.
And in the absence of an original gateway, I would close the topic.
Thank you.

closed 2021-06-23 by The-Deep-Sea