Koenkk/zigbee2mqtt #7854: Setting brightness and color at the same time results in the publish of two MQTT status messages

TheTmirror, 2021-06-22:
Hello to the community,

I'm having a problem / bug with the zigbee2mqtt software. The situation is the following:

## What happened
I'm using the [IKEA LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html) and want to set the brightness and the color (x, y) at the same time. Therefore I published the following MQTT message to zigbee2mqtt/Wohnzimmer/set

`{
  "brightness": 150,
  "color": {
    "x": 0.5,
    "y": 0.5
  }
}`

In return two messages are published by zigbee2mqtt to zigbee2mqtt/Wohnzimmer, containing the updated informations. In the first message only the brightness was changed, in the the second message the updated color was also present.

## What did you expect to happen
I expected that my published set message would result in only one reply message by zigbee2mqtt to zigbee2mqtt/Wohnzimmer, where all values were updated.

## How to reproduce it (minimal and precise)
Assuming zigbee2mqtt is running and IKEA LED1624G9 has the friendly name Wohnzimmer.

`sudo mosquitto_pub --username username --pw password -t zigbee2mqtt/Wohnzimmer/set -m "{\"brightness\": 150, \"color\": {\"x\": 0.5, \"y\": 05}}"`

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2652R (Electrolama zig-a-zig-ah! (zzh!))
Adapter firmware version: CC2652R_coordinator_20210120

## Question
I'm assuming what happens in the background that firstly the brightness is updated, then the new state is published, then the color is updated and at last the new state is published. What I'm trying to achieve is that first the brightness is updated, then the color is updated and then finally only one message, containing the updated status, is published to zigbee2mqtt/Wohnzimmer. Is this somehow possible?

Why I'm asking?
In my home automation system I subscribed to zigbee2mqtt/Wohnzimmer to sync the system with all status updates. The double "feedback" message by zigbee2mqtt as response to a single published message (with the wanted color / brightness) disturbes keeping track of the correct state. Won't go into detail here, because it has nothing to do with zigbee2mqtt direcetly. 

Hope you can help me out. Thanks in advance!

Best regards,
Tristan

sjorge, 2021-06-23:
https://www.zigbee2mqtt.io/information/configuration.html#device-specific-configuration

You probably want to add the `debounce` option

Koenkk, 2021-06-23:
@sjorge debounce doesn't fix this since it's only applied to receiving messages. These message are assumed states after a convertSet() has been executed. 

Currently one message is published for each used converter, e.g. `{"brightness": 150, "color": {"x": 0.5, "y": 0.5}}` uses the `brightness` and `color` converter leading to 2 published messages. I've fixed this now, each received `/set` MQTT message will result in one state publish now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-23 by Koenkk

sjorge, 2021-06-23:
> @sjorge debounce doesn't fix this since it's only applied to receiving messages. These message are assumed states after a convertSet() has been executed.

Interesting, I was under the assumption those would also get debounced

TheTmirror, 2021-06-24:
Thanks for the soloution @Koenkk and @sjorge for the suggestion. I'll check out the dev branch this evening.

Just out of curiosity: Was this an intended behaviour or just never thought about?

Koenkk, 2021-06-24:
@TheTmirror  never thought about

geforce28, 2023-11-30:
Hi @Koenkk 

Have you changed this ?
If i set 
{"brightness": 150, "color": {"x": 0.5, "y": 0.5}} for a Lamp, it results in 2 Messages
First Brightness and then Color.

How can I change this ?