Koenkk/zigbee2mqtt #7858: Executed scene: bulbs and device rgb is blinked latest state color

shirou93, 2021-06-22:
## What happened
Executed scene: bulbs and device rgb is blinked latest state color
[image]
i try: all rgb zigbee devices i have:
https://www.zigbee2mqtt.io/devices/AC25697.html#ledvance-ac25697
https://www.zigbee2mqtt.io/devices/TS0505B.html#tuya-ts0505b
https://www.zigbee2mqtt.io/devices/HG06106C.html#lidl-hg06106c

## What did you expect to happen
No blink, only color setuped in script.
## How to reproduce it (minimal and precise)
Change color in bulb, > turn off bulp > execute scene
## Debug info
Zigbee2MQTT version: 1.19.1 commit: 1.19.1
Adapter hardware: CC2652RB
Adapter firmware version: 20210120

 herdsman debug log attached:

`2021-06-22T21:14:20.903Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,56,254]
2021-06-22T21:14:20.903Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,56,254]
2021-06-22T21:14:20.903Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,56] - 254
2021-06-22T21:14:20.904Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":56}
2021-06-22T21:14:20.904Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-22T21:14:20.907Z zigbee-herdsman:controller:endpoint Read 0x60a423fffeff4829/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-06-22T21:14:20.907Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffeff4829:42393/1 (0,0,5)
2021-06-22T21:14:20.912Z zigbee-herdsman:controller:endpoint Read 0x60a423fffef8dbd5/1 haElectricalMeasurement(["rmsVoltage","rmsCurrent","activePower"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-06-22T21:14:20.913Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffef8dbd5:57761/1 (0,0,6)
2021-06-22T21:14:20.943Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-06-22T21:14:20.944Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-06-22T21:14:20.944Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-06-22T21:14:20.944Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-06-22T21:14:20.945Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-22T21:14:20.947Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":57987,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":58,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,48,0,5,5,8,5,11,5]}}
2021-06-22T21:14:20.947Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,131,226,1,1,4,11,58,0,30,9,16,48,0,5,5,8,5,11,5,86]
2021-06-22T21:14:20.948Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,57,255,254,41,68,129,0,0,4,11,2,255,1,1,0,63,0,1,80,76,0,0,21,24,46,1,5,5,0,33,232,0,8,5,0,33,40,0,11,5,0,41,4,0,2,255,29,16]
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,57,255,254,41,68,129,0,0,4,11,2,255,1,1,0,63,0,1,80,76,0,0,21,24,46,1,5,5,0,33,232,0,8,5,0,33,40,0,11,5,0,41,4,0,2,255,29,16]
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,57] - 255
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":57}
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,68,129,0,0,4,11,2,255,1,1,0,63,0,1,80,76,0,0,21,24,46,1,5,5,0,33,232,0,8,5,0,33,40,0,11,5,0,41,4,0,2,255,29,16]
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 41 - 2 - 4 - 129 - [0,0,4,11,2,255,1,1,0,63,0,1,80,76,0,0,21,24,46,1,5,5,0,33,232,0,8,5,0,33,40,0,11,5,0,41,4,0,2,255,29] - 16
2021-06-22T21:14:20.949Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":65282,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":63,"securityuse":0,"timestamp":5001217,"transseqnumber":0,"len":21,"data":{"type":"Buffer","data":[24,46,1,5,5,0,33,232,0,8,5,0,33,40,0,11,5,0,41,4,0]}}
2021-06-22T21:14:20.951Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":46,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1285,"status":0,"dataType":33,"attrData":232},{"attrId":1288,"status":0,"dataType":33,"attrData":40},{"attrId":1291,"status":0,"dataType":41,"attrData":4}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":65282,"endpoint":1,"linkquality":63,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-06-22T21:14:20.953Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-22T21:14:20.959Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,68,129,0,0,4,11,248,213,1,1,0,111,0,5,85,76,0,0,21,24,47,1,5,5,0,33,227,0,8,5,0,33,13,0,11,5,0,41,0,0,248,213,29,106]
2021-06-22T21:14:20.959Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,68,129,0,0,4,11,248,213,1,1,0,111,0,5,85,76,0,0,21,24,47,1,5,5,0,33,227,0,8,5,0,33,13,0,11,5,0,41,0,0,248,213,29,106]
2021-06-22T21:14:20.959Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 41 - 2 - 4 - 129 - [0,0,4,11,248,213,1,1,0,111,0,5,85,76,0,0,21,24,47,1,5,5,0,33,227,0,8,5,0,33,13,0,11,5,0,41,0,0,248,213,29] - 106
2021-06-22T21:14:20.959Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":54776,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":111,"securityuse":0,"timestamp":5002501,"transseqnumber":0,"len":21,"data":{"type":"Buffer","data":[24,47,1,5,5,0,33,227,0,8,5,0,33,13,0,11,5,0,41,0,0]}}
2021-06-22T21:14:20.963Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":47,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1285,"status":0,"dataType":33,"attrData":227},{"attrId":1288,"status":0,"dataType":33,"attrData":13},{"attrId":1291,"status":0,"dataType":41,"attrData":0}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":54776,"endpoint":1,"linkquality":111,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-06-22T21:14:20.965Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-22T21:14:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,58,252]
2021-06-22T21:14:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,58,252]
2021-06-22T21:14:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-06-22T21:14:20.971Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-06-22T21:14:20.971Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,58,252]
2021-06-22T21:14:20.972Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,58] - 252
2021-06-22T21:14:20.972Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":58}
2021-06-22T21:14:20.972Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-22T21:14:20.972Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":38649,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":59,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data": [... cut ...] 1,"status":0,"dataType":41,"attrData":0}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":42393,"endpoint":1,"linkquality":36,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-06-22T21:14:21.133Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []`

Koenkk, 2021-06-24:
The attached herdsman debug logging does not seem to be correct. I don't see any commands being send to the bulb there.

Please provide the STDOUT (this willl contain both the herdsman and zigbee2mqtt debug logginig), start capturing right before you send the command till 10 seconds after that. Please post the log on pastebin.com and link it here.

shirou93, 2021-06-24:
It's ok now? 
https://pastebin.com/sWCbPun2

Koenkk, 2021-06-26:
Yes, can you also reproduce this bug by just sending the payload `{"state": "ON", "brightness": 254, "color_temp": 150}` to `zigbee2mqtt/Światło Pokój/set`, if yes, does it also occur when sending:  `{"state": "ON", "brightness": 254}`?

shirou93, 2021-06-27:
{"state": "ON", "brightness": 254, "color_temp": 150 same issue blink 

{"state": "ON", "brightness": 254} turned on with last color before turn off. no blink.

Koenkk, 2021-06-28:
Does it also blink when sending: `{"state": "ON", "brightness": 254}`?

shirou93, 2021-06-28:
> Does it also blink when sending: `{"state": "ON", "brightness": 254}`?

No blink

Koenkk, 2021-06-28:
And with: 
- `{"state": "ON", "color_temp": 150}`
- `{"brightness": 254, ""color_temp": 150}`
- `{"color_temp": 150}`

shirou93, 2021-06-28:
> And with:
> 
> * `{"state": "ON", "color_temp": 150}`
> * `{"brightness": 254, ""color_temp": 150}`
> * `{"color_temp": 150}`

 {"state": "ON", "color_temp": 150} = Blink
 {"brightness": 254, "color_temp": 150} = Blink

{"color_temp": 150} No blink (if bulb turned off this command don't power on bulb)

Koenkk, 2021-06-28:
Thanks, can you now provide the herdsman debug log when just sending: `{"state": "ON", "color_temp": 150}`?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

shirou93, 2021-06-28:
https://pastebin.com/YcMuiA8D

Koenkk, 2021-06-29:
I've checked your log, but I think this issue is inevitable. `{"state": "ON", "color_temp": 150}` results in 2 zigbee commands
- turn on
- set color temp

The time between the commands is the blink, however there is no Zigbee command to turn on a bulb with a specific color temperature so always 2 need to be send.

shirou93, 2021-06-29:
Thanks for check.

closed 2021-07-31 by shirou93