Koenkk/zigbee2mqtt #7859: Lonsonho QS-Zigbee-C01 - Problem cover entity with 'state'

falbalala, 2021-06-23:
Zigbee2MQTT:error
> No converter available for 'get' 'state' ()

The value for the 'state' is 'null'

For the 'calibration', it's good. No problem.

PS: All is updated to last release.

SimonSezKossel, 2021-07-09:
I think i have the same problem.

Suddenly having a greyed out move blind entity.
Adding a new device does not even add this entity. 
[image]

In the Zigbee2mqtt its just functional:
[image]

RamblingGeekUK, 2021-07-10:
I'm having the same issue with IKEA E1926, it works in Zigbee2MQTT but can't see position, open, close state's in HA.  This was working, I had to start due to SD card corruption, and now it's not pulling through.

closed 2021-08-18 by github-actions[bot]

s3rgi, 2021-08-18:
I'm having the same issue, when re-pairing the device does not even add "cover" entity, only sensors like quality, motor_reversal and moving are available

SimonSezKossel, 2021-08-18:
I was able to solve my isssues by updating to the latent version of zigbee2mqtt.