Koenkk/zigbee2mqtt #7860: Some entities missing in HA when using external converter for TS011F with power monitoring. OK in z2m frontend

jcobreros, 2021-06-23:
So I got a couple of BlitzWolf BW-SHP15 smartplugs with power and voltage monitoring [here](https://www.banggood.com/2-Pcs-BlitzWolf-BW-SHP15-Zigbee-3_0-16A-3680W-Smart-Plug-Wireless-Power-Socket-Outlet-EU-Plug-APP-Remote-Control-or-Voice-Control-or-Multiple-Timer-Modes-Compatible-With-Amazon-Alexa-or-Google-Assistant-p-1859041.html?rmmds=myorder&cur_warehouse=CN)

Zigbee2mqtt detects it as `modelID: 'TS011F', manufacturerName: '_TZ3000_mraovvmm'` and complains that it's currently unsupported.

So I tried to make an external converter for it, seeing as it's pretty similar to other smartplugs based on the same model.
This is the code for my external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mraovvmm'}],
        model: 'TS011F_plug',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
    }
];
```

Zigbee2mqtt detects it correctly now. In z2m frontend, if I go to the device properties, and the Exposes tab, I can see things like: Status, Voltage, Current, Power, Link Quality... all of them updating with correct values.
[image]

Back to Home assistant, the device is detected successfully. I can control it and I can see things like Power and Current. But I can't see Voltage or Link quality. Those two variables are shown under "disabled entities" in the device properties panel.
[image]
[image]

More interestingly, if I select any of the working variables, like Power, and look at the history graph, under the graph there's a dropdown menu called Attributes, and that dropdown menu contains all the variables, including voltage and link quality.
This is happens with every one of the "working" entities. They all containg every variable under Attributes.
[image]

Has anyone had a similar issue? Is this possibly related to my external converter?

rindlerblabla, 2021-06-23:
Guess it is related to https://github.com/Koenkk/zigbee2mqtt/pull/7684.

jcobreros, 2021-06-23:
Oh man. That was so simple. I actually had tried to enable them and restart HA, but this didn't seem to work the first time. However I just did it again and it worked.

Thank you!

closed 2021-06-23 by jcobreros

maureis, 2021-08-11:
@jcobreros could you make PR with this new model converter? :)

biejay, 2021-08-11:
Would love to have these integrated in the next release!

jcobreros, 2021-08-13:
I will do it when I'm back home next week :)

lassieee, 2021-09-06:
I think with the latest zigbee2mqtt:1.21.1 the bw-shp15 powerplugs work out-of-the-box, can anyone confirm that?

maureis, 2021-09-06:
@lassieee I can confirm :)

biejay, 2021-09-06:
I got problems with the reading of the Power consumptions. Anyone Else?

lassieee, 2021-09-06:
In what sense are you having problems? I think it's working properly in my setup:

[image]
[image]
[image]

But I can't be 100% sure because I didn't actually measure the power consumption.

There is an issue though because there is no `total_consumption` `state_class`, so you can't select the entity in the HA energy dashboard to show up under 'individual devices', but as a workaround you can make a customization in HA to add the missing state_class, found here: https://github.com/StyraHem/ShellyForHASS/issues/578

So you can do something like this to add the state_class to all sensors that match the glob:
```
customize_glob:
    sensor.*_energy:
      last_reset: '1970-01-01T00:00:00+00:00'
      state_class: total_increasing
      device_class: energy
```

wvdmaarl, 2021-11-11:
I am trying to connect the BW-SHP15 smart plug with zigbee2mqtt but are trouble getting it to work. I have Zigbee2mqtt 1.22.0 installed. I have created BW-SHP15.js in the data folder:
---
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
        fingerprint: [{manufacturerID: '4098', manufacturerName: '_TZ3000_mraovvmm'}],
        model: 'BWP-SHP15',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
    }
];
---
When connecting I cannot find modelID, only manufacturer:"_TZ3000_mraovvm"

this is a part of the debug log:
"clusters":{"input":["genBasic","genIdentify","genGroups","genScenes","genOnOff","seMetering","haElectricalMeasurement","manuSpecificTuya_3"],"output":["genOta","genTime"]},"configured_reportings":[],"scenes":[]},"242":{"bindings":[],"clusters":{"input":[],"output":["greenPower"]},"configured_reportings":[],"scenes":[]}},"friendly_name":"0x84fd27fffece7ef3","ieee_address":"0x84fd27fffece7ef3","interview_completed":true,"interviewing":false,"manufacturer":"_TZ3000_mraovvmm","network_address":28129,"power_source":"Mains (single phase)","supported":false,"type":"Router"}]'

Thanks for help.

sygys, 2022-01-04:
im having the same problem that when i use the external converter the device keeps showing as unsupported in z2m. Im i doing something wrong or has the code been changed so this doesnt work anymore?