Koenkk/zigbee2mqtt #7861: Router lost after power outage

sineverba, 2021-06-23:
Zigbee2MQTT version: 1.19.1

Hi to all,
after an electricity outage of about 15 minutes, I have lost communication with my router (and some other device)

[image]

I tried to restart docker of zigbee2mqtt without luck.
I tried also to poweroff and poweron the coordinator, removing from the power socket.

How can I force it to rejoin?

Coordinator: CC2652RB
Router: CC2530

Thank you for the support!

MattWestb, 2021-06-24:
I think then the power was coming back the coordinator was not online and all the routers is doing one device accouterments and was not finding the coordinator for "rejoining" the network.

Try re power some of the easy accessed router and see if they is finding  the coordinator and rejoining the network.

sineverba, 2021-06-25:
Thank you but coordinator was online (it is connected to a Raspberry UPS powered). Btw I solved enabling joining and re-powering the router.
I'm closing, thank you!

closed 2021-06-25 by sineverba