Koenkk/zigbee2mqtt #7865: Pairing Xbee 3 to CC2531 fails

Genoil, 2021-06-24:
## What happened
Pairing / interviewing of a Digi Xbee 3 device fails.

## What did you expect to happen
Expected succesful pairing

## How to reproduce it (minimal and precise)
Using CC2531 dongle and vanilla Xbee3, used X-CTU to set protocol stack to 2 (Zigbee-PRO)

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2531
Adapter firmware version:   zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +1ms

## Log
https://hastebin.com/cekovamope.yaml

Genoil, 2021-06-25:
fixed by setting correct encryption settings

closed 2021-06-25 by Genoil

ClaudioABBueno, 2021-08-09:
which configuration did you use, I have the same problem?

Genoil, 2021-08-09:
https://www.digi.com/support/knowledge-base/zigbee-home-automation