Koenkk/zigbee2mqtt #7869: New Device - Tuya RGB-CCT Controller - HI004 / TS0505B

Stele77, 2021-06-24:
## Support for new device
Hi!
Ist just bought a ton of RGBCCT controllers, and stupidly did not check if they are supported. 
https://de.aliexpress.com/item/1005002092208628.html?spm=a2g0s.9042311.pcrcomd.1.27424c4duCmLbD&gps-id=pcOrderList&scm=1007.30114.213637.0&scm_id=1007.30114.213637.0&scm-url=1007.30114.213637.0&pvid=93ce71b7-470c-44e8-84cd-9750e47c8d4b&_t=gps-id:pcOrderList,scm-url:1007.30114.213637.0,pvid:93ce71b7-470c-44e8-84cd-9750e47c8d4b,tpp_buckets:668%230%23131923%230_668%23888%233325%239_20114%230%23213637%235_20114%233258%2310449%23185_668%232846%238108%231977_668%235811%2327186%2375_668%232717%237565%23723__668%233374%2315176%23480
They are made by Tuya, and are now flodding the market. I suspect there will be many others like me missing support, because they are by far the cheapest RGB-CCT controllers (8-10€) with high power input able to drive long light strips.
[image]

The package says: "Model No: HI004"
When i add it into Zigbee2Mqtt GUI, it pairs but says "unsupported" and shows Tuya as brand and as Model: TS0505B.
[image]

I searched this repo for open issues, and came up with this one: 
#https://github.com/Koenkk/zigbee-herdsman-converters/pull/2608

I says merged in it, that should mean it would be supported now by the stable or at least the dev Zigbee2mqtt HA addon, right?
But it sadly does not work.

Could you please help me? I am not very tech savvy, and coming so far to get HA and Zigbee to work was a hard journey already. Thanks in advance!

## Debug Info
Zigbee2MQTT version: latest stable and dev Home Assistant addons both tested
Adapter hardware: CC2652R zzh Electrolama
Adapter firmware version: Zigbee 3 - 20210120

marc86lavergne, 2021-06-28:
Hello I have exactly the same problem, Any help would be welcom.  I think the solution is with adding a convertor to the device but I do not know how to do it.

Stele77, 2021-06-29:
> Hello I have exactly the same problem, Any help would be welcom. I think the solution is with adding a convertor to the device but I do not know how to do it.

Yeah, setting up HA is already pushing me to the edge of my abilities. And looking at the docs for adding a converter makes my head explode.. And then those Tuya-specific extra steps?? 
Good to see i am not alone with this. Lets just hope there is someone with the same problem who is able to understand the docs and how the hell this converter should be made.
🙏🏻

marc86lavergne, 2021-06-29:
I am running Hassio on a raspberry PI and I have downloaded the fill package of herdsman-converters but I have no idea how to install it or where to copy it. 
If someone can tell me how, I have found no manual about this part that seems very basic....

drgrandios, 2021-06-30:
Did you look at the log file? I'm quite sure it reports an not (yet) known manufacturer name. 

The currently supported ones can be found here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/52ad2b3d8cc3cad9f6a30110059afe744db20d15/devices/tuya.js#L85

It is possibly just an additional entry needed there...

drgrandios, 2021-06-30:
If the manufacturer name is "_TZ3210_5snkkrxw", the issue has been fixed some hours ago... 

See https://github.com/Koenkk/zigbee2mqtt/issues/7906

I tried this with my LED controller and it works. Unfortunately the colors are completely off (e.g. red is green etc.) but basically it does the job!

Stele77, 2021-06-30:
> If the manufacturer name is "_TZ3210_5snkkrxw", the issue has been fixed some hours ago...
> 
> See #7906
> 
> I tried this with my LED controller and it works. Unfortunately the colors are completely off (e.g. red is green etc.) but basically it does the job!
> 
> edit: Seems to be just a naming mismatch (or incorrect labeling...). After swapping the channels on the device, everything works fine.

Thank you (and Koenkk )so much!! It does work, you made my day!
Will mark this as closed!

closed 2021-06-30 by Stele77

Slaw3km, 2022-01-16:
Did you manage to connect multiple such controllers? 
My instance connects always to only one controller - even if I have 3 of them connected next to each other. 
Any tip how to handle that?

ravaz, 2022-01-29:
Hello,

I got a similar problem. I bought the following lamp:

[https://www.aliexpress.com/item/1005003717039854.html?spm=a2g0o.9042311.0.0.18414c4dI5p8Lf](url)

The manufacturer is reported as "**_TZ3210_leyz4rju**", would it be possible to add it as well?

Koenkk, 2022-01-29:
@ravaz could you provide the data/database.db entry of `_TZ3210_leyz4rju` ?

Slaw3km, 2022-01-30:
@Koenkk any idea what could be the root cause of my issue?
Where to look for it? I am a dev so I could debug that. I wonder why other devices are not visible...

Koenkk, 2022-01-30:
@Slaw3km I don't understand what your problem is, but you should be able to pair multiple devices, just enable pairing and reset them one by one.

Slaw3km, 2022-01-30:
@Koenkk it works now... (as it always does when you reach out to "support")
Apparently I was not able to properly reset devices before. 
Now they reset correctly and pair nicely - thanks for helping :)

Like-electronics, 2022-02-19:
this morning i bought the bulb from Livarno in the Lidl.
The typenumber on the lamp is HG07843C
It is recognized in MQTT as TS0505B
But is said the manufacturer is unsupported.
On the website of ZIGBEE2MQTT it is a known divice. https://www.zigbee2mqtt.io/devices/TS0505B.html
How can it be sloved

Tsury, 2022-03-09:
> @ravaz could you provide the data/database.db entry of `_TZ3210_leyz4rju` ?

@Koenkk 
I bought this device as well. Here's the entry:

`{"id":58,"type":"Router","ieeeAddr":"0xa4c1386657c65725","nwkAddr":56855,"manufId":4417,"manufName":"_TZ3210_leyz4rju","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":94,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1646812680352,"defaultSendRequestWhen":"immediate"}`

I'd appreciate if support is added for it, I'm willing to supply whatever more information you require, thanks.

Koenkk, 2022-03-10:
@Tsury added

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

cronner, 2022-03-27:
I change to edge but still does not work, anyone else got it working?

Tsury, 2022-04-05:
> @Tsury added
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@Koenkk Is there a way to debug the pairing process? I'm unable to pair it for some reason. Using the new 1.25 version. Nothing happens when I try to permit it to join.

I've plugged it and unplugged it tens of times already, which is supposed to be the trigger to start the pairing.

I made sure it's not in the block list as well.

almirus, 2022-04-18:
new (same) model 
model Zigbee
**TS0505B**
manufacturer Zigbee
**_TZ3210_klv2wul0**

[image]
[image]

MegoVaTx, 2022-06-23:
> This morning i bought the bulb from Livarno in the Lidl. The typenumber on the lamp is HG07843C It is recognized in MQTT as TS0505B But is said the manufacturer is unsupported. On the website of ZIGBEE2MQTT it is a known divice. https://www.zigbee2mqtt.io/devices/TS0505B.html How can this be solved?

Having a Similar issue, some of the lights paired as Taya but some as _tz3210_iwbaamgh brand.
[image]

MegoVaTx, 2022-06-23:
> Having a Similar issue, some of the lights paired as Taya but some as _tz3210_iwbaamgh brand. [image]

{"id":19,"type":"Router","ieeeAddr":"0xa4c138ec36eed4a6","nwkAddr":59229,"manufId":4417,"manufName":"_TZ3210_iwbaamgh","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_iwbaamgh","powerSource":1,"zclVersion":3,"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":98,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1656026708966,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-06-24:
@MegoVaTx `_tz3210_iwbaamgh ` is already supported in the latest dev. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

dE1l, 2022-06-29:
@Koenkk 
new (same) model
model Zigbee
TS0504B
manufacturer Zigbee
_TZ3210_elzv6aia

[image]
[image]
[image]

Unsupported at Z2M Edge. Please add it

Koenkk, 2022-07-26:
@copelandchris added 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Fazer1000, 2022-09-13:
Hello,

I got a similar problem with my TS0505B.
The manufacturer is reported as "_TZ3210_r5afgmkl", would it be possible to add it as well?
I use the edge version of Zigbee2mqtt

Koenkk, 2022-09-14:
@Fazer1000 `_TZ3210_r5afgmkl` is already supported, you probably need to uninstall/install the addon the get the latest edge version.

Fazer1000, 2022-09-14:
> @Fazer1000 `_TZ3210_r5afgmkl` is already supported, you probably need to uninstall/install the addon the get the latest edge version.
@Koenkk Thank you for your quick response. Unfortunately, I cannot install new Z3M edge. When I go to Settings → Add-ons → Add-ons store → ⋮ → Check for updates, nothing happens at all. What can I do?

Fazer1000, 2022-09-14:
@Koenkk thx for the quick answer.

harfox, 2022-09-15:
Hi @Koenkk , can you look into mine? Couldn't complete the interview. Manufacturer model: _TZ3000_jtmhndw2. 

{"id":13,"type":"Router","ieeeAddr":"0x60a423fffe8951b9","nwkAddr":13434,"manufId":4098,"manufName":"_TZ3000_jtmhndw2","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1662912072226,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-09-16:
@harfox please check https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair first; if not provide the herdsman debug logging when pairing the device fails.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

harfox, 2022-09-16:
@Koenkk , Is this the correct log?

[z2m_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9585957/z2m_log.txt)

Koenkk, 2022-09-19:
@harfox since you are using the experimental EZSP adapter; today a fix has been integrated to improve reliability procedure (https://github.com/Koenkk/zigbee-herdsman/pull/580). Please try repairing with that fix.

These changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SteffenBMS, 2022-12-05:
@Koenkk , Could you also check if it´s possible to support the manufacturer "_TZ3210_w7ge4ldo" for a Zigbee controller of LED stripes? Thank you so much :)

SteffenBMS, 2022-12-07:
> @SteffenBMS can you check if it is already supported in z2m 1.28.4? If not provide me the data/database.db entry of this device.

I have to be honest, I dont know what you mean and what i should do :( Sorry

Koenkk, 2022-12-08:
Update to zigbee2mqtt 1.28.4 and check if the device works out-of-the-box.

SteffenBMS, 2022-12-08:
Thank you so much for the support and the good work. After updating it worked!!!!

xm4rcell0x, 2022-12-17:
Guys, I'm sorry if OT but I'm struggling to pair these devices. I bought 10 LED controller and can't pair none of them. Is there a reset procedure? I already tried to switch on-off 5 times, 7 times, 10 times ... but nothing . 
Thank you for your help.

falkotronic, 2023-02-28:
@SteffenBMS 

Hello,

I got a similar problem with my TS0505B.
The manufacturer is reported as "_TZ3210_eejm8dcr", would it be possible to add it as well? (Received message from unsupported device with Zigbee model 'TS0505B' and manufacturer name '_TZ3210_eejm8dcr')

srvduplex, 2023-04-07:
@SteffenBMS Hi!
Have been able to solve your problem /w _TZ3210_eejm8dcr manufacturer.
I got my led stripe from amazon and can get it work with homeassistant as it says unsupported.
cheers mate

falkotronic, 2023-04-15:
> 

Solved it by updating to the newest integration of zigbee2mqtt. It didn't update because an OS version update changed the data folder location.